DiedB/Homey-SolarPanels #103: Error retrieving data and slow refreshment

aswinvdkamp, 2020-04-27:
I have a data retrieving error (500) and there is no frequent sync off the energy in the app (zonnepanelen app homey). The energiemeter is frequently updatet. Sometimes the energy takes an our and the energymeter takes minutes.

DiedB, 2020-04-27:
Do you use other SolarEdge integrations as well? The API has a rate limit and the Homey app uses that to its limit.

aswinvdkamp, 2020-04-27:
I use the solar edge app, nothing else. I don't know of you mean this. Why is there a difference in refreshment between the energy and energymeter (I added a screenshot)

DiedB, 2020-04-27:
Each of those values is retrieved using a separate API call. One is succeeding (probably), the other hits the limit, which explains the difference between the two.

Could you logout of the SolarEdge app for a day and check whether it changes anything?

aswinvdkamp, 2020-04-29:
Did it, yesterday there where some data retrieving errors (403 and 500). Today no errors. The refreshment today was sometimes still an our or more for the energy, the energymeter had a shorter refreshment time. The same yesterday.I only use the energy for the flows.Greetings

DiedB, 2020-05-07:
How has it been going the past week?

aswinvdkamp, 2020-05-09:
The thirst 4 days there where stil some errors, after that no errors. I also use the solar edge app again en there last 3 days there where nog errors and the refreshment frequenty is around the 30min.Greetings,Aswin

DiedB, 2020-05-16:
Still no errors? In that case I will close the issue.

aswinvdkamp, 2020-05-16:
Only yesterday, today there where no errors.

DiedB, 2020-05-18:
Could you submit an error report (should be possible on the app settings page in the Homey app)? Want to make sure the app is not making too many calls.

aswinvdkamp, 2020-05-18:
A report from the app6ac52b2e-8906-4736-af3f-a53576efe35cGrAswin

DiedB, 2020-05-18:
Thanks! The one you submitted looks completely normal. Sometimes, no new data is available upon requesting the API, but that is just a limitiation on the refresh rate of data that is made available through the API.

Can you do this again when you encounter errors?

aswinvdkamp, 2020-05-31:
Can you do this again when you encounter errors?Dit it!1267642d-6ad2-4989-a8b5-d7c2eb6f3ffeGreetings,Aswin

DiedB, 2020-05-31:
Thanks! Seems like the error is on the side of SolarEdge, they probably do not have enough capacity to handle all requests. I thought it would be a rate limit previously, but that is not the case given the error code that is returned. I'm afraid we'll just have to wait until SolarEdge improves their service.

closed 2020-05-31 by DiedB