DiedB/Homey-SolarPanels #104: Omnik solar

Niekvb-homey, 2020-05-14:
Hi Diederik,
I have some problems (re)connecting my sunpanels (omnik). Today I received this letter from omnik. I think they've changed some necessary things.

"
Due to some issues of Omnik, the original portal can't be accessed recently, of which you might already notice. Therefore, we have prepared a temporary address to continue our monitoring services. 
Please kindly check the website below.
https://omnik1.solarmanpv.com

Omnik APP can't be accessed now due to the same reason. Developer of Omnik APP will adjust the APP. You will get update notification from the APP later.
"

DiedB, 2020-05-14:
Omnik's internet connectivity is a hot mess and they have demonstrated it again, unfortunately.

I will check whether I can make the Homey app work with the new URL, otherwise we'll just have to wait for them to fix it.

DiedB, 2020-05-16:
https://gathering.tweakers.net/forum/list_messages/1984936/1

People are reporting that the SolarmanPV portal is working, also with Omnik accounts. Could you give that a try and report back?

Niekvb-homey, 2020-05-16:
Thanks for the heads up. I was able to add the solar panels in homey (y) :D thanks for the good work, we're enjoying the app.

DiedB, 2020-05-16:
Interesting! Was it as simple as adding your Omnik account in the app under the SolarMAN driver?

Niekvb-homey, 2020-05-16:
It was as simple as adding my Omnik account in the app under "Omnik", not SolarMAN. Then my solarpanels popped-up. It seems to update the panels as usual.

DiedB, 2020-05-16:
So the integration is working like it normally did?

closed 2020-05-16 by DiedB

Niekvb-homey, 2020-05-16:
Indeed, I don't know if you changed something or omnik but it's working the the old way.
Sorry reading your text carefully now (was in the car when i first read it and tried it).

DiedB, 2020-05-18:
I think Omnik restored their API connection. Thanks!