DiedB/Homey-SolarPanels #108: Feature Request - Fronius Smart Meter integration

prj32, 2020-06-29:
Hi Team 

Great work with Homey integration with Fronius inverter.  Like most Solar investors understanding consumption is almost as equally important as understand production. It is my understanding many Fronius customers also have their companion Smart Meter (Smart Meter 63A-1) to understand a variety of other metrics, most importantly consumption in KWh. These metrics appear to be available to Fronius Solar Manager. Any plans to expand would be welcome. Happy to assist with testing.

DiedB, 2020-06-30:
Could you make a screenshot of this page?

http://**[fronius_ip]**/solar_api/v1/GetInverterRealtimeData.cgi?Scope=Device&DeviceID=1&DataCollection=CommonInverterData

prj32, 2020-06-30:
As requested 

{
   "Body" : {
      "Data" : {
         "DAY_ENERGY" : {
            "Unit" : "Wh",
            "Value" : 185.30000000000001
         },
         "DeviceStatus" : {
            "ErrorCode" : 0,
            "LEDColor" : 2,
            "LEDState" : 0,
            "MgmtTimerRemainingTime" : -1,
            "StateToReset" : false,
            "StatusCode" : 7
         },
         "FAC" : {
            "Unit" : "Hz",
            "Value" : 50.039999999999999
         },
         "IAC" : {
            "Unit" : "A",
            "Value" : 2.3399999999999999
         },
         "IDC" : {
            "Unit" : "A",
            "Value" : 1.74
         },
         "PAC" : {
            "Unit" : "W",
            "Value" : 511
         },
         "TOTAL_ENERGY" : {
            "Unit" : "Wh",
            "Value" : 31407200
         },
         "UAC" : {
            "Unit" : "V",
            "Value" : 238.5
         },
         "UDC" : {
            "Unit" : "V",
            "Value" : 342.5
         },
         "YEAR_ENERGY" : {
            "Unit" : "Wh",
            "Value" : 101641.10000000001
         }
      }
   },
   "Head" : {
      "RequestArguments" : {
         "DataCollection" : "CommonInverterData",
         "DeviceClass" : "Inverter",
         "DeviceId" : "1",
         "Scope" : "Device"
      },
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2020-07-01T07:53:44+10:00"
   }
}

DiedB, 2020-07-27:
Can you also get `http://[fronius_ip]/solar_api/v1/GetMeterRealtimeData.cgi?Scope=System` and post the result here?

ThomasHoussin, 2020-10-09:
Hi. I have a Fronius SmartMeter, the values are not reported with GetInverterRealtimeData. My results with http://[fronius_ip]//solar_api/v1/GetMeterRealtimeData.cgi?Scope=System below : 

{
   "Body" : {
      "Data" : {
         "0" : {
            "Current_AC_Phase_1" : 14.752000000000001,
            "Current_AC_Sum" : 14.752000000000001,
            "Details" : {
               "Manufacturer" : "Fronius",
               "Model" : "Smart Meter 63A-1",
               "Serial" : "20200200"
            },
            "Enable" : 1,
            "EnergyReactive_VArAC_Phase_1_Consumed" : 260,
            "EnergyReactive_VArAC_Phase_1_Produced" : 13610,
            "EnergyReactive_VArAC_Sum_Consumed" : 260,
            "EnergyReactive_VArAC_Sum_Produced" : 13610,
            "EnergyReal_WAC_Minus_Absolute" : 567,
            "EnergyReal_WAC_Phase_1_Consumed" : 2039,
            "EnergyReal_WAC_Phase_1_Produced" : 567,
            "EnergyReal_WAC_Plus_Absolute" : 2039,
            "EnergyReal_WAC_Sum_Consumed" : 2039,
            "EnergyReal_WAC_Sum_Produced" : 567,
            "Frequency_Phase_Average" : 50,
            "Meter_Location_Current" : 0,
            "PowerApparent_S_Phase_1" : 3285.27,
            "PowerApparent_S_Sum" : 3285.27,
            "PowerFactor_Phase_1" : 0.97999999999999998,
            "PowerFactor_Sum" : 0.97999999999999998,
            "PowerReactive_Q_Phase_1" : -423.31,
            "PowerReactive_Q_Sum" : -423.31,
            "PowerReal_P_Phase_1" : 3233.5,
            "PowerReal_P_Sum" : 3233.5,
            "TimeStamp" : 1602254495,
            "Visible" : 1,
            "Voltage_AC_Phase_1" : 222.69999999999999
         }
      }
   },
   "Head" : {
      "RequestArguments" : {
         "DeviceClass" : "Meter",
         "Scope" : "System"
      },
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2020-10-09T16:41:35+02:00"
   }
}

DiedB, 2020-10-10:
@prj32 You have both the Smart Meter and a Fronius inverter, I suppose? And @ThomasHoussin only has the Smart Meter?

I think the Smart Meter should be implemented as a separate device in the Homey app (which you can pick during pairing). I am planning to make some individual apps because the Solar Panels app is getting a little complicated and cluttered. Fronius would be on the list for a separate app too. I can only start development once I can install Homey firmware 5.0.0 on my own Homey, so that will take a short while. I'll get back to this thread when I'm ready.

Bascht74, 2020-10-10:
I'm in, too :-)

Cheers, 

Sebastian

ThomasHoussin, 2020-10-11:
I have both a SmartMeter and an inverter (request is the same as shown by prj32) - you may need both. After browsing through the API, I think that GetPowerFlowRealtimeData my be useful and easier. The result for request http://[fronius_ip]/solar_api/v1/GetPowerFlowRealtimeData.fcgi : 

{
   "Body" : {
      "Data" : {
         "Inverters" : {
            "1" : {
               "DT" : 81,
               "E_Day" : 222.19999694824219,
               "E_Total" : 11254.0009765625,
               "E_Year" : 11254.900390625,
               "P" : 458
            }
         },
         "Site" : {
            "E_Day" : 222.19999694824219,
            "E_Total" : 11254.0009765625,
            "E_Year" : 11254.900390625,
            "Meter_Location" : "grid",
            "Mode" : "meter",
            "P_Akku" : null,
            "P_Grid" : 511.19,
            "P_Load" : -969.19000000000005,
            "P_PV" : 458,
            "rel_Autonomy" : 47.2559560044986,
            "rel_SelfConsumption" : 100
         },
         "Version" : "12"
      }
   },
   "Head" : {
      "RequestArguments" : {},
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2020-10-11T12:20:05+02:00"
   }
}

with for example : 
#this value is null if inverter is not running ( + production ( default ) )
number P_PV;
#this value is null if no meter is enabled ( + from grid , - to grid )
number P_Grid;
#this value is null if no meter is enabled ( + generator , - consumer )
number P_Load;
#this value is null if no battery is active ( - charge , + discharge )
number P_Akku;

DiedB, 2020-10-12:
I wonder if `GetPowerFlowRealtimeData` is accessible for users without a smart meter?

ThomasHoussin, 2020-10-12:
According to the full documentation, without a smart meter P_Grid and P_Load have a null return value. But you still have P_PV, E_day and E_year.
(without an inverter these one should be null and you have values for P_Grid and P_Load)

ThomasHoussin, 2020-11-29:
Hi, 

For my own use I created an app that supports Fronius inverter and Fronius Smart Meter, and provides more data. It also fixes for me #128. I guess it should also fix issue #106, may be #134 (not tested however). Hope it can help

App is here : https://github.com/ThomasHoussin/com.thomashoussin.fronius

prj32, 2020-12-05:
> Can you also get `http://[fronius_ip]/solar_api/v1/GetMeterRealtimeData.cgi?Scope=System` and post the result here?

Sorry guys, I missed your last update. Here is the output from additional command with latest firmware running.

{
   "Body" : {
      "Data" : {
         "0" : {
            "Current_AC_Phase_1" : 15.188000000000001,
            "Current_AC_Sum" : 15.188000000000001,
            "Details" : {
               "Manufacturer" : "Fronius",
               "Model" : "Smart Meter 63A-1",
               "Serial" : "16510612"
            },
            "Enable" : 1,
            "EnergyReactive_VArAC_Phase_1_Consumed" : 21272120,
            "EnergyReactive_VArAC_Phase_1_Produced" : 73667480,
            "EnergyReactive_VArAC_Sum_Consumed" : 21272120,
            "EnergyReactive_VArAC_Sum_Produced" : 73667480,
            "EnergyReal_WAC_Minus_Absolute" : 59603505,
            "EnergyReal_WAC_Phase_1_Consumed" : 59603505,
            "EnergyReal_WAC_Phase_1_Produced" : 0,
            "EnergyReal_WAC_Plus_Absolute" : 0,
            "EnergyReal_WAC_Sum_Consumed" : 59603505,
            "EnergyReal_WAC_Sum_Produced" : 0,
            "Frequency_Phase_Average" : 49.899999999999999,
            "Meter_Location_Current" : 1,
            "PowerApparent_S_Phase_1" : 3684.5999999999999,
            "PowerApparent_S_Sum" : 3684.5999999999999,
            "PowerFactor_Phase_1" : -0.95999999999999996,
            "PowerFactor_Sum" : -0.95999999999999996,
            "PowerReactive_Q_Phase_1" : 286.35000000000002,
            "PowerReactive_Q_Sum" : 286.35000000000002,
            "PowerReal_P_Phase_1" : -3537.9400000000001,
            "PowerReal_P_Sum" : -3537.9400000000001,
            "TimeStamp" : 1607205559,
            "Visible" : 1,
            "Voltage_AC_Phase_1" : 242.59999999999999
         }
      }
   },
   "Head" : {
      "RequestArguments" : {
         "DeviceClass" : "Meter",
         "Scope" : "System"
      },
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2020-12-06T07:59:20+10:00"
   }
}

prj32, 2020-12-05:
Yes I have both Smart Meter and Inverter

DiedB, 2020-12-15:
@ThomasHoussin great work!

I was planning on slowly adding SDKv3 support for this app by releasing separate apps for most inverter types. Could we collaborate on upgrading your app to SDKv3 and release it separately in the Homey App Store? I can then deprecate support for the Fronius part of this app.

ThomasHoussin, 2020-12-20:
Yes, thanks ! I just migrated to Homey v5 to be able to try SDK v3 ; I just made some quick tests but migrating to v3 doesn't seem difficult (for now...).

DiedB, 2020-12-20:
Athom documentation is quite clear on the things that have been deprecated. Main thing is the cron library that has been removed.

closed 2022-01-22 by DiedB