DiedB/Homey-SolarPanels #110: Enphase Envoy Local API

KvDaalen, 2020-07-10:
I have an Enphase micro inverter installation and would like to use the Homey SolarPanels App to integrate data that the Envoy Metered produces into my Homey. The Envoy metered is a type of Envoy that next to production also measures data for import/export and consumption. It does this via measuring devices that need to be installed separately in the fuse box.
The current SolarPannels app does not support the Local API of the Enphase Envoy and it does also not deliver the Import/Export data nor consumption data. Because of the usage of the Cloud API, also the polling frequency is very low and on-boarding is cumbersome.

I suggest to enhance the SolarPanels app for enphase by implementing the local Envoy API instead of the Cloud API.
This enables the app to:

- Have a much easier on-boarding and configuration (actually none)
- Have a very fast update frequency (up to 1 second is doable from an envoy perspective)
- Expose detailed data about consumption and import/export.
- include Envoy Battery state (not installed in my system)
- Give more detail on this data by including the lines

The local API can be called via the following URL (no authentication is needed)

GET http://<local-ip>/production.json?details=1
or
GET http://envoy.local/production.json?details=1

The output will be:
`
{
	"production": [
		{
			"type": "inverters",
			"activeCount": 28,
			"readingTime": 1593511436,
			"wNow": 3341,
			"whLifetime": 12388248
		},
		{
			"type": "eim",
			"activeCount": 1,
			"measurementType": "production",
			"readingTime": 1593511437,
			"wNow": 1694.568,
			"whLifetime": 12458555.621,
			"varhLeadLifetime": 0.0,
			"varhLagLifetime": 6479514.629,
			"vahLifetime": 17158417.314,
			"rmsCurrent": 7.622,
			"rmsVoltage": 693.064,
			"reactPwr": 421.765,
			"apprntPwr": 1758.862,
			"pwrFactor": 0.96,
			"whToday": 8593.621,
			"whLastSevenDays": 305566.621,
			"vahToday": 12488.314,
			"varhLeadToday": 0.0,
			"varhLagToday": 5186.629,
			"lines": [
				{
					"wNow": 540.521,
					"whLifetime": 4009258.266,
					"varhLeadLifetime": 0.0,
					"varhLagLifetime": 2132369.76,
					"vahLifetime": 5681543.437,
					"rmsCurrent": 2.447,
					"rmsVoltage": 230.907,
					"reactPwr": 137.9,
					"apprntPwr": 564.291,
					"pwrFactor": 0.95,
					"whToday": 2731.266,
					"whLastSevenDays": 98080.266,
					"vahToday": 4128.437,
					"varhLeadToday": 0.0,
					"varhLagToday": 1707.76
				},
				{
					"wNow": 475.897,
					"whLifetime": 3376262.813,
					"varhLeadLifetime": 0.0,
					"varhLagLifetime": 1824771.386,
					"vahLifetime": 4596161.622,
					"rmsCurrent": 2.139,
					"rmsVoltage": 231.04,
					"reactPwr": 119.74,
					"apprntPwr": 493.494,
					"pwrFactor": 0.97,
					"whToday": 2430.813,
					"whLastSevenDays": 85166.813,
					"vahToday": 3439.622,
					"varhLeadToday": 0.0,
					"varhLagToday": 1461.386
				},
				{
					"wNow": 678.15,
					"whLifetime": 5073034.543,
					"varhLeadLifetime": 0.0,
					"varhLagLifetime": 2522373.483,
					"vahLifetime": 6880712.255,
					"rmsCurrent": 3.036,
					"rmsVoltage": 231.117,
					"reactPwr": 164.125,
					"apprntPwr": 701.077,
					"pwrFactor": 0.97,
					"whToday": 3431.543,
					"whLastSevenDays": 122319.543,
					"vahToday": 4920.255,
					"varhLeadToday": 0.0,
					"varhLagToday": 2017.483
				}
			]
		}
	],
	"consumption": [
		{
			"type": "eim",
			"activeCount": 1,
			"measurementType": "total-consumption",
			"readingTime": 1593511437,
			"wNow": 402.496,
			"whLifetime": 7757008.735,
			"varhLeadLifetime": 10143924.871,
			"varhLagLifetime": 6822949.748,
			"vahLifetime": 21724196.048,
			"rmsCurrent": 0.985,
			"rmsVoltage": 693.209,
			"reactPwr": -1158.324,
			"apprntPwr": 682.951,
			"pwrFactor": 0.59,
			"whToday": 6625.735,
			"whLastSevenDays": 98733.735,
			"vahToday": 15448.048,
			"varhLeadToday": 8274.871,
			"varhLagToday": 5239.748,
			"lines": [
				{
					"wNow": 29.248,
					"whLifetime": 800125.515,
					"varhLeadLifetime": 3317291.561,
					"varhLagLifetime": 2141228.044,
					"vahLifetime": 6463775.681,
					"rmsCurrent": 0.024,
					"rmsVoltage": 230.819,
					"reactPwr": -348.292,
					"apprntPwr": 5.49,
					"pwrFactor": 1.0,
					"whToday": 402.515,
					"whLastSevenDays": 6385.515,
					"vahToday": 4677.681,
					"varhLeadToday": 2585.561,
					"varhLagToday": 1708.044
				},
				{
					"wNow": 233.336,
					"whLifetime": 2946940.308,
					"varhLeadLifetime": 2784437.965,
					"varhLagLifetime": 2154574.904,
					"vahLifetime": 6411584.138,
					"rmsCurrent": 0.612,
					"rmsVoltage": 231.161,
					"reactPwr": -357.569,
					"apprntPwr": 141.577,
					"pwrFactor": 1.0,
					"whToday": 2400.308,
					"whLastSevenDays": 44066.308,
					"vahToday": 4515.138,
					"varhLeadToday": 2614.965,
					"varhLagToday": 1460.904
				},
				{
					"wNow": 139.912,
					"whLifetime": 4009942.912,
					"varhLeadLifetime": 4042195.344,
					"varhLagLifetime": 2527146.801,
					"vahLifetime": 8848836.229,
					"rmsCurrent": 0.349,
					"rmsVoltage": 231.229,
					"reactPwr": -452.464,
					"apprntPwr": 80.689,
					"pwrFactor": 1.0,
					"whToday": 3822.912,
					"whLastSevenDays": 48281.912,
					"vahToday": 6255.229,
					"varhLeadToday": 3074.344,
					"varhLagToday": 2070.801
				}
			]
		},
		{
			"type": "eim",
			"activeCount": 1,
			"measurementType": "net-consumption",
			"readingTime": 1593511437,
			"wNow": -1292.072,
			"whLifetime": 5244504.564,
			"varhLeadLifetime": 10143924.871,
			"varhLagLifetime": 343435.119,
			"vahLifetime": 21724196.048,
			"rmsCurrent": 6.637,
			"rmsVoltage": 693.353,
			"reactPwr": -736.559,
			"apprntPwr": 1530.874,
			"pwrFactor": -0.84,
			"whToday": 0,
			"whLastSevenDays": 0,
			"vahToday": 0,
			"varhLeadToday": 0,
			"varhLagToday": 0,
			"lines": [
				{
					"wNow": -511.273,
					"whLifetime": 492580.932,
					"varhLeadLifetime": 3317291.561,
					"varhLagLifetime": 8858.283,
					"vahLifetime": 6463775.681,
					"rmsCurrent": 2.424,
					"rmsVoltage": 230.731,
					"reactPwr": -210.392,
					"apprntPwr": 556.457,
					"pwrFactor": -0.91,
					"whToday": 0,
					"whLastSevenDays": 0,
					"vahToday": 0,
					"varhLeadToday": 0,
					"varhLagToday": 0
				},
				{
					"wNow": -242.561,
					"whLifetime": 2036429.36,
					"varhLeadLifetime": 2784437.965,
					"varhLagLifetime": 329803.518,
					"vahLifetime": 6411584.138,
					"rmsCurrent": 1.527,
					"rmsVoltage": 231.282,
					"reactPwr": -237.829,
					"apprntPwr": 352.291,
					"pwrFactor": -0.69,
					"whToday": 0,
					"whLastSevenDays": 0,
					"vahToday": 0,
					"varhLeadToday": 0,
					"varhLagToday": 0
				},
				{
					"wNow": -538.238,
					"whLifetime": 2715494.271,
					"varhLeadLifetime": 4042195.344,
					"varhLagLifetime": 4773.318,
					"vahLifetime": 8848836.229,
					"rmsCurrent": 2.687,
					"rmsVoltage": 231.34,
					"reactPwr": -288.339,
					"apprntPwr": 622.125,
					"pwrFactor": -0.87,
					"whToday": 0,
					"whLastSevenDays": 0,
					"vahToday": 0,
					"varhLeadToday": 0,
					"varhLagToday": 0
				}
			]
		}
	],
	"storage": [
		{
			"type": "acb",
			"activeCount": 0,
			"readingTime": 0,
			"wNow": 0,
			"whNow": 0,
			"state": "idle"
		}
	]
}
`

I found that in order to get the most current power production you must follow the path $.production[1].wNow (eim). I do not know what $.production[0].wNow (inverters) means, but it is not the most current value. Might be a rolling average.

DiedB, 2020-07-10:
Thanks for the detailed information! Will look into implementing this somewhere in the next few weeks.

KvDaalen, 2020-07-10:
> Thanks for the detailed information! Will look into implementing this somewhere in the next few weeks.

Great.

Retrospected, 2020-07-14:
nice, i was looking into this api last week as well. would be great if this can be implemented.

Not sure which value to take though, I'll run some analysis tomorrow during the day to see if i can figure that out.

Retrospected, 2020-07-16:
> 
> 
> nice, i was looking into this api last week as well. would be great if this can be implemented.
> 
> Not sure which value to take though, I'll run some analysis tomorrow during the day to see if i can figure that out.

Ran it for 24 hours and compared the local API value of **.production[0].wNow** with the ones sent via the Envoy API and it seems to be quite spot on in contrast to what @KvDaalen mentioned

[image]

KvDaalen, 2020-07-16:
> Ran it for 24 hours and compared the local API value of **.production[0].wNow** with the ones sent via the Envoy API and it seems to be quite spot on in contrast to what @KvDaalen mentioned
>
> [image]

I think you misunderstood what I was explaining. Your findings are not in contrast to mine. The .production[0].wNow gives a rolling average (also used within the cloud API). If you however need the actual production at the moment you call the API the .production[1].wNow gives the right value. 

Evidence:
1. If you call the API many times, each time the .production[1].wNow value is different. The .production[0].wNow value remains the same for a long duration (at least 15 minutes). 
2. The values as seen in the webpage http://envoy.local/home are reflecting the .production[1].wNow value
3. Using the Installer toolkit the Meters Tab.Production shows the .production[1].wNow data (including the lines). 
 
So you are correct. The data as reported in the Cloud API has a very low resolution. It has intervals of 15 minutes of reporting. This is reflected in the local API via the .production[0].wNow value. 

However I think that if you have the possibility to obtain data which is more accurate and has a higher resolution you should use that source. Automation on realtime data is much better than on averages.

Example of usage:
- Turn on boiler when more then 2000KW is exported.
- Turn off boiler when you start to import power from the grid.

When you use the low quality, low resolution data you are bound to keep the boiler running whilst you are actually importing power from the grid. This is because you are using a data point which by itself is a rolling average. After a while the fact that a cloud appeared (leading to lower production) will also be reflected in you rolling average data points. but that takes time. Better to automate on realtime actuals then on averages .

Retrospected, 2020-07-16:
Ah yes I see, I misunderstood you because for me .production[1].wNow does not return a valid value at all:
```
user@server:~ $ curl -s http://envoy.local/production.json?details=1 | jq '.production[1].wNow'
-6.379
```

I think this has to do with the fact that "metering" was not enabled by the installers. This is also shown on the /home page:

[image]

I contacted the installing party (CoolBlue) before about this and they confirmed that they do not connect the "Metered" function as part of their service yet :(. I checked out the Envoy manual and it requires the installation and wiring of an extra "CT" (Current Transformer) component, so it is not just a simple configuration on the software side.

Long story short: the value of **.production[1].wNow** indeed sounds like the most accurate value.

Ideally the feature would support both unmetered as well as the metered installation. _I think_ this can be check via the same json results using **.production[1].activeCount**. For me (unmetered) this is on 0, for @KvDaalen (metered) this is 1.

Alternatively i can also fork it afterwards and change the JSON query myself to support the local API on my unmetered version, until i figured out how to wire and configure metering without blowing up my house. :)

DiedB, 2020-07-23:
Thanks for the extended research on the differences between the metered and unmetered setups. I wonder why the unmetered Envoy does not have a higher resolution, it should have more granular production values right? Most inverters report at least every few seconds (on an LCD or something). Maybe I don't understand the Enphase setup quite well.

Could you guys check whether your Envoy responds to mDNS queries? This would improve the pairing process. You can check this using https://apps.apple.com/us/app/discovery-dns-sd-browser/id305441017 or https://hobbyistsoftware.com/bonjourbrowser or https://play.google.com/store/apps/details?id=com.druk.servicebrowser&hl=en

If it does, please make a screenshot of all values for the Envoy service.

Retrospected, 2020-07-24:
> Thanks for the extended research on the differences between the metered and unmetered setups. I wonder why the unmetered Envoy does not have a higher resolution, it should have more granular [...]

Sure:

envoy._enphase-envoy._tcp.local.
Enphase Energy Envoy

envoy.local:80
192.168.1.114:80
fd50:d85:9682:1:aae2:c1ff:fe5a:217c:80

protovers = 4.10.35
serialnum = 122015061668
txtvers = 1

DiedB, 2020-07-24:
Perfect, that should make pairing a lot more user-friendly ;-)

DiedB, 2020-07-24:
@Retrospected could you send your output of `http://envoy.local/production.json?detail=1`, like @KvDaalen did?

DiedB, 2020-07-24:
I have built an implementation, could you give it a try? It should work for both situations (metered and unmetered).

You can find it here: https://homey.app/nl-nl/app/it.diederik.solar/Zonnepanelen/test/

Please check if it works and send me a crash report after a few minutes (from the Homey app). Chances are it doesn't, since I have no way to test it myself 😄

Retrospected, 2020-07-24:
> @Retrospected could you send your output of `http://envoy.local/production.json?details=1`, like @KvDaalen did?

Thanks! i will test the build

In case you still need my production.json output:

[production.txt](https://github.com/DiedB/Homey-SolarPanels/files/4971682/production.txt)

DiedB, 2020-07-24:
Ah, your JSON output is different than I anticipated. If the app works, you will probably get some strange production/consumption values.

I'll implement the `activeCount` check you mentioned. Still, let me know if you are able to pair your inverter and if you see any output.

Retrospected, 2020-07-24:
I tried to pair it, but the device can’t be found
I double checked and i still do see the mDNS broadcasting on the same device.

[image]

DiedB, 2020-07-24:
Strange, I have no idea why it wouldn't find your Envoy. Sure Homey is on the same network (segment)?

Retrospected, 2020-07-24:
> 
> 
> Strange, I have no idea why it wouldn't find your Envoy. Sure Homey is on the same network (segment)?

Hmm that's strange... i still have the issue. Envoy, Homey and my phone are in the same WiFi network/VLAN and subnet. My phone can see the mdns just fine. 

Is there any way i can help out with debugging? If you can add some console.log's and push it to a remote branch i can give you the output by running the version manually myself.

DiedB, 2020-07-25:
Do you have experience with running Homey apps? You can try to change the mDNS query in `.homeycompose/discovery/enphase-envoy.json`. Maybe try changing the name to `enphase-envoy`?

Retrospected, 2020-07-25:
> Do you have experience with manually running Homey apps? You can try to change the mDNS query in `.homeycompose/discovery/enphase-envoy.json`. Maybe try changing the name to `enphase-envoy`?

Yes i know how to run local apps based on the source code repo. But i cant seem to find the branch you pushed it to :o did you push it to github? If so can you link it 🙈?

DiedB, 2020-07-25:
Forgot to push, see b43fc9f.

KvDaalen, 2020-07-26:
> I have built an implementation, could you give it a try? It should work for both situations (metered and unmetered).
>
> You can find it here: [...]

Hi Tried to install the new test version. That seems to have succeeded as it now mentions that the zonnepanelen app is of version 4.1.

Then tried to add a new device (enphase Envoy) But it does not find any.

geurti64, 2020-07-26:
Here it also doesn't find anything when I try to install Enphase Envoy.
Should I give permission somewhere in my Envoy to work with a third party?

DiedB, 2020-07-26:
It's an issue with the (beta) app, not your Envoy. @Retrospected have you found the correct mDNS name?

Retrospected, 2020-07-26:
I have just ran it, strangely enough when running it with "homey app run", discovery works just fine.. 👍 
However now i end up with " Only HTTP(S) protocols are supported" when looking at the device status.

this is the console output:

```
2020-07-26 22:53:09 [log] [ManagerDrivers] [enphase-envoy] {
  '122015061668': DiscoveryResultMDNSSD {
    _events: [Object: null prototype] {},
    _eventsCount: 0,
    _maxListeners: undefined,
    address: '192.168.1.50',
    host: 'envoy',
    port: 80,
    name: 'envoy',
    fullname: 'envoy._enphase-envoy._tcp.local.',
    txt: { serialnum: '122015XXXXXX', protovers: '4.10.35', txtvers: '1' },
    id: '122015XXXXXX',
    lastSeen: 2020-07-26T20:52:57.698Z,
    [Symbol(kCapture)]: false
  }
}
2020-07-26 22:53:11 [log] [ManagerDrivers] [enphase-envoy] [0] Initializing device
2020-07-26 22:53:11 [log] [ManagerDrivers] [enphase-envoy] [0] Added device
2020-07-26 22:53:11 [log] [ManagerDrivers] [enphase-envoy] [0] Checking production
2020-07-26 22:53:11 [log] [ManagerDrivers] [enphase-envoy] [0] Tried fetching production before API init or device unavailable!
2020-07-26 22:53:11 [log] [ManagerDrivers] [enphase-envoy] [0] Discovery available
2020-07-26 22:53:11 [err] [ManagerDrivers] [enphase-envoy] Error executing Device.onDiscoveryResult TypeError: Only HTTP(S) protocols are supported
    at getNodeRequestOptions (/node_modules/node-fetch/lib/index.js:1303:9)
    at /node_modules/node-fetch/lib/index.js:1404:19
    at new Promise (<anonymous>)
    at fetch (/node_modules/node-fetch/lib/index.js:1401:9)
    at EnphaseEnvoyApi.apiRequest (/drivers/enphase-envoy/api.js:9:35)
    at EnphaseEnvoyApi.getProductionData (/drivers/enphase-envoy/api.js:20:27)
    at EnphaseEnvoy.onDiscoveryAvailable (/drivers/enphase-envoy/device.js:18:31)
    at /opt/homey-client/system/manager/ManagerApps/bootstrap/sdk/v2/lib/Driver.js:1:4290
    at processTicksAndRejections (internal/process/task_queues.js:97:5)
2020-07-26 22:53:11 [log] [ManagerDrivers] [enphase-envoy] [0] Cron job f8c4f77223ad45b2b1649f8cb311a145 created successfully
2020-07-26 22:53:12 [log] [ManagerDrivers] [enphase-envoy] [0] Initialized cron job f8c4f77223ad45b2b1649f8cb311a145
2020-07-26 22:53:15 [log] [ManagerDrivers] [enphase-envoy] [0] Running task f8c4f77223ad45b2b1649f8cb311a145
2020-07-26 22:53:15 [log] [ManagerDrivers] [enphase-envoy] [0] Checking production
2020-07-26 22:53:15 [log] [ManagerDrivers] [enphase-envoy] [0] Tried fetching production before API init or device unavailable!```

DiedB, 2020-07-26:
Interesting! I changed the mDNS name to `enphase-envoy` in the branch I sent you and apparently that is the correct one (it was `envoy` in the app store version).

An updated version has been pushed to the app store (test channel), that should also fix the other bug you encountered and implements the `activeCount` check.

Retrospected, 2020-07-26:
> Interesting! I changed the mDNS name to `enphase-envoy` in the branch I sent you and apparently that is the correct one (it was `envoy` in the app store version).
>
> An updated version has been pushed [...]

Cool thx for picking it up so fast, I'm getting a step further now. Now i get the following error:

Error retrieving data (ReferenceError: currentEnergy is not defined)

DiedB, 2020-07-26:
Thanks for the quick feedback! Please try the next build (4.1.2).

Retrospected, 2020-07-26:
> 
> 
> Thanks for the quick feedback! Please try the next build (4.1.2).

same thing

DiedB, 2020-07-26:
That is strange, could you run the app manually and see if you get more error output?

Retrospected, 2020-07-26:
> 
> 
> That is strange, could you run the app manually and see if you get more error output?

```
2020-07-26 23:45:36 [log] [ManagerDrivers] [enphase-envoy] [0] Cron job 5a6df3ce267f49eea5b271c8d60b27d5 created successfully
2020-07-26 23:45:36 [log] [ManagerDrivers] [enphase-envoy] [0] Initialized cron job 5a6df3ce267f49eea5b271c8d60b27d5
2020-07-26 23:45:40 [log] [ManagerDrivers] [enphase-envoy] [0] Running task 5a6df3ce267f49eea5b271c8d60b27d5
2020-07-26 23:45:40 [log] [ManagerDrivers] [enphase-envoy] [0] Checking production
2020-07-26 23:45:40 [log] [ManagerDrivers] [enphase-envoy] [0] Unavailable (ReferenceError: currentEnergy is not defined)
2020-07-26 23:45:45 [log] [ManagerDrivers] [enphase-envoy] [0] Running task 5a6df3ce267f49eea5b271c8d60b27d5
2020-07-26 23:45:45 [log] [ManagerDrivers] [enphase-envoy] [0] Checking production
2020-07-26 23:45:45 [log] [ManagerDrivers] [enphase-envoy] [0] Tried fetching production before API init or device unavailable!
2020-07-26 23:45:50 [log] [ManagerDrivers] [enphase-envoy] [0] Running task 5a6df3ce267f49eea5b271c8d60b27d5
2020-07-26 23:45:50 [log] [ManagerDrivers] [enphase-envoy] [0] Checking production
2020-07-26 23:45:50 [log] [ManagerDrivers] [enphase-envoy] [0] Tried fetching production before API init or device unavailable!
2020-07-26 23:45:55 [log] [ManagerDrivers] [enphase-envoy] [0] Running task 5a6df3ce267f49eea5b271c8d60b27d5
2020-07-26 23:45:55 [log] [ManagerDrivers] [enphase-envoy] [0] Checking production
2020-07-26 23:45:55 [log] [ManagerDrivers] [enphase-envoy] [0] Tried fetching production before API init or device unavailable!
```

DiedB, 2020-07-26:
Please try 4.1.3 :wink:

geurti64, 2020-07-26:
Is it possible it doesn't work now because of sleep modus of Envoy?
My SMA doesn't do anything after sunset...

Retrospected, 2020-07-26:
> 
> 
> Please try 4.1.3 😉

alright no more errors :) Energy shows 0 Watt now though.. so let's see what it does in the morning

Energymeter for me shows a "-".. i would expect it to show around 14.00kWh that i have according to the cloud version

DiedB, 2020-07-26:
@Retrospected that is (still) an issue with unmetered setups and the local version. As you can see in your JSON output, the local version does not contain a `whToday` field. Will probably have to record a value from `whLifetime` at the start of each day and then subtract it from the latest value, but haven't implemented that yet.

DiedB, 2020-07-26:
@geurti64 That is possible, but do you get errors with version 4.1.3?

Retrospected, 2020-07-26:
> @Retrospected that is (still) an issue with unmetered setups and the local version. As you can see in your JSON output, the local version does not contain a `whToday` field. Will probably have to [...]

ah of course.. makes sense. I have already fixed it the way you describe using the cloud data anyway :)
Thanks for the feature! 

Maybe @KvDaalen can test how it works on a metered Envoy.

geurti64, 2020-07-26:
I get device is not available.
Error retrieving data.
TypeError: cannot read property activeCount of undefined.

DiedB, 2020-07-26:
@geurti64 Can you try to access http://envoy.local/production.json?details=1

If that works, can you send the output it gives?

geurti64, 2020-07-26:
> @geurti64 Can you try to access http://envoy.local/production.json?details=1
> 
> If that works, can you send the output it gives?

This is what I got:

{"production":[{"type":"inverters","activeCount":6,"readingTime":0,"wNow":0,"whLifetime":5106119}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}

DiedB, 2020-07-26:
I made a new build that should work with all the three JSON outputs in this thread, see 4.1.5. I wonder what other variations we are going to encounter :wink:

@Retrospected The energy value might work now.

geurti64, 2020-07-26:
> I made a new build that should work with all the three JSON outputs in this thread, see 4.1.5. I wonder what other variations we are going to encounter 😉
> 
> @Retrospected The energy value might work now.

I get the same error.
{"production":[{"type":"inverters","activeCount":6,"readingTime":0,"wNow":0,"whLifetime":5106119}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}

geurti64, 2020-07-26:
[image]

DiedB, 2020-07-26:
@geurti64 Sure you are running 4.1.5? Can you remove the app and reinstall? Do you get the exact same error?

geurti64, 2020-07-26:
Oops 😬
[image]

geurti64, 2020-07-26:
I'll try again after sunrise 😉

DiedB, 2020-07-26:
That error won't be fixed after sunrise, but it will in 4.1.6 :smile:

geurti64, 2020-07-26:
> That error won't be fixed after sunrise, but it will in 4.1.6 😄

I know, there was an other message behind it...
Give me a second, then I try once again before sleeping.

geurti64, 2020-07-26:
Seems like it's good now. You're a master!
[image]

geurti64, 2020-07-26:
Good night.

geurti64, 2020-07-27:
Seems like the "energiemeter" is not working correctly as the meter in my other tile says 0.18 kWh.

{"production":[{"type":"inverters","activeCount":6,"readingTime":1595833466,"wNow":134,"whLifetime":5106337}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}

[image]

geurti64, 2020-07-27:
An other question.
Is it also possible to make an option that we choose one panel?
My solar panel set is East/West. I would like to know the Energie on East side to control my sunscreens.
Maybe just make a new topic for this question?

Retrospected, 2020-07-27:
Hmm on 4.1.7 it doesn’t seem to be working on my unmetered envoy anymore

[image]

Retrospected, 2020-07-27:
> Hmm on 4.1.7 it doesn’t seem to be working on my unmetered envoy anymore
> 
> [image]

Nevermind, re-adding the device fixed it

DiedB, 2020-07-27:
@geurti64 @Retrospected I pushed a new version (4.1.8) that might fix the energy meter staying at zero. If it doesn't work, please remove the inverter and reinstall the app.

@geurti64 That's not possible from the data we have, but maybe on a different endpoint than the `production.json` one? You will have to do some research.

geurti64, 2020-07-27:
The result is the same.
To be honest, i removed the inverter and installed the last version over 4.1.7 Is it really necessary to remove app 4.1.7 first?

geurti64, 2020-07-27:
> @geurti64 That's not possible from the data we have, but maybe on a different endpoint than the `production.json` one? You will have to do some research.

All research I already did is asking around and found this.
Maybe someone else can do something with this information.

https://m.facebook.com/story/graphql_permalink/?graphql_id=UzpfSTc0MjIwODQ4MTpWSzoxOTEzMTY0NDEyMTUwMDI3

The link is from the Enphase users + owners group.

DiedB, 2020-07-27:
@geurti64 Can you check whether http://envoy.local/api/v1/production/inverters gives any output?

geurti64, 2020-07-27:
> @geurti64 Can you check whether http://envoy.local/api/v1/production/inverters gives any output?

I've to find out how to login. The user name and password is different from what I'm used to.

Retrospected, 2020-07-27:
> > @geurti64 Can you check whether http://envoy.local/api/v1/production/inverters gives any output?
> 
> I've to find out how to login. The user name and password is different from what I'm used to.

Read here:https://thecomputerperson.wordpress.com/2016/08/28/reverse-engineering-the-enphase-installer-toolkit/

Install the apk that is listed (on your own risk ;-)). It will retrieve the “installer” credentials to access that api. Worked fine for me

JSON:

‘’’[
  {
    "serialNumber": "122018001310",
    "lastReportDate": 1595840610,
    "devType": 1,
    "lastReportWatts": 80,
    "maxReportWatts": 308
  },
  {
    "serialNumber": "122017037787",
    "lastReportDate": 1595840615,
    "devType": 1,
    "lastReportWatts": 78,
    "maxReportWatts": 306
  },
  {
    "serialNumber": "122017037789",
    "lastReportDate": 1595840606,
    "devType": 1,
    "lastReportWatts": 90,
    "maxReportWatts": 296
  },
  {
    "serialNumber": "122018000241",
    "lastReportDate": 1595840612,
    "devType": 1,
    "lastReportWatts": 90,
    "maxReportWatts": 296
  },
  {
    "serialNumber": "122017037914",
    "lastReportDate": 1595840618,
    "devType": 1,
    "lastReportWatts": 87,
    "maxReportWatts": 296
  },
  {
    "serialNumber": "122017037772",
    "lastReportDate": 1595840620,
    "devType": 1,
    "lastReportWatts": 85,
    "maxReportWatts": 296
  },
  {
    "serialNumber": "122018000891",
    "lastReportDate": 1595840621,
    "devType": 1,
    "lastReportWatts": 89,
    "maxReportWatts": 296
  },
  {
    "serialNumber": "122018001281",
    "lastReportDate": 1595840607,
    "devType": 1,
    "lastReportWatts": 79,
    "maxReportWatts": 309
  },
  {
    "serialNumber": "122018000625",
    "lastReportDate": 1595840614,
    "devType": 1,
    "lastReportWatts": 91,
    "maxReportWatts": 296
  },
  {
    "serialNumber": "122017037320",
    "lastReportDate": 1595840611,
    "devType": 1,
    "lastReportWatts": 78,
    "maxReportWatts": 305
  },
  {
    "serialNumber": "122018001358",
    "lastReportDate": 1595840616,
    "devType": 1,
    "lastReportWatts": 92,
    "maxReportWatts": 296
  }
]
’’’

Other note: the energymeter rounds the production of the current day to a whole number, while the cloud version shows it with 2 decimals. Is this based on the api or did you include a round function?

DiedB, 2020-07-27:
The round is not necessary indeed, I pushed a new build to fix that.

geurti64, 2020-07-27:
> Read here:https://thecomputerperson.wordpress.com/2016/08/28/reverse-engineering-the-enphase-installer-toolkit/
>
> Install the apk that is listed (on your own risk ;-)). It will retrieve the [...]

I got stuck on "911 is an inside job" 🙈
The app is Android only right?

@DiedB 
It's because of this shitty summer that it didn't work.
Normally at this time it should be far above one.
Good job again, it works now.
[image]

[18 more replies]

closed 2020-09-18 by DiedB