DiedB/Homey-SolarPanels #112: GoodWe Solar via Zonnepanelen app Homey

TomCat1971, 2020-07-29:
Hi ,

I try to connect my Goodwe converter GW-2000 XS via zonnepanelen app on my Homey.  
Installing is working but when I want to login with the right Creditials it doesnt work .  Error :Wrong credtials  
I am pretty sure the creditials are oke.

Something wrong with zonnepanelen app? Can someone help me
Version V4.0.13 Diederik Bakker

Grz Tom

DiedB, 2020-07-29:
Are you able to login at [GoodWe SEMS](https://www.semsportal.com/home/login)?

TomCat1971, 2020-07-29:
Yep but thats via the cloud, do I need this creditials to login to the app?

DiedB, 2020-07-29:
Yes, what other credentials do you have?

TomCat1971, 2020-07-29:
Credentials from the webinterface of the converter.
I thought that it was directly connected to inverter. Put its getting data from sems portal. Thank you for your answer

closed 2020-07-29 by DiedB