DiedB/Homey-SolarPanels #113: issue: since Homey core 5 beta no contacts with Solaredge

mveck, 2020-07-31:
Since introduction of Homey firmware 5.0 Beta no contacts with Solaredge. 
Please check (during pairing "no keypair found").

DiedB, 2020-07-31:
When do you get the error, after entering your API credentials? Do you have a screenshot?

mveck, 2020-08-01:
Hi Diederik, I did firstly get the problem when upgrading to 5.0 release candidates... the existing device (Solaredge) wasn't available anymore.

Then I decided to re-pair my device (and checking if that would help). During pairing of the device I do get the following message (see screenshot), ID and key are blurred to avoid problems.
[image]

mveck, 2020-08-01:
After that I tried to pair to PVoutput (which is also reading my Solaredge, as a backup), and this one gives similar results... It looks like an equal problem as happened when Homey 2.0 was introduced... Formally only Zibee would be touched, but in reality also SDK 1.0 code is obsoleted.

DiedB, 2020-08-01:
Interesting, sounds familiar indeed...

I'm too dependent on some Zigbee apps to try the new firmware out myself, currently. I have a few more changes to make to the app in its current state. I will release that as the latest version for SDK v2, and then look at rewriting against SDK v3. The transition will take some time but will be made eventually.

mveck, 2020-08-01:
Hi Diederik, Don't worry, we are all in it (just finalized some work on Bitron Video) ;-)

mveck, 2020-09-13:
Hi Diederik,
Quite late but please check this diagnostic report:
535e4579-b8f5-4c33-b93a-b0a4db46e2cd

Made in combination with 5.0.0. rc 31

Cheers, Menno

DiedB, 2020-09-13:
Thanks, unfortunately the error report does not contain any further information (just a log message that the app has started). Can you describe what's happening for you?

mveck, 2020-09-13:
Hi Diederik,
That's a pitty, the log is belonging to the screenshot in my earlier
message. You requested a log actually months ago, and announced that you
will need some more time before solving the problem for the 5.0.0
firmware...

DiedB, 2021-03-07:
Does it now work for you? Planning to take the app to SDK v3 but that will take some more time. Limited testing with the stable version of 5.0.0 shows that it works for me.

mveck, 2021-03-07:
Hi Diederik, Solaredge is working fine now. It really looks like a temporary disorder in the kernel, but now its working fine for me.  Using 4.3.4 and Solaredge. Homey kernel 5.0.1 rc1

DiedB, 2021-03-07:
Yeah, there is a reason it's called RC ;-) Will keep this issue open until SDK v3 is implemented.

mveck, 2021-03-08:
Hi Diederik, I would promote to close this, since the temporary kernel mishap was solved in one of het 64 RC's of R5.0.0! I need to run that kernel due to my development activities. Now Solaredge is working fine on my side, and (my catch) is that you don't have to move to SDK 3.0 on your side (unless you are bound to Zigbee).

DiedB, 2021-03-08:
In the end I have to, SDK v2 will ultimately be end of life. So while your issue has been resolved, this thread serves as a reminder that the app has to be upgraded some time in the future.

DiedB, 2021-10-31:
Hey - I am working on an SDK v3 implementation. Current test version supports SolarEdge, so feel free to try and report back.

https://homey.app/a/it.diederik.solar/test/

closed 2023-06-10 by DiedB