DiedB/Homey-SolarPanels #114: API delay

nickmvanslooten, 2020-08-03:
Hi Diederik,

First of all: I like your app. Keep up the good work. 👍

One thing I've noticed with my SolarEdge connection, is that the data has a pretty large delay. While watching the data in the SolarEdge app itself, there's sometimes a delay of 10 minutes. But when I look at the data in Homey on that moment, it's even older.

I use this data to open / close my sunscreens. So when it's getting cloudy, nothing happens. When the sun returns in about half an hour, the sunscreens suddenly open, while they need to be closed on that moment. So that flow isn't working for me now.
Is there anything you can do about this, so the delay will be the same as the delay in the SolarEdge app itself?
If you want to help me, and need something to test with, just let me know.

DiedB, 2020-08-03:
Could you try the beta version? Some things were changed in the timings and it might help.

https://homey.app/nl-nl/app/it.diederik.solar/Zonnepanelen/test/

Although I'm afraid that the SolarEdge API is not going to give you the resolution you will need for your sunscreens. 10 minutes is still too much of a delay. Maybe use a different power metering solution at your panel, like the Qubino DIN power module or a YouLess with a cheap power meter (I used an ABB C11 previously). Those solutions will give you power values to the second.

nickmvanslooten, 2020-08-04:
Thanks for your fast response. I have installed the beta and will share my experiences.

A 10-minute-delay might still be too large for this flow, If that's the case, I might just go for a weather station or something, but I'll keep the DIN-modules in mind. 😊

nickmvanslooten, 2020-08-04:
The good thing: the delay is gone.
The bad thing: it only shows a 500 error.

I've created a report, which has this code:
fd3b9d4e-997c-4942-9e49-8eb51b25ef2a

DiedB, 2020-08-12:
It seems to work for a while, and then starts giving 500 errors. Does it recover at a later stage?

nickmvanslooten, 2020-08-21:
Just re-installed the test version, because I didn't want my solar-related-flows to stop working while having a tropical heat outside.
First impression: it seems to work, I'll keep you posted. 🙂

DiedB, 2020-08-24:
I don't think it's going to get any better. SolarEdge also disabled direct access to inverters in a new firmware, making a local connection impossible for most inverters.

closed 2020-08-24 by DiedB