DiedB/Homey-SolarPanels #119: Error for Trannergy

BertWi, 2020-09-07:
Hi,

I'm getting this error while setting it up:

"You've been logged out or you have no authorization to use the service".

What does this mean?
And how can it be solved?

Best regards
Bert from NL

DiedB, 2020-09-08:
Sounds like some issue with credentials. Are you sure you have the right credentials, do they work at https://log.trannergy.com/LoginPage.aspx?

BertWi, 2020-09-09:
Hi,
for my energie readings I need to login here: 

http://log1.trannergy.com/login.html

So that seems to be a different login location.

BertWi, 2020-10-04:
Hi Diederik
Any plan on solving this? 
No push, just curious. 
Thanks!

DiedB, 2020-10-05:
Yes, but I would need more technical information about their (new) API. It seems to still work for most users, as I have not yet seen heard of other users with problems. Once more people chime in, we might be able to resolve this, but it depends on support from Trannergy as well.

enbach, 2020-12-14:
I recently got my Trannergy system installed. Getting the same error code

Cootjec, 2020-12-18:
I recently got my Trannergy system installed as well,  got the same error when I try to connect with the homey app

[ERR_UNHANDLED_ERROR]:
Unhandled error. ("You've been logged out or you have no authorization to use the service".)

Please let me know where I can help you

DiedB, 2021-01-09:
@MaxvandeLaar could you take a look?

enbach, 2021-02-11:
@DiedB Do you have an update on this issue ? Or is there any way I could help/assist you ?

MaxvandeLaar, 2021-02-11:
I have not yet have a change to look at it, however, I can tell you that I am not experiencing any issues. At the same time, I am running my own custom version of this app before Trannergy got merged back in.

enbach, 2021-02-12:
@MaxvandeLaar Are you using the http://log1.trannery.com page ? Seems like that is the main issue for me and BertWi

BertWi, 2021-02-12:
Yes. Same error and that seems to be the issue.

MaxvandeLaar, 2021-02-12:
No I am using the aspx page. I can't register a logger on the page you use so I can't test it.

enbach, 2021-02-14:
@MaxvandeLaar, I could give you my login credentials, if that could be of use.

MaxvandeLaar, 2021-02-14:
@enbach no need, I managed to register my logger. I cannot promise you I can commit any time to this, but I will try to have a look.

BertWi, 2021-02-14:
Hi
As an alternative I bought a HomeWizard WiFi Energy Monitor (Euro 39,95) that you just plug in to your smart meter at home and now using the HomeWizard Energy App on my Homey. This now allows me to read all energy I produced and used for T1 and T2 and even per phase.  As a bonus can also read my gas usage.

Harinozo, 2021-02-18:
I had the same problem but I have a solution for people who log in to the new monitoring portal and can not use the app. You have to register on the 'old' monitoring portal. https://log.trannergy.com/. When registering, use the logger s/n that can be found in your 'new' portal under devices. After a while the data will arrive and you can use this app in Homey.

enbach, 2021-02-18:
@Harinozo Thanks for the info, I've just registered on the old monitoring portal. Hopefully this is a quick fix.

hanshoevenaars, 2021-11-01:
I am sorry to say the old monitoring portal is not working anymore. I have the same problem. I can not log in (Error: could not find that PairSession) is the message I get.

Sorry guys

closed 2023-06-10 by DiedB