DiedB/Homey-SolarPanels #123: Zonnepanelen App does not work anymore for me (SolarEdge) in Homey

SolarEdgeUser, 2020-10-12:
Hello, 

I have been using an Athom Homey (normal, not Pro) with firmware version 4.1.0 and NodeVersion 8.16.0. I have installed the Zonnepanelen app. I am using a SolarEdge omvormer. 

This worked great for about the first 2 weeks. Then it stopped working. The whole Zonnepanelen app crashes, and gives me the notification:  The app ZOnnepanelen is gestopt met werken. De ontwikkelaar is op de hoogte gesteld. 

If I go to the app in the app list, it says that the app shut down unexpectedly, and it gives the following information: 

[image]

It would be nice if this issue could be fixed :) 

Thanks. 

Regards, 
SolarEdgeUser

DiedB, 2020-10-12:
This is an issue with the newest update. I used a language feature that is not supported on Node versions that were used before Homey version v4.2.0. I didn't know and therefore didn't set the compatibility settings. This is causing crashes for users that haven't updated their Homey to the latest version.

Can you check which firmware your Homey is using, and update to the latest version (do not install experimental updates)? The Solar Panels app should then continue working normally.

SolarEdgeUser, 2020-10-12:
I am running 4.1.0. I have just now started the update process to update to 4.2.0 Once finished, I will let you know if it has been solved. Will be back in a few minutes....hopefully.

SolarEdgeUser, 2020-10-12:
That did the trick. Now that I am on 4.2.0 it is working again. Thanks! :)

closed 2020-10-12 by DiedB