DiedB/Homey-SolarPanels #128: Solar panel generated power is added up to total usage in energy tab

hendriksenremco, 2020-11-04:
The power which is generated by the solar-panels is added up to the total power-usage (see biggest number in the circle).
My total usage is 325w according to p1mon and my solar-panel generates 117w in this screenshot, which makes exactly 442w. 
Those numbers shouldn't be added up in my opinion.

[image]

DiedB, 2020-11-04:
I'm a bit puzzled where the figure comes from. It is in the Homey software so I cannot really influence it.

The Solar Panels app adds devices with the class 'Solar Panel', adding them to the outer ring in Homey Energy. I believe Homey does not make calculations based on that figure, and instead shows consumption separately based on what it knows from other devices (the number you are mentioning). I do not understand why that is not the case in your screenshot. Another screenshot from another issue in this repository, showing different behaviour:
https://user-images.githubusercontent.com/32201989/93337187-14879980-f829-11ea-8b6f-03883669361c.png

Using the information from the page below, can you explain the behaviour of Homey Energy for your Homey?
https://apps-sdk-v2.developer.athom.com/tutorial-Drivers-Energy.html

BillyHekken, 2020-11-14:
Hi. I have the same issue as @hendriksenremco. In my case also the power which is generated by the solar panels is added up to the total power-usage. Of course I use Zonnepanelen, but the Plugewise app on Homey for the data from the P1 port of my smart meter.

There is one difference in your screenshot to the situaton of hendriksenremco and my Homey energy flow view. In your screenshot (https://user-images.githubusercontent.com/32201989/93337187-14879980-f829-11ea-8b6f-03883669361c.png) the P1 reading shows as a negative value. As I see it the screenshot seems not to show your P1 value as total consumption: might you have set the view to exclude the energy consumption from the P1 output (throught the settings in the energy flow? I have set to include the P1 power consumption value in the total power consumption as I, by far, don't have my major energy consuming devices enabled with power meters.

Happy to share more details if needed.

hendriksenremco, 2020-11-20:
@DiedB Thanks for responding. I think indeed that this is a bug in Homey energy and not in your app. Keep up the good work and i will pass the issue through to Athom.

closed 2020-11-20 by hendriksenremco