DiedB/Homey-SolarPanels #129: Login Omnik Portal does not work

geurtd, 2020-11-05:
Er is een onbekende fout opgetreden
An unknown error occured followed by chinese characters.

You can only click Ok and this comes back tot the login page

DiedB, 2020-11-05:
They took their API down. I have no information (yet) on a new API, does someone want to reverse engineer the app again?

geurtd, 2020-11-06:
Thats a Shame,

OllemGit, 2021-03-03:
(Most?) OMNIK inverters can also be accessed locally: 
http://\<local ip address OMNIK inverter\>/js/status.js

for this url no userid or password is needed, just IP Address

More details: https://github.com/robbinjanssen/home-assistant-omnik-inverter

The output is a very long string with the required production data at fixed position, which is relatively easy to parse.
However, according to the code of robbinjanssen - see link above - different type of output can be expected:
```

# Find the webData
        if result.find('webData="') != -1:
            matches = re.search(r'(?<=webData=").*?(?=";)', result)
        else:
            matches = re.search(r'(?<=myDeviceArray\[0\]=").*?(?=";)', result)

        # Split the values
        if matches is not None:
            data = matches.group(0).split(',')
            self.result = [
                data[0],
                int(data[5]),
                int(data[6]),
                int(data[7])
            ]

```
My OMNIK inverter shows the 'webData' output:

```
var version="H4.01.XXXX.0.XXXX.0.08";var m2mMid="1612345678";var wlanMac="AA:11:AA:BB:AA:FF";var m2mRssi="0%";var wanIp="192.168.2.xxx";var nmac="AA11AABBAAFF";var fephy="off";var webData="NLEN1234567DK012,NL2-V1.0-12345,V5.3-80169,omnik3000tls,3000,1176,201,40583,,2,";function initPageText(){
           var A=window.parent.reList("status");
           ........... a lot of output which can be ignored...........................
};
```

a few hours later:

```
var version="H4.01.XXXX.0.XXXX.0.08";var m2mMid="1612345678";var wlanMac="AA:11:AA:BB:AA:FF";var m2mRssi="0%";var wanIp="192.168.2.xxx";var nmac="AA11AABBAAFF";var fephy="off";var webData="NLEN1234567DK012,NL2-V1.0-12345,V5.3-80169,omnik3000tls,3000,417,646,40627,,4,";function initPageText(){
var A=window.parent.reList("status");
           .........................................................
};
```
This part is the data part:
3000,**419,657,40628**,,1,

419 -> current power output / 'vermogen' (W) => 419 W
657 -> energy production today / 'energie'  (kWh) => 6.57 kWh  ( $value / 100 kWh )
40627 - total energy production (kWh) => 4062.7 kWh ($value / 10 kWh)

DiedB, 2021-03-03:
And `http://[omnik-ip]/status.json?CMD=inv_query&rand=38219389`?

OllemGit, 2021-03-03:
Result (after login pop-up for uid-pwd):

```
404 Not Found
File not found. 
```

uid-pwd are okay because IP address without any parameter/.js/.json in URL results in status page after login

DiedB, 2021-03-04:
Ah, I see.

I also saw a Home Assistant implementation where the inverter was approached over a raw TCP connection on port 8899. Do you have Netcat or Telnet to try and see what happens on that port?

OllemGit, 2021-03-04:
unfortunately no Telnet not around somewhere for now, but Netcat (not familiar with it) did report a port listing:

`root@abc:~# netcat -v -n  192.168.2.xyz 1000-9000`
`(UNKNOWN) [192.168.2.xyz] 8899 (?) open`

OllemGit, 2021-03-04:
`root@abc:~# netcat -l 192.168.2.xyz 8899`

Generates no output and nothing happens, but might be due to being dark outside :-) 
Will retry tomorrow

DiedB, 2021-03-04:
Without the `-l` it might generate some output. I can also look at the implementation you linked.

DiedB, 2021-03-04:
Can someone port forward port 8899 from their Omnik inverter and send me the IP and WiFi serial number? I am willing to create a PoC for local connection to the Omnik inverter.

DiedB, 2021-03-06:
[image]

Just implemented it, seems to work well. Will monitor it and check whether it keeps working during the night, when the inverter powers down.

DiedB, 2021-03-07:
Released it to https://homey.app/a/it.diederik.solar/test/, wrote some documentation at https://app.gitbook.com/@diederik/s/homey-solar-panels-app/inverters/omnik. Please try it and let me know if this works!

KimKodde, 2021-03-08:
Thanks for the better solution. I can connect to it now and gives me three values:
Energie (4.1) = ok
Vermogen (132667) = not ok, hmm  should now be around 1600 watt
Voltage = 4449.43 = not ok, don't need this value.
So energy is ok but Power is not correct

part of status.js output
var version = "H4.01.38Y1.0.08W1.0.07"; var m2mMid = "XXXXXXX"; var wlanMac = "XX: XX: XX: XX: XX: XX"; var m2mRssi = "70%"; var wanIp = "192.168.0.XX"; var nmac = "XXXXXXX"; var fephy = "off"; var webData = "NLDN302015299999, NL1-V1.0-0108-4, V2.0-0028, omnik3000tl, 3000,1508,359, 160359 ,, 3, ";

DiedB, 2021-03-08:
Interesting, seems like some offsets are incorrect when parsing the response from your inverter. What are the first three digits of your WiFi S/N?

Guess I will have to add some more logging to the app, will request you to submit a diagnostics report afterwards.

KimKodde, 2021-03-08:
S/N:160689....

DiedB, 2021-03-08:
Could you update to the latest version from the test link above (v4.4.4), wait a few minutes and send a diagnostic report from the app settings page, with an approximation of the current power value in the attached message?

KimKodde, 2021-03-08:
cc2ae772-b5b4-4452-84a5-5f8e5f3ff91b and around 400 watt
App was automatic updated

DiedB, 2021-03-08:
> cc2ae772-b5b4-4452-84a5-5f8e5f3ff91b and around 400 watt
> App was automatic updated

Thanks a lot, has been fixed! Your inverter is slightly different from that of @OllemGit, returning `ff` for fields that contain no data instead of `00`, hence the app breaking. The new implementation is fine with both.

KimKodde, 2021-03-08:
Tested and works! Even the voltage now shows 233.3 volt. Thanks for the quick fix

RalfGroot, 2021-03-11:
@DiedB Is there already a solution for my Omnik. Could it help to give you access to mine on port 8899?
https://community.athom.com/t/app-solar-panels/44273/12

DiedB, 2021-03-20:
@RalfGroot Sure, I will have a look. Just send me the IP address in private, on the Homey Community Forum for example.

RalfGroot, 2021-03-22:
> @RalfGroot Sure, I will have a look. Just send me the IP address in private, on the Homey Community Forum for example.

Send you a pm on https://community.athom.com/

RalfGroot, 2021-03-28:
> > @RalfGroot Sure, I will have a look. Just send me the IP address in private, on the Homey Community Forum for example.
> 
> Send you a pm on https://community.athom.com/

@DiedB did you had time to look at it allready? When you need more info, let me know!

RalfGroot, 2021-04-04:
Nice to hear that data will be transfered to the SOLARMAN platform. We have to wait till we get an email, after the data have been migrated. 
- 1 month free trail
-  after the free month 6 EUR/year

@DiedB issue can be closed now?

[https://omnik1.solarmanpv.com/](url)
Dear Users,

As Omnik has gone bankrupt and is not able to provide service in the future, we are sorry to inform you that this platform will be closed at the end of April.

With the purpose of providing continuous service for you, all data at Omnik platform will be migrated to SOLARMAN platform from 9 a.m. on 19th April, 2021 (UTC+8). Please notice that any operation on Omnik platform from that time will not be synchronized to SOLARMAN platform.

The overall migration process will take about 72 hours. In order to ensure the data integrity and accuracy, please DO NOT modify any data on Omnik platform during this period. If you need to register new accounts during the migration, please log in to SOLARMAN platform (For end-users, please log in https://home.solarmanpv.com; for installers and distributors, please log in https://pro.solarmanpv.com).

After the migration is completed, you will be informed at the very first time. Notice: Services including Open API, APP and data forwarding at Omnik platform will not be available after migration.

You could log in SOLARMAN platform using your current account and password. SOLARMAN will provide one-month free trial for your Omnik devices. During this period, you will be able to use all the functions at the new platform. When the free trial period is expired, you will need to pay 6 EUR/year for each device, otherwise the access to the device and data will be restricted.

SOLARMAN Smart (https://home.solarmanpv.com) is designed for end-users. With new visual experience, user-friendly data display, and well-established monitoring function, SOLARMAN Smart makes smart energy management easier for everyone.

SOLARMAN Business (https://pro.solarmanpv.com) is designed for installers and distributors, which is more suitable for distributed system management. With comprehensive and sophisticated functions, SOLARMAN Business will provide global leading intelligent O&M solutions for installers and distributors.

Sorry for the inconvenience and we are looking forward to providing better service for you at SOLARMAN platform.

Please feel free to contact us for any questions via customerservice@solarmanpv.com._

closed 2023-06-10 by DiedB