DiedB/Homey-SolarPanels #131: Request for SolarEdge solar converter app for Homey

GerardLeemkuil, 2020-11-07:
Diederik is it possible to add in the Solarcels app to add the object for the SolarEdge converter.

DiedB, 2020-11-07:
What do you mean with 'object'?

GerardLeemkuil, 2020-11-08:
Object = device in Homey

DiedB, 2020-11-08:
That's exactly what the app already does right?

Voel je vrij om het in het Nederlands uit te leggen, als dat makkelijker is.

GerardLeemkuil, 2020-11-08:
Oké zou je de omvormer van solidEdge willen toe voegen aan jouw zonnepanelen app
Is dat mogelijk

DiedB, 2020-11-08:
De omvormer van SolarEdge zit in de Zonnepanelen-app, toch? Ik snap niet precies wat je bedoelt.

GerardLeemkuil, 2020-11-08:
Daar zit een fout in. 
De omvormer  6 outputs zichtbaar in Homey Insights scherm
Energie data deze is ok
Energiemeter data deze is ok
Stroomverbruik (null geen data)
Stroomverbruik wel data deze is ok
Stroomverbruik vandaag geen data
Temperatuur omvormer data deze is ok

GerardLeemkuil, 2020-11-08:
Wanneer ik ga naar het device scherm dan zie ik 5 variabelen
Energie is ok
Stroomverbruik (null) geen Gata
Energiemeter is ok
Stroomverbruik vandaag (null) geen data
Temperatuur omvormer is ok
Hopelijk heb je hier iets aan om de fout te vinden.

GerardLeemkuil, 2020-11-08:
[image]
Outputs scherm Inverter

GerardLeemkuil, 2020-11-08:
[image]

GerardLeemkuil, 2020-11-08:
[image]

DiedB, 2020-11-09:
Het is niet echt een fout, je SolarEdge apparaat geeft die waarden gewoon niet terug.

Kun je https://homey.app/a/it.diederik.solar/test/ installeren? Als je daarna je omvormer verwijdert en opnieuw toevoegt zouden de ongebruikte capabilities moeten zijn verdwenen.

GerardLeemkuil, 2020-11-09:
Prima , ik ga het installeren
Bedankt voor je inzet

Gerard Leemkuil
gleemkuil@ziggo.nl
0637276710

GerardLeemkuil, 2020-11-09:
Device scherm is nog steeds het zelfde
Op HomeyInsights scherm werken de eerste 3 output en de temperatuur
De tweede Stroomverbruik heeft geen data evenals de Stroomverbruik vandaag heeft geen data

closed 2020-11-09 by DiedB