DiedB/Homey-SolarPanels #133: Enphase

Barblo1986, 2020-11-19:
Dear,

In the Enphase module, the summary of the total energy used today is displayed incorrectly. The comma or unit is wrong. According to the example, 4911.6 kWh has been consumed, while in reality this is 4.9 kWh. This is also the case with the total yield of the day. The screenshot shows that 958.84 kWh has been used while in reality this is only 0.96 kWh. The word "energiemeter" should be better adapted to "Total generated power" and the word "energie" to generating power.
[image]

DiedB, 2020-11-19:
Interesting. What Enphase product are you using? I think it's something other than most users, otherwise this discrepancy would have been noticed before.

I do not have control over the strings 'energie' and 'energiemeter', it is Athom controlling those. I cannot customize them or Homey Energy support would break.

Barblo1986, 2020-11-19:
I am using a Envoy-S-Metered-EU with the newest firmware R4.10.35

Barblo1986, 2020-11-19:
[image]

DiedB, 2020-11-19:
Can you visit `http://{enphase_ip}/production.json?details=1` and post the result here?

Barblo1986, 2020-11-21:
{"production":[{"type":"inverters","activeCount":6,"readingTime":1605960738,"wNow":157,"whLifetime":1546719},{"type":"eim","activeCount":1,"measurementType":"production","readingTime":1605961173,"wNow":253.375,"whLifetime":1519928.012,"varhLeadLifetime":0.0,"varhLagLifetime":475714.77,"vahLifetime":1877561.999,"rmsCurrent":1.219,"rmsVoltage":230.104,"reactPwr":87.98,"apprntPwr":280.134,"pwrFactor":0.9,"whToday":662.012,"whLastSevenDays":16410.012,"vahToday":1797.999,"varhLeadToday":0.0,"varhLagToday":1220.77,"lines":[{"wNow":253.375,"whLifetime":1519928.012,"varhLeadLifetime":0.0,"varhLagLifetime":475714.77,"vahLifetime":1877561.999,"rmsCurrent":1.219,"rmsVoltage":230.104,"reactPwr":87.98,"apprntPwr":280.134,"pwrFactor":0.9,"whToday":662.012,"whLastSevenDays":16410.012,"vahToday":1797.999,"varhLeadToday":0.0,"varhLagToday":1220.77}]}],"consumption":[{"type":"eim","activeCount":1,"measurementType":"total-consumption","readingTime":1605961173,"wNow":435.875,"whLifetime":1688278.998,"varhLeadLifetime":1040640.696,"varhLagLifetime":485371.607,"vahLifetime":2817336.857,"rmsCurrent":2.75,"rmsVoltage":230.155,"reactPwr":-345.606,"apprntPwr":632.909,"pwrFactor":0.69,"whToday":4083.998,"whLastSevenDays":58465.998,"vahToday":5632.857,"varhLeadToday":3039.696,"varhLagToday":1225.607,"lines":[{"wNow":435.875,"whLifetime":1688278.998,"varhLeadLifetime":1040640.696,"varhLagLifetime":485371.607,"vahLifetime":2817336.857,"rmsCurrent":2.75,"rmsVoltage":230.155,"reactPwr":-345.606,"apprntPwr":632.909,"pwrFactor":0.69,"whToday":4083.998,"whLastSevenDays":58465.998,"vahToday":5632.857,"varhLeadToday":3039.696,"varhLagToday":1225.607}]},{"type":"eim","activeCount":1,"measurementType":"net-consumption","readingTime":1605961173,"wNow":182.5,"whLifetime":1195282.363,"varhLeadLifetime":1040640.696,"varhLagLifetime":9656.837,"vahLifetime":2817336.857,"rmsCurrent":1.531,"rmsVoltage":230.207,"reactPwr":-257.625,"apprntPwr":352.437,"pwrFactor":0.52,"whToday":0,"whLastSevenDays":0,"vahToday":0,"varhLeadToday":0,"varhLagToday":0,"lines":[{"wNow":182.5,"whLifetime":1195282.363,"varhLeadLifetime":1040640.696,"varhLagLifetime":9656.837,"vahLifetime":2817336.857,"rmsCurrent":1.531,"rmsVoltage":230.207,"reactPwr":-257.625,"apprntPwr":352.437,"pwrFactor":0.52,"whToday":0,"whLastSevenDays":0,"vahToday":0,"varhLeadToday":0,"varhLagToday":0}]}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}

Barblo1986, 2020-11-21:
> Can you visit `http://{enphase_ip}/production.json?details=1` and post the result here?
{"production":[{"type":"inverters","activeCount":6,"readingTime":1605960738,"wNow":157,"whLifetime":1546719},{"type":"eim","activeCount":1,"measurementType":"production","readingTime":1605961173,"wNow":253.375,"whLifetime":1519928.012,"varhLeadLifetime":0.0,"varhLagLifetime":475714.77,"vahLifetime":1877561.999,"rmsCurrent":1.219,"rmsVoltage":230.104,"reactPwr":87.98,"apprntPwr":280.134,"pwrFactor":0.9,"whToday":662.012,"whLastSevenDays":16410.012,"vahToday":1797.999,"varhLeadToday":0.0,"varhLagToday":1220.77,"lines":[{"wNow":253.375,"whLifetime":1519928.012,"varhLeadLifetime":0.0,"varhLagLifetime":475714.77,"vahLifetime":1877561.999,"rmsCurrent":1.219,"rmsVoltage":230.104,"reactPwr":87.98,"apprntPwr":280.134,"pwrFactor":0.9,"whToday":662.012,"whLastSevenDays":16410.012,"vahToday":1797.999,"varhLeadToday":0.0,"varhLagToday":1220.77}]}],"consumption":[{"type":"eim","activeCount":1,"measurementType":"total-consumption","readingTime":1605961173,"wNow":435.875,"whLifetime":1688278.998,"varhLeadLifetime":1040640.696,"varhLagLifetime":485371.607,"vahLifetime":2817336.857,"rmsCurrent":2.75,"rmsVoltage":230.155,"reactPwr":-345.606,"apprntPwr":632.909,"pwrFactor":0.69,"whToday":4083.998,"whLastSevenDays":58465.998,"vahToday":5632.857,"varhLeadToday":3039.696,"varhLagToday":1225.607,"lines":[{"wNow":435.875,"whLifetime":1688278.998,"varhLeadLifetime":1040640.696,"varhLagLifetime":485371.607,"vahLifetime":2817336.857,"rmsCurrent":2.75,"rmsVoltage":230.155,"reactPwr":-345.606,"apprntPwr":632.909,"pwrFactor":0.69,"whToday":4083.998,"whLastSevenDays":58465.998,"vahToday":5632.857,"varhLeadToday":3039.696,"varhLagToday":1225.607}]},{"type":"eim","activeCount":1,"measurementType":"net-consumption","readingTime":1605961173,"wNow":182.5,"whLifetime":1195282.363,"varhLeadLifetime":1040640.696,"varhLagLifetime":9656.837,"vahLifetime":2817336.857,"rmsCurrent":1.531,"rmsVoltage":230.207,"reactPwr":-257.625,"apprntPwr":352.437,"pwrFactor":0.52,"whToday":0,"whLastSevenDays":0,"vahToday":0,"varhLeadToday":0,"varhLagToday":0,"lines":[{"wNow":182.5,"whLifetime":1195282.363,"varhLeadLifetime":1040640.696,"varhLagLifetime":9656.837,"vahLifetime":2817336.857,"rmsCurrent":1.531,"rmsVoltage":230.207,"reactPwr":-257.625,"apprntPwr":352.437,"pwrFactor":0.52,"whToday":0,"whLastSevenDays":0,"vahToday":0,"varhLeadToday":0,"varhLagToday":0}]}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}

DiedB, 2020-11-23:
Strange, your output aligns with what I would expect. What Homey software version are you running?

Barblo1986, 2020-11-23:
I have a Homey (Early 2019) running on 4.2.0

DiedB, 2020-11-24:
Could you try the version at https://homey.app/a/it.diederik.solar/test/?

Barblo1986, 2020-11-25:
Hi,

Have the latest version installed and the problem with the unit of total generated power is solved :) 👍 :) 

Only the actual power generated will now be made negative in the evenings (when there is no sun). This because  the micro-inverters from Enphase use a little power.

As a result, the total generated power is incorrect. If I can see it properly, the negative values will be added to the total.

DiedB, 2020-11-27:
Can you be more clear in the values you are seeing (W / kWh)? From your description, I think the app is behaving correctly.

closed 2020-11-27 by DiedB