DiedB/Homey-SolarPanels #137: Ferroamp support?

jampeef, 2020-12-30:
Any chance to get ferroamp support in this nice app. 
Someone made something for HA

https://github.com/henricm/ha-ferroamp

DiedB, 2021-01-09:
If others are in need for Ferroamp support too, please let me know here.

ChristerSjo, 2021-02-09:
I'm in need of support for Ferroamp Energyhub I will replace my Solar Edge with Ferroamp Energyhub soon.

DiedB, 2021-03-07:
Does the device have a local API you can connect to? Does it show production data? I prefer not going the MQTT route, that works well for Homekit but is a little more involved for Homey Apps.

jampeef, 2021-03-08:
I'm not sure other then that it has that mqtt broker and the cloud portal.
It's supposed to have an open API according the manual. I can call them and
check to see the possibilities.

ChristerSjo, 2021-03-08:
Yes it have an open API. I have write to Ferroamp to get hold of the documentation around the API. I will be back.

DiedB, 2021-03-08:
Great, will implement it once the documentation is here!

ChristerSjo, 2021-03-09:
Here is the API documentations I get from Ferroamp.

ChristerSjo, 2021-03-10:
I don´t now if you need it but here is the Modbus API.
[Ferroamp Modbus TCP Specification RevB.pdf](https://github.com/DiedB/Homey-SolarPanels/files/6113032/Ferroamp.Modbus.TCP.Specification.RevB.pdf)

jampeef, 2021-03-15:
> Could you port forward the Modbus port (502) to the outside and send me your IP address (preferably in private, can be done via the Athom Community Forum)? I will have a look.
> 
> Also, do you have access to Energy Cloud and can you generate API keys there?

Hey. I can make them open up the cloud api.  You should be able to generate apikeys there

jampeef, 2021-03-15:
So I talked to a technician om ferroamp and all the apis will on open in mine tomorrow. 
I'd really like to add flowcards to trigger battery discharge and so on

DiedB, 2021-03-15:
@jampeef What is needed to get Cloud API keys? Would this be an easy process for every user?

jampeef, 2021-03-15:
> @jampeef What is needed to get Cloud API keys? Would this be an easy process for every user?

I'm not sure yet. They're updating my software atm so I will be able to trigger battery discharge from the portal. I'll know tomorrow if it's easy getting api keys. I would assume it is.

jampeef, 2021-03-19:
so been busy with other stuff, I now have all Apis open , seems pretty bad for getting cloud api keys, as I cant seem to find any documentation how to obtain keys other that from ferroamp support.
I've quickly just logged into the mqtt broker and checked to see and here where you get everything.
I'm not sure how to do anything with the modbus tcp connection.

jampeef, 2021-03-20:
> @jampeef What is needed to get Cloud API keys? Would this be an easy process for every user?

it would be easy if you just email ferroamp they give you a key, so should be ok, i thought there was a way to make a token yourself from the cloud API.   To trigger anything on the battery side you need to connect to the mqtt broker

DiedB, 2021-03-20:
If there's not a way to make a token yourself, then that option is not user-friendly enough and the options for MQTT or Modbus are left. While I would prefer Modbus, MQTT will probably be the easiest to support, I cannot find any Modbus documentation. I will need access to a Ferroamp inverter on port 1883 in order to implement it, can someone make a port forward and send me the address in private?

jampeef, 2021-03-20:
> If there's not a way to make a token yourself, then that option is not user-friendly enough and the options for MQTT or Modbus are left. While I would prefer Modbus, MQTT will probably be the [...]

I could do that yes, not sure how to do a PM on github though :)

DiedB, 2021-03-20:
Not possible, send it to diederik@outlook.com

jampeef, 2021-03-26:
> Not possible, send it to [diederik@outlook.com](mailto:diederik@outlook.com)

So did you have a look at this or?

nichlaz, 2021-08-14:
I am also interested in Ferroamp support!

jampeef, 2021-08-14:
Nothing happen here so i just added it to a home assistant docker and then
use the home assistant app in homey. Not optimal at all but get the info

nichlaz, 2021-08-14:
I've tried ha-ferroamp but I can't seem to get it to work. What's the name of the Home Assistant app in Homey (can't find it)?

jampeef, 2021-08-14:
Think it's only available in the  homey community store.

https://store.homey.community/

handtema, 2021-12-30:
I think I would love ferroamp support also. I can open up ports for you to my Energyhub if its still noone who has done that.

I also have the modbus + local api documentation if needed.

Teel76, 2022-04-03:
Also interested in support for Ferroamp in homey. Any news on this?

matsskoglund, 2022-05-07:
I am also interested in Ferroamp support in Homey.

johhiq, 2022-08-21:
Howdy! Also eager to integrate Ferroamp to Homey. Any news on this topic?
If you need support with test just let me know.
//J

GurraGe, 2023-06-21:
Would love an FerroAmp - Homey integration.

ZeroDmg, 2024-02-19:
Any news on this topic?

jampeef, 2024-02-19:
A homey ferroamp app was done by ola

https://homey.app/sv-se/app/io.olae.ferroamp/Ferroamp/

ZeroDmg, 2024-02-19:
Hi, 

Ok nice thanks for info :) 

//Peter 

2024-02-19 17:22 skrev jampeef:

> A homey ferroamp app was done by ola 
> 
> https://homey.app/sv-se/app/io.olae.ferroamp/Ferroamp/ 
> 
> 
>> Any news on this topic? 

  

Links:
------
[1]
https://github.com/DiedB/Homey-SolarPanels/issues/137#issuecomment-1952809955
[2]
https://github.com/
--=_9ecb2ffa99b4d59fca10bc4850af1b70
Content-Type: multipart/related;
 boundary="=_09fc650b6e26eee266b5d9299a685dc8"

--=_09fc650b6e26eee266b5d9299a685dc8
Content-Transfer-Encoding: quoted-printable
Content-Type: text/html; charset=UTF-8

<html><head><meta http-equiv=3D"Content-Type" content=3D"text/html; charset=
=3DUTF-8" /></head><body style=3D'font-size: 10pt; font-family: Verdana,Gen=
eva,sans-serif'>
<p>Hi,</p>
<p>Ok nice thanks for info :)</p>
<p>//Peter</p>
<p><br /></p>
<p id=3D"reply-intro">2024-02-19 17:22 skrev jampeef:</p>
<blockquote type=3D"cite" style=3D"padding: 0 0.4em; border-left: #1010ff 2=
px solid; margin: 0">
<div id=3D"replybody1">
<p><br /></p>
A homey ferroamp app was done by ola <br /> <br /> https://homey.app/sv-se/=
app/io.olae.ferroamp/Ferroamp/ <br /> <br /> Den m&aring;n 19 feb. 2024 17:=
18ZeroDmg ***@***.***&gt; skrev: <br /> <br /> &gt; Any news on this topic?=
view it on GitHub <br /> &gt; &lt;https://github.com/DiedB/Homey-SolarPanel=
FUU3ZLELYUN3OHAVCNFSM4VOTPVIKU5DIOJSWCZC7NNSXTN2JONZXKZKDN5WW2ZLOOQ5TCOJVGI=
<p style=3D"font-size: small; -webkit-text-size-adjust: none; color: #666;"=
iedB/Homey-SolarPanels/issues/137#issuecomment-1952809955" target=3D"_blank=
" rel=3D"noopener noreferrer">view it on GitHub</a>, or <a href=3D"https://=
FSM4VOTPVIKU5DIOJSWCZC7NNSXTN2JONZXKZKDN5WW2ZLOOQ5TCOJVGI4DAOJZGU2Q" target=
=3D"_blank" rel=3D"noopener noreferrer">unsubscribe</a>.<br />You are recei=
ving this because you commented.<img src=3D"cid:170836307665d38d44007b48270=
rent; font-size: 0; display: none; visibility: hidden; overflow: hidden; op=
acity: 0; width: 0px; height: 0px; max-width: 0; max-height: 0; mso-hide: a=
pan><span>@</span><span>github</span><span>.</span><span>com&gt;</span></sp=
an></p>
</div>
</blockquote>
<div id=3D"vtPopupMenu"> </div>
</body></html>

--=_09fc650b6e26eee266b5d9299a685dc8
Content-Transfer-Encoding: base64
Content-ID: ***@***.***>
Content-Type: image/gif;
 name=blocked.gif
Content-Disposition: inline;
 filename=blocked.gif;
 size=118

R0lGODlhZAAyAIAAAPrOzgAAACH5BAAAAAAALAAAAABkADIAAAJNhI+py+0Po5y02ouz3rz7D4bi
SJbmiabqyrbuC8fyTNf2jef6zvf+DwwKh8Si8YhMKpfMpvMJjUqn1Kr1is1qt9yu9wsOi8fksvls
KwAAOw==
--=_09fc650b6e26eee266b5d9299a685dc8--

--=_9ecb2ffa99b4d59fca10bc4850af1b70--