DiedB/Homey-SolarPanels #138: Wil toch niet connecten met Solis 4G Inverter:

Jansiesling, 2021-01-09:
Beste,
ik heb nu ook een inverter van Solis.
Heb bovenstaande uitgevoerd via ( m.ginlong.com) ook het plant num: wat daar vernoemt wordt geeft geen verbinding met de inverter?
Heb ook het serienum van de datalogging stick ingevoerd die herkent hij ook niet.
weet jij hier meer over? @DiedB

Vr GR Jan

DiedB, 2021-01-09:
Haal je het plant ID uit de URL-balk? Volgens mij was het een 5 of 6-cijferig nummer.

Jansiesling, 2021-01-09:
Dag Diederik,

Snel antwoord dank hiervoor, heb onder V5 de app opnieuw op de homey gezet, zie hierboven het screenschot ik gebruik het nummer vernoemt zoals bij plant ID: 799153

Het andere nummer is van mijn Data Logging stick (die verbind de inverter met de router via WIFI) num: 4023709387

Vriendelijke groet,

 

Jan Siesling

Mobiel: 0642015915

 

Deze e-mail is gecontroleerd op virussen door AVG.
http://www.avg.com

DiedB, 2021-01-09:
Het Plant ID is de juiste. Volgens de Ginlong API zijn er geen omvormers gekoppeld aan het account. Wellicht is de omvormer te nieuw en wordt hij daardoor niet getoond? Als je inlogt op het Ginlong Portal, krijg je dan wel productiestatistieken?

Jansiesling, 2021-01-09:
Dag Diederik,

Zie bij de andere tabbladen geen verdere info staan! Dus ik zie geen productie statistieken.

Via de app (Solis home) zie ik wel de info.

Wat ik kan doen voor je is het boekje kopiëren via pdf jou dan opsturen, misschien dat je daar dan wat mee kan doen?

 

Vriendelijke groet,

 

Jan Siesling

Mobiel: 0642015915

 

http://www.avg.com

Jansiesling, 2021-02-14:
Beste Diederik,

Was je er nog mee verder aan de gang geweest om het mogelijk te maken dat mensen die een nieuwere Ginlong hebben die kunnen koppelen aan de homey via jou app?

 

Vriendelijke groet,

 

Jan Siesling

Snipweg 26

7331LV Apeldoorn 

 

http://www.avg.com

DiedB, 2021-02-17:
Nog niet, ik heb vrij weinig tijd en ik heb geen documentatie van een (publieke) API voor de nieuwe Solis inverters. Voorlopig gaat dit dus denk ik niet gebeuren, maar ik laat de issue open staan mocht dat ik de toekomst veranderen.

Jansiesling, 2021-02-17:
Ik heb wel een installation and Operation Manual voor je die bij de huidige Inverter zat.

Kan die wel omzetten naar PDF, indien je dat zou willen?

 

Vriendelijke groet,

 

Jan Siesling

 

http://www.avg.com

DiedB, 2021-02-17:
Helaas, de informatie die ik nodig heb staat daar niet in. Ik denk overigens dat Ginlong dit helemaal niet officieel ondersteunt, en zolang dat niet zo is, is het lastig om ondersteuning te bieden via een Homey-app

SvB4u, 2023-03-01:
> Nog niet, ik heb vrij weinig tijd en ik heb geen documentatie van een (publieke) API voor de nieuwe Solis inverters. Voorlopig gaat dit dus denk ik niet gebeuren, maar ik laat de issue open staan mocht dat ik de toekomst veranderen.

@DiedB hier is de api documentatie:

https://solis-service.solisinverters.com/support/solutions/articles/44002212561-api-access-soliscloud

https://solis-service.solisinverters.com/helpdesk/attachments/2043393248854

Upe-G, 2023-10-28:
Is er al nieuws? Kan Ginlong/Solis mogelijk toegevoegd worden aan de app voor HP23?