DiedB/Homey-SolarPanels #140: Enphase Envoy discovery op ander subnet

ElStupid, 2021-03-05:
Discovery werkt prima, mijn Enphase was direct gevonden toen ik m installeerde.
Inmiddels ben ik mijn netwerk wat aan het scheiden. De Enphase zit op een ander subnet dan de Homey en nu is de connectie weg met mijn omvormer. Is er een mogelijkheid om zelf een IP adres in te voeren?

DiedB, 2021-03-06:
Hmm, dat is een interessant geval. Is er geen mogelijkheid om mDNS te relayen tussen je subnets? Dan ben je als het goed is ook van het probleem af.

Ik gebruik de SDK om discovery te doen, dus het wordt een beetje omslachtig om daar buitenom ook nog handmatig IP-adressen in te laten voeren. Wil niet zeggen dat het niet kan, maar ik doe het liever niet.

ElStupid, 2021-03-06:
Okee, snap dat je dat niet wilt. Op mijn Toon (thermostaat) kan ik gewoon het IP adres van de Envoy instellen en dat gaat goed (ook ander subnet). Ik zal es kijken wat ik voor elkaar kan krijgen met mDNS. Niet mijn sterke punt ;)

DiedB, 2021-03-07:
Wat voor router gebruik je? Ik kan je wellicht wat resources doorsturen. Op mijn EdgeRouter was het simpelweg het installeren van een daemon via de CLI.

ElStupid, 2021-03-07:
Ik gebruik er eentje van TP-Link, de TL-ER7206. Als je wat kan vinden zou het mooi zijn, maar anders ga ik mijn Envoy in hetzelfde LAN hangen

DiedB, 2021-03-08:
Ik kan hier weinig over vinden helaas, zelf ook geen (recente) ervaring met TP-Link spul. Wellicht dat Athom ooit een handmatige override implementeert voor mDNS discovery, is naar mijn optiek wel een must namelijk.

closed 2021-03-08 by DiedB

ElStupid, 2021-03-08:
Thanks voor de hulp tot zover. Ik heb de vraag inmiddels ook bij TP-Link neergelegd. Als ik meer weet meld ik het nog wel even.