DiedB/Homey-SolarPanels #143: SAJ inverter

psvrob, 2021-03-29:
Hi,

I have a problem connecting my new SAJ inverter to homey. It keeps saying that my ip adres is wrong. This is not true because i have checked the ip adres on my router. 

Any ideas how to solve this?

DiedB, 2021-03-30:
Can you open a browser and go to `http://[ip_address]/real_time_data.xml`? If that doesn't work, can you visit the IP address directly?

POfransz, 2021-06-09:
Hi Diederik...I have the same problem. Do you mean the access point IP-adress or the original IP-adress when you start installing the SAJ inverter. I shall try both....when I it does not work shall I come back with this.
The comment in your app is a little bit a riddle when not a success. Is it possible to change that?

DiedB, 2021-06-09:
The IP address currently assigned to the SAJ inverter. Can be found in your router, maybe in the SAJ app, maybe on the screen of the inverter (if it has one).

POfransz, 2021-06-09:
1) I have tried the access point  "http://[ip_address]/real_time_data.xml" and I get the sign-in screen of the access point.
   After signing in ...I get in the firefox browser "404: The requested file cannot be found."
2) I have also tried the IP-adress 10.10.176.1....but I get nothing..

OK....I am not so an IT-guy...and I will explain it as good as I can....
I renamed te access point. changed the admin name and password.
The SAJ-inverter has an Webserver which create an access point, but I think you know that already.
Here the link https://solarnrg.nl/downloads/handleidingen/solarnrg-handleiding-saj-wifi-module.pdf#page=7 for your information

DiedB, 2021-06-10:
You need to connect the SAJ inverter to your home network (the same network Homey is connected to). That should be possible using the web server you already found.

Afterwards, you will have to find the IP address of the inverter in your home network. Possibly, it gives you the IP address after connecting, otherwise, you will have to have a look in the web interface of your router.

POfransz, 2021-06-11:
Hi Diederik,
I think we misunderstand each other. The connection runs trough the WiFi adapter of the inverter. The IP-adress is known to me.
When I use this in my browser ...a sign-in menu where I get a sort of router menu....appears....for adjusting data or name.
The ...xml does not exist...because I see no files.--I cannot see what is on the adapter.
I contacted my solarpanel supplier and asked him for technical advise where I can reach the requested data because this is visible in my browser trough a kind of menu. I shall come back with this .....when this known to me and perhaps solved.

hendrik607, 2021-06-15:
Same here, Inverter is live, XML not found, see images
[http://challenger.zone/diversen/inverter1.JPG](http://challenger.zone/diversen/inverter1.JPG)
[http://challenger.zone/diversen/inverter12JPG](http://challenger.zone/diversen/inverter2.JPG)

DiedB, 2021-06-15:
Can you find any production statistics in that web interface?

hendrik607, 2021-06-16:
yes there is [http://challenger.zone/diversen/saj.jpg](http://challenger.zone/diversen/saj.jpg)

POfransz, 2021-06-17:
Hi Diederik and Hendrik,
@hendrik - you have the same Webserver version as I have....so the data must be available.
I just received an answer from my solarpanel supplier. That was a dead end...because they do not sell that kind of inverters anymore. But I got an e-mail adress from SAJ. Here the link....mailto:Service.nl@saj-electric.com.
I shall send an e-mail today.

DiedB, 2021-06-17:
@hendrik607 Is there by any chance also a `status.json`?

What type of SAJ inverter do you have exactly? I think this is a newer version with a different web interface and API and we should add support for it.

POfransz, 2021-06-17:
@DiedB ....Mine inverter is Sununo Plus 2.5K....Installation date 25-9-2017.

hendrik607, 2021-06-17:
hi, i think mine is year 2018, do not know exactle it was once changed.
here is the label [http://challenger.zone/diversen/saj4.jpg](http://challenger.zone/diversen/saj4.jpg)

How should i send this (status.json?) ? 
When doing http://<ip>/status.json? i get nothing, also nothing with an extra /status/ in URL

DiedB, 2021-06-23:
I meant `http://ip/i18n/en/status.json` (just replace `xml` with `json` in your URL).

Just paste it here in code tags.

hendrik607, 2021-06-23:
gives me nothing (404: The requested file cannot be found.)
http://192.168.2.133/i18n/en/status.xml does give me output;

<m k="statusTittle" v="Device Status"/>
<m k="statistics" v="Statistics"/>
<m k="totalGen" v="Total generated"/>
<m k="totalRunTime" v="Total Running Time"/>
<m k="todayGen" v="Today generated"/>
<m k="todayRunTime" v="Today Running Time"/>
<m k="DCinput" v="DC Input"/>
<m k="PV1vol" v="PV1 Voltage"/>
<m k="PV1cur" v="PV1 Current"/>
<m k="PV2vol" v="PV2 Voltage"/>
<m k="PV2cur" v="PV2 Current"/>
<m k="PV3vol" v="PV3 Voltage"/>
<m k="PV3cur" v="PV3 Current"/>
<m k="PV1StrCurr1" v="PV1 StrCurr1"/>
<m k="PV1StrCurr2" v="PV1 StrCurr2"/>
<m k="PV1StrCurr3" v="PV1 StrCurr3"/>
<m k="PV1StrCurr4" v="PV1 StrCurr4"/>
<m k="PV2StrCurr1" v="PV2 StrCurr1"/>
<m k="PV2StrCurr2" v="PV2 StrCurr2"/>
<m k="PV2StrCurr3" v="PV2 StrCurr3"/>
<m k="PV2StrCurr4" v="PV2 StrCurr4"/>
<m k="PV3StrCurr1" v="PV3 StrCurr1"/>
<m k="PV3StrCurr2" v="PV3 StrCurr2"/>
<m k="PV3StrCurr3" v="PV3 StrCurr3"/>
<m k="PV3StrCurr4" v="PV3 StrCurr4"/>
<m k="ACouput" v="AC Output"/>
<m k="gridConPwr" v="Grid-connected Power"/>
<m k="gridConFreq" v="Grid-connected Frequency"/>
<m k="line1vol" v="Line1 Voltage"/>
<m k="line2vol" v="Line2 Voltage"/>
<m k="line3vol" v="Line3 Voltage"/>
<m k="line1cur" v="Line1 Current"/>
<m k="line2cur" v="Line2 Current"/>
<m k="line3cur" v="Line3 Current"/>
<m k="devStatus" v="Other Status"/>
<m k="runState" v="Running State"/>
<m k="busvol" v="Bus Voltage"/>
<m k="devtemp" v="Device Temperature"/>
<m k="co2red" v="CO<sub>2</sub>emission reduction"/>
<m k="Statistchart" v="Statist chart"/>
<m k="chart1" v="Today"/>
<m k="chart2" v="This month"/>
<m k="chart3" v="This year"/>
<m k="chart4" v="Total"/>
<m k="cx1" v="Time"/>
<m k="cx2" v="Days"/>
<m k="cx3" v="Month"/>
<m k="cx4" v="Year"/>
<m k="kwh" v="kWh"/>
<m k="w" v="W"/>
<m k="m0" v="Not connect"/>
<m k="m1" v="Waiting"/>
<m k="m2" v="Normal"/>
<m k="m3" v="Error"/>
<m k="m4" v="Upgrading"/>
<m k="p0" v=""/>
<m k="p1" v="×10 "/>
<m k="p2" v="×100 "/>
<m k="p3" v="×1K "/>
<m k="p4" v="×10K "/>
<m k="p5" v="×100K "/>
<m k="p6" v="×1M "/>
<m k="p7" v="×10M "/>
<m k="p8" v="×100M "/>
<m k="p9" v="×1G "/>
<m k="u1" v="kWh"/>
<m k="u2" v="h"/>
<m k="u3" v="kWh"/>
<m k="u4" v="h"/>
<m k="u5" v="V"/>
<m k="u6" v="A"/>
<m k="u7" v="V"/>
<m k="u8" v="A"/>
<m k="u9" v="V"/>
<m k="u10" v="A"/>
<m k="u11" v="A"/>
<m k="u12" v="A"/>
<m k="u13" v="A"/>
<m k="u14" v="A"/>
<m k="u15" v="A"/>
<m k="u16" v="A"/>
<m k="u17" v="A"/>
<m k="u18" v="A"/>
<m k="u19" v="A"/>
<m k="u20" v="A"/>
<m k="u21" v="A"/>
<m k="u22" v="A"/>
<m k="u23" v="W"/>
<m k="u24" v="Hz"/>
<m k="u25" v="V"/>
<m k="u26" v="A"/>
<m k="u27" v="V"/>
<m k="u28" v="A"/>
<m k="u29" v="V"/>
<m k="u30" v="A"/>
<m k="u31" v="V"/>
<m k="u32" v="℃"/>
<m k="u33" v="Kg"/>
<m k="mon0" v="Jan"/>
<m k="mon1" v="Feb"/>
<m k="mon2" v="Mar"/>
<m k="mon3" v="Apr"/>
<m k="mon4" v="May"/>
<m k="mon5" v="Jun"/>
<m k="mon6" v="Jul"/>
<m k="mon7" v="Aug"/>
<m k="mon8" v="Sep"/>
<m k="mon9" v="Oct"/>
<m k="mon10" v="Nov"/>
<m k="mon11" v="Dec"/>

POfransz, 2021-06-30:
I have read your comments, Here my listing.
[SAJ-Listing.pdf](https://github.com/DiedB/Homey-SolarPanels/files/6741889/SAJ-Listing.pdf)

rsimmers, 2021-08-02:
I have the same problem, is it solved yet?

DiedB, 2021-08-02:
@hendrik607 @POfransz Those are translation files, as far as I can see no production data is in there. @rsimmers Did you find production data in XML format?

rsimmers, 2021-08-04:
No nothing. Is it possible to connect Anyway?

René Simmers

DiedB, 2021-08-04:
Not without more information. Mind you, I don't have this inverter, and not much public documentation is available. I will only be able to implement this when assisted by someone with quite some technical knowledge who owns one of these inverters.

POfransz, 2021-08-04:
I still follow this issue and perhaps can help.
I received from solarpanel supplier the attached file which is the internal protocol from the SAJ inverter

[Modbus-protocol.pdf](https://github.com/DiedB/Homey-SolarPanels/files/6931282/Modbus-protocol.pdf)

POfransz, 2021-08-04:
I also got a remark that the software to read the SAJ-inverter is also on github.
Here the link https://github.com/jwillemsen/saj_collector
I hope this will help

DiedB, 2021-08-04:
@POfransz ModBUS works via UART, no way to connect that to Homey (unless you would add some hardware on the inverter side, not really possible for the average user). The SAJ Collector you mention is using the same method for fetching production data as the current implementation of this app.

DiedB, 2021-08-04:
> Of course, i / we want to help because this is a nice feature. Tell me how to help. The standard xml seems not to be there. There is a php that gives the values as output. Can that be used?

Maybe, but it would probably be quite hacky and I'm not sure if the interface is consistent across all inverter models. So preferably not.

rsimmers, 2021-08-04:
I’ll be glad to help. I have such a converter and what do you need?

Rene

rsimmers, 2021-08-04:
I have installeer zonnepanelen and add a Saj device. I files in the richt IP adres but it return that the data is wrong. What can i do?

Rene

drieshooghe, 2021-09-29:
> yes there is http://challenger.zone/diversen/saj.jpg
> 
> http://ip/status/status.php gives me;
> 
> 1,1460417,126931,3093,163,1472,4,1774,4,65535,65535,65535,65535,65535,65535,65535,65535,65535,65535,65535,65535,65535,65535,0,5000,2260,8,65535,65535,65535,65535,3546,357,49106,2
> 
> with above string, page looks like:
> 
> http://challenger.zone/diversen/saj2.JPG
> 
> also there is http://ip/i18n/en/status.xml
> 
> geving me;
> 
> http://challenger.zone/diversen/saj3.JPG

For the WiFi D module, which communicates real time data through http://[IP]/status/status.php I've created a PR: https://github.com/DiedB/Homey-SolarPanels/pull/164

rsimmers, 2021-09-30:
Does this mean i can try it with my homey?

René Simmers

chrisboer, 2022-02-27:
@DiedB I have 2 of these SAJ and I also have Solarman so all of my solarpanels can not be monitored anymore. I can help you setup the SAJ because it worked perfectly with previous versions.

chrisboer, 2022-02-27:
If I connect to my local network I can see you can readout param.js 

var xmlhttp_par;
function display_par()
{
if(xmlhttp_par.readyState==4&& xmlhttp_par.status==200)
{
var xmlDoc=xmlhttp_par.responseXML.documentElement;

with(document){
getElementById("v-grid").value=xmlDoc.getElementsByTagName("v-grid")[0].childNodes[0].nodeValue+" V";
getElementById("i-grid").value=xmlDoc.getElementsByTagName("i-grid")[0].childNodes[0].nodeValue+" A";
getElementById("f-grid").value=xmlDoc.getElementsByTagName("f-grid")[0].childNodes[0].nodeValue+" Hz";
getElementById("p-ac").value=xmlDoc.getElementsByTagName("p-ac")[0].childNodes[0].nodeValue+" W";
getElementById("temp").value=xmlDoc.getElementsByTagName("temp")[0].childNodes[0].nodeValue+" ℃";
getElementById("e-today").value=xmlDoc.getElementsByTagName("e-today")[0].childNodes[0].nodeValue+" kWh";
getElementById("t-today").value=xmlDoc.getElementsByTagName("t-today")[0].childNodes[0].nodeValue+" h";
getElementById("e-total").value=xmlDoc.getElementsByTagName("e-total")[0].childNodes[0].nodeValue+" kWh";
getElementById("t-total").value=xmlDoc.getElementsByTagName("t-total")[0].childNodes[0].nodeValue+" h";
getElementById("CO2").value=xmlDoc.getElementsByTagName("CO2")[0].childNodes[0].nodeValue+" kg";
getElementById("v-pv1").value=xmlDoc.getElementsByTagName("v-pv1")[0].childNodes[0].nodeValue+" V";
getElementById("i-pv1").value=xmlDoc.getElementsByTagName("i-pv1")[0].childNodes[0].nodeValue+" A";
getElementById("v-pv2").value=xmlDoc.getElementsByTagName("v-pv2")[0].childNodes[0].nodeValue;
getElementById("i-pv2").value=xmlDoc.getElementsByTagName("i-pv2")[0].childNodes[0].nodeValue;
getElementById("v-bus").value=xmlDoc.getElementsByTagName("v-bus")[0].childNodes[0].nodeValue+" V";
getElementById("Inverter-Stare").value=xmlDoc.getElementsByTagName("state")[0].childNodes[0].nodeValue;
}
}
}

DiedB, 2022-03-14:
If someone wants to implement this into a separate app, I am willing to help. In the current app, SAJ will likely be deprecated because there are too many different inverters and software versions, all requiring their own implementation. Will likely be confusing for users and thus lead to lots of questions.

hendrik607, 2022-03-14:
if i can help, let me know, isnt it a option to get the data from the portal? https://fop.saj-electric.com/saj?login

chrisboer, 2023-01-14:
Hi Dirrick I have time now to fix and test it with you maybe it is wise to extract the data locally via XML 

[image]

closed 2023-06-10 by DiedB