DiedB/Homey-SolarPanels #144: Enphase connection issues

arnoarnoo, 2021-04-10:
Hi Diederik,
Goed dat je voor Homey ontwikkelt!
Ik kan mijn Enphase envoy niet activeren als apparaat; error melding;
Error retrieving data (Fetcherror, invalid response body at http..nr.... error reason unexpected token < in JSON at position 4.

daarna dacht ik met Enlighten te werken, maar de activatie vraagt naar de Api key en die heb je niet vrijgegeven.

bedankt voor je hulp,
Arno

DiedB, 2021-04-10:
Dank voor je donatie!

Ik zou de Enlighten integratie niet aanraden, door limitaties aan de kant van Enphase moet je je eigen developer account registeren, je systeem koppelen en dan de API-key opvragen. Zie het eerder als een fallback als je omvormer Envoy niet blijkt te ondersteunen.

Zou je eens kunnen kijken wat je krijgt als je naar `http://[inverter-ip]/production.json` gaat?

arnoarnoo, 2021-04-10:
Hi,
Die url bestaat niet, geeft een 401 error.
De production menu item gaat naar http://-IP-adres/production?locale=en 
De envoy web toegang werkt gewoon en ik zie alle details van alle 18 (micro)omvormers inclusief dagelijkse productie per omvormer.
Het PV-Systeem heeft een omvormer per paneel. De Envoy ondersteunt dus de omvormers.
Groeten,

Arno

DiedB, 2021-04-12:
Interessant, welk model Envoy heb je? Zit er enige wachtwoordbeveiliging op?

Zie https://enphase.com/en-us/support/what-envoy voor een referentie. Ik heb de app getest op een Envoy S, als ik het me goed herinner, maar het zou inderdaad best kunnen dat andere modellen ander gedrag vertonen.

arnoarnoo, 2021-04-12:
Hi Diederik,
ik heb de gewone Envoy, de ovale. Volgens mij de voorloper van de Envoy-S.
Hij heeft geen wachtwoordbeveiliging: plug and play. Wel een enlighten manager account; dus de functionaliteit van de envoy is beschikbaar via de account. De Enlighten app werkt prima.
ik heb ff de API technical brief instructie geprobeerd: http://{ip-envoy}/api/v1/production geeft exact de respons die in de technical brief staat.

en als ik naar {IP-Envoy}/production ga, krijg ik gewoon het openingsscherm, met het hele menu.
\Het is die .json toevoeging in jouw email die de problemen geeft.

groeten,
Arno

DiedB, 2021-04-14:
Ik zie het, interessant. Dat zal dan voorbehouden zijn aan de Envoy-S.

Dit artikel beschrijft het een beetje: https://thecomputerperson.wordpress.com/2016/08/03/enphase-envoy-s-data-scraping/

Ik zie nu ook waarom de keuze is gemaakt om de JSON op te halen, er zit een stuk meer data in (en die wordt ook gebruikt bij systemen die een batterij hebben of ook de consumptie van het huis meten). Ik zal dus even een fallback moeten implementeren die `/api/v1/production` benadert.

Ik ben de komende tijd een beetje druk maar ik ga even kijken wanneer ik eraan toe kom. Dank voor het rapporteren!

arnoarnoo, 2021-04-14:
Zou fijn zijn als je eraan toe komt!

Met vriendelijke groet,
Kind regards,

Arno van Oosten

arnoarnoo, 2023-01-20:
Hi Diederik,Ik zag in het changelog op de Homey dat je bezig bent geweest met de Enphase, maar die nieuwe versie nog niet Live hebt gezet. Is er een kans dat de Envoy-S nu ook gaat werken met de 5.1 versie? Ik blijf hopen…Met vriendelijke groet,Arno van Oosten

vierlijner, 2024-08-02:
Nu is de nieuwe versie wel live, werkt het nu?

arnoarnoo, 2024-08-21:
Diederik,In de homey app store staat nog steeds versie 5.3.2 van december 2023?Kind regards,Met vriendelijke groet,Arno van Oosten