DiedB/Homey-SolarPanels #145: Support Sungrow Sh10rt - iSolar-Cloud / Homey Pro

CrazyMelodie, 2021-04-12:
Hello, my english is not very good but I'll try anyway..
I have a Sungrow Sh10rt inverter with 11 kW battery storage. 
I can read out all data via the iSolar cloud app. Is it possible to process these data in my Homey Pro?
I need help with the configuration because a Sungrow photovoltaic system is not suggested in Homey Pro.

Many thanks for the support.
Greetings Frank

DiedB, 2021-04-14:
I had a quick look and it seems that there is no public Sungrow API (https://github.com/kronicd/sungrow2pvoutput/issues/1). For the time being, I'll refrain from implementing it for that reason. Might there be a lot of support we can have a look if this can be worked around, so I'll keep this issue open for others to chime in.

mellbratt, 2024-02-18:
@DiedB There is no public API but a GoLang implementation here https://github.com/MickMake/GoSungrow