DiedB/Homey-SolarPanels #148: Omnik migrated to SolarMan

JorissP, 2021-05-17:
Thanx for maintaining this app. It has been very usefull with a Omnik inverter! As you might know Omnik is bankrupt, the Omnik portal is down and for this reason the Omnik part in the app does not work anymore. However, quite recently they migrated all Omnik data to Solarman. I tried to connect Homey with the 'Zonnepanelen' app with 'Solarman' but it does not work. I think I know why it doesn't work and hopefully this can be relatively easily fixed.

The regular Solarman portal is at: https://solarmanpv.com but the Omnik users have been migrated to: https://home.solarmanpv.com. My best guess would be that if you use the Solarman part of the app for a new Omnik part with reference to the correct address that it would work.

Hope you have some time to look at this. If necessary I would like tot test this for you

deBru1n, 2021-05-20:
Hi, if needed I can also do some testing. Looking forward to get a solution. I have some follow depending on the Solar information

DiedB, 2021-07-24:
Do you have any programming experience? You can try to change the URL in the SolarMAN driver and run the app yourself. Let me know if it works!

The URL is defined in this file:
https://github.com/DiedB/Homey-SolarPanels/blob/master/drivers/solarman/device.js

deBru1n, 2021-07-24:
Dear Diederik,

I am having not much programming experience, but know my way around in computers.
So, if you let me know where this driver is located, I can replace it.

Thank you

Met vriendelijke groet, / Kind Regards,

M. de Bruin

https://github.com/DiedB/Homey-SolarPanels/blob/master/drivers/solarman/device.js

DiedB, 2021-08-18:
I don't think addressing the new SolarMAN portal is going to work, I prefer implementing a local connection. For that, several issue threads are open currently.

closed 2021-08-18 by DiedB