DiedB/Homey-SolarPanels #151: Enphase

robkeemink, 2021-06-08:
After today the app does not regognize my 2 enphase envoy systems anymore

Can you see why?

robkeemink, 2021-06-09:
Ik zie ze allebei wel op ip adres. Ze zijn lange tijd wel gewoon gevonden die de app. Maar na stroomonderbreking was er geen connectie meer. Opnieuw toevoegen geeft: geen nieuwe apparaten gevonden

DiedB, 2021-06-09:
Homey zoekt de Enphase omvormers via mDNS, een lokaal netwerkprotocol voor discovery van hosts. Kan het zijn dat er iets aan je netwerkconfiguratie is veranderd, waardoor Homey in een ander subnet zit dan je Enphase-systemen? In die situatie kun je de systemen prima via hun IP-adres benaderen maar zal Homey ze niet vinden (omdat mDNS niet zonder meer buiten een subnet reikt).

robkeemink, 2021-06-09:
Niets veranderd aan netwerk/subnet. Gewoon een home netwerk met ethernet
(zitten de 2 enphases in) en wifi (Homey). Tot gisteren werkte het wel Na
stroomonderbreking kan ik de enphases niet meer in de homey vinden

DiedB, 2021-06-09:
Hmm, router herstarten?

robkeemink, 2021-06-09:
als eerste al gedaan... er is op de meterkast gisteren een
loadbalancer aangesloten....Voor laadpaal, maar die staat los van netwerk

ik ga wat meer dingen resetten: alvast bedankt voor je snelle reactie....
enne: het heeft maanden fluitend gewerkt

robkeemink, 2021-06-09:
Homey zegt bij toevoegen enphase: geen NIEUW apparaten gevonden.... Kan het
zijn dat de oude instellingen nog ergen in de homey staan?

DiedB, 2021-06-09:
Hmm, je noemde dat ze opeens niet meer werkten, heb je ze ergens in het proces zelf verwijderd?

robkeemink, 2021-06-09:
Gisteren toon de stroom er weer op kwam, werden de enphases niet meer
gevonden.  Ik heb ze toen in de homey verwijderd..... en bij opnieuw
toevoegen: geen nieuwe apparaten gevonden

robkeemink, 2021-06-10:
Hoi Diederik,

Heb je nog tips? Je app vindt de enphases envoy nog steeds niet (wel de
enlighten met api key). Maar ik wil de twee enphases apart monitoren. Veel
gezocht op mDNS: maar daar ook nog geen oplossing gevonden. Zelfde subnet.
Maar melding blijft: er zijn geen nieuwe apparaten gevonden

DiedB, 2021-06-23:
Je zou kunnen kijken (vanaf een computer of laptop) of je mDNS verkeer kan opvangen.

https://www.tobias-erichsen.de/software/zeroconfservicebrowser.html

Komen de Enphases daarin terug?

robkeemink, 2021-06-23:
Hij ziet 1 enphase energy (geen details zichtbaar ??), ik heb 2
enphasekastjes, die eerder wel zichtbaar waren voor je app (nog wel ip
adres)

zie bijlage

DiedB, 2021-06-24:
Bijlages komen niet door als je mail gebruikt, je zult 'm direct even op GitHub moeten posten.

JoostBerk, 2021-07-08:
Hi Diederik,

Ik loop tegen het zelfde probleem aan met mijn Envoy. Gister toegevoegd in Homey, alles werkte perfect. Vanmorgen zag ik dat het apparaat niet beschikbaar is. Ik heb alles nagelopen en gerestart. De Envoy is gewoon beschikbaar en te pingen in mijn netwerk, maar de app ziet hem niet meer.

DiedB, 2021-07-24:
@JoostBerk Zou je eens kunnen kijken wat je binnenkrijgt als je naar `[enphase-ip]/production.json?details=1` gaat?

Heeft de app ooit data opgehaald uit de Enphase? Kun je wellicht achterhalen met Homey Insights door de grafiek te bekijken op de dag dat je de Enphase toevoegde.

JoostBerk, 2021-07-24:
@DiedB Dan krijg ik dit: {"production":[{"type":"inverters","activeCount":5,"readingTime":1627150849,"wNow":20,"whLifetime":417963}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}

Yes de setup heeft precies een dag gewerkt, daarna ga de app aan dat het apparaat niet beschikbaar was. Ik doe het nu direct via you app maar haal de data op via de Enphase API, dat werkt nu al een paar weken vlekkeloos.

DiedB, 2021-07-25:
De app gaat ervan uit dat er een veld `whToday` zit in de JSON output. Die lijkt hier te missen en dat is waar de app op stuk gaat.

Welk model Envoy heb je?

DiedB, 2021-07-25:
Net geprobeerd, wat ik net noemde lijkt niet het probleem te zijn. Dan zit het probleem wellicht in de discovery van het apparaat in Homey. Ik heb wat extra logging toegevoegd aan het Enphase-gedeelte van de app.

Zou je alle Enphase apparaten uit Homey willen verwijderen, de app zelf willen verwijderen, en vervolgens https://homey.app/a/it.diederik.solar/test/ installeren en de Envoy koppelen?

Vervolgens een diagnostic report versturen 10 minuten nadat je hebt gekoppeld, en zo snel mogelijk nadat het stuk gaat. Dan zal ik even kijken waar het probleem uit voortkomt. Alvast bedankt!

closed 2021-10-31 by DiedB