DiedB/Homey-SolarPanels #152: Growatt voltage

gooooba, 2021-06-16:
Hi,

growatt integration works. Is there any way to control the inverter(change the mode AC source, Charge source?) The controls are available on web interface, so I guess it should be possible.

Other thing is that I'm missing is Voltage readings. This could be used for detecting Grit outage for example.

pro-sumer, 2021-06-18:
Not sure I understand your request.

I’m using a flow that triggers if the solar panels have not reported for 10 minutes (after sunrise and before sunset).

Would that be useful? (as a workaround?)

Which inverter do you have?

pro-sumer, 2021-06-20:
Additionally, I don't see any voltages in Growatt's ShinePhone App or in it's API responses (for my inverter), so I'm afraid I won't be able to help with this.