DiedB/Homey-SolarPanels #154: Huawei Sun2000-17ktl-m2

github-selguson-se, 2021-07-23:
Hi,
having issues add my inverter:
Huawei Sun2000-17ktl-m2. Is it not supported? If not, is it possible to add it?
Thanks
Andreas

DiedB, 2021-07-24:
Unfortunately, only the SUN2000L range is supported for now. I haven't found any documentation on connecting to SUN2000 inverters yet. If you do, let me know and I'm happy to implement it.

closed 2021-07-24 by DiedB

github-selguson-se, 2021-08-05:
Can this links help you out?

https://forum.huawei.com/enterprise/en/firmware-v100r001c00spc120/thread/745393-100027?page=2
https://github.com/ccorderor/huawei-sun2000-modbus-mqtt

Thanks.

DiedB, 2021-08-08:
Do you have the first dongle [here](https://solar.huawei.com/en/download?p=%2F-%2Fmedia%2FSolar%2Fattachment%2Fpdf%2Feu%2Fdatasheet%2FSmartDongle.pdf)?

github-selguson-se, 2021-08-08:
Yes, that’s the one connected to my inverter.

Med vänlig hälsning,
Andreas

Skickat från min iPhone

DiedB, 2021-08-09:
And if you execute a port scan, using `nmap -sC -sV -p502 <inverter-ip>` for example, does it show that the ModBUS port is open? Does this change when you change from a wireless to a wired connection (or vice versa)?

github-selguson-se, 2021-08-09:
What I can understand both are opened (cable and WiFi)
See screenshots.

[cid:E37AB7F2-90F5-476F-ADBE-DE0EBDE9B2AA-L0-001][cid:2C997633-B25E-49B4-8FEE-56170274E39F-L0-001]

Med vänlig hälsning,
Andreas

Skickat från min iPhone

DiedB, 2021-08-09:
You need to post the screenshots directly to GitHub, email doesn't work for that.

github-selguson-se, 2021-08-09:
[image]
[image]

DiedB, 2021-08-09:
Hmm, interesting, do you know how to forward ports to the outside? I would like to have a look at why the app is not working then, but I would need to debug the ModBUS connection with your inverter.

github-selguson-se, 2021-08-09:
Yeah, I think I can arrange that.
Port 502?
Your ip so I can open only against you?
Maybe you can send me a private email for exchange of ipaddress.

Med vänlig hälsning,
Andreas

Skickat från min iPhone

DiedB, 2021-08-09:
I will, but I can't access your e-mail address from GitHub. Could you post it here (fine to delete directly afterwards)?

github-selguson-se, 2021-08-09:
github@selguson.se

github-selguson-se, 2021-08-16:
Any updates yet?

DiedB, 2021-08-18:
I have taken a look. Seems like we're experiencing the same issues here, like other users in the other Huawei issue thread. SUN2000L inverters with integrated WiFi seem to work fine, but any inverters that need a smart dongle to connect are not responding to ModBUS packets on most firmware versions (although they do publish the corresponding port).

Do you know what firmware version your smart dongle is running?

[This](https://forum.huawei.com/enterprise/en/sun2000-smart-dongle-modbus-tcp-issues/thread/675885-100027?page=3) thread mentions that SPC122 on the SDongleA-05 should work.

github-selguson-se, 2021-08-18:
The dongle firmware is:
V100R001C00SPC120
I will check if i can upgrade it. I will come back soon.

BR
Andreas

github-selguson-se, 2021-08-18:
I can find the release notes but not the binaries or find them in Fusion Solar Portal for upgrade. So im a bit locked right now.

BR
Andreas

DiedB, 2021-08-18:
[This](https://community.home-assistant.io/t/integration-solar-inverter-huawei-2000l/132350) thread also seems to contain lots of information on firmware and compatibility. Seems like many users experiencing the same issues. Someone also mentions that SPC123 would be released in the next days and that version should provide a fix.

github-selguson-se, 2021-08-18:
I see!
I probably have to wait than.
I atm running SPC120 on donlge and SPC135 on inverter.

BR
Andreas

github-selguson-se, 2021-09-01:
Hi,
patching Dongle with SPC123 at the moment.
Up for testing for you in few minutes.

BR,
Andreas

Skickat från min iPhone

DiedB, 2021-09-01:
Great, curious to see whether this is going to work. Would add support for Huawei to this app for lots of users.

github-selguson-se, 2021-09-07:
Hi, did you figured out something?

Br
Andreas

DiedB, 2022-05-03:
How do you verify that Modbus is working?

blexter, 2022-05-04:
By two different apps on my phone, screenshots below.

[image]

blexter, 2022-05-04:
[image]

blexter, 2022-05-04:
[image]

blexter, 2022-05-04:
[image]