DiedB/Homey-SolarPanels #158: Problems with huawei sun2000

tuomaskilpimaa, 2021-08-18:
I try to connect homey to Huawei inverter, but when I fill the right ip, comes message "error: connect ECONNREFUSED 192.168.xx.xxx:502. What I can do?

DiedB, 2021-08-18:
I would need the following information to help you further:
- What (exact) inverter model do you have?
- Is it connected via WiFi or ethernet?
- Do you use a Smart Dongle (SDongleA-05) to add connectivity to the inverter?
- What firmware versions are running on the inverter and (if applicable) the Smart Dongle?

tuomaskilpimaa, 2021-08-19:
> Lähetetty: keskiviikko 18. elokuuta 2021 21.25
> Vastaanottaja: DiedB/Homey-SolarPanels ***@***.***>
> Kopio: tuomaskilpimaa ***@***.***>; Author ***@***.***>

Hi, answers in blue

  *   What (exact) inverter model do you have? SUN2000-5KTL-M0
  *   Is it connected via WiFi or ethernet? via WiFi
  *   Do you use a Smart Dongle (SDongleA-05) to add connectivity to the inverter? yes SDongleA-05
  *   What firmware versions are running on the inverter and (if applicable) the Smart Dongle? I'm not exactly sure is this the firmware version number, software version V100R001C00SPC133 and dongle V100R001C00SPC109

-

tuomaskilpimaa, 2021-08-19:
And if you need. The homey version is 7.1.1

DiedB, 2021-08-19:
Thanks, that's all I needed! There are several users that do not get this combination working, mainly because the Smart Dongle doesn't behave the way it should on all firmware versions. Do you have a way to upgrade the firmware of the Smart Dongle, maybe by contacting your installer?

A new firmware version for the Smart Dongle will be released soon, ending in `SPC123`, which should solve the problems and that might immediately make the app compatible. There are two other issue threads tracking this, so maybe it's best to watch those until people confirm that things start working with new firmware, and contact your installer after that.

See #154 and #78.

closed 2021-08-19 by DiedB

tuomaskilpimaa, 2021-08-19:
Ok thanks, hopely new firmware solve the problems :)