DiedB/Homey-SolarPanels #163: Connection with eSolar AIO3

hmook, 2021-09-29:
Hi is it posible to connect with my eSolar AIO3 interface? When I try to connect with my SAJ inverter which is connected via eSolar AIO3. The Homey App gives me an error message telling me I've entered a wrong IP address.

drieshooghe, 2021-09-29:
For the WiFi D module, when the inverter isn't producing any energy, the local API goes offline. Perhaps the same goes for the AIO3 module.

DiedB, 2021-10-31:
Does this work with the latest version?

hmook, 2021-11-01:
Sorry for the late reply, I’ll try to connect again later to see if I am able to connect when producing power.

hmook, 2021-11-01:
No seems like also when producing power I cannot connect.

hmook, 2021-11-27:
Kinda found the issue. The AIO3 module doesnt create a webserver to read out locally. Power data is directly pushed to the eSolar Portal at https://fop.saj-electric.com/ . So I cannot connect to the unit locally but have to use the external API. Is this possible in this HomeyApp?

DiedB, 2021-12-01:
The external API is not meant for public use. While it would technically be possible, building and supporting it is difficult, especially because I don't have an account myself.

closed 2022-01-22 by DiedB