DiedB/Homey-SolarPanels #169: App lost after power outage

FonsdeRond, 2021-10-31:
This month there were two power failures in our neighborhood.  Both times the app appeared to be lost. I have to re-install the app and check all the flowcards. As I'm writing this, re-installing is not possible. I'll try later again.
The app should be able to withstand a power outage.
I use the Solax converter on a Homey.

DiedB, 2021-10-31:
Does the Solax Cloud show data? Can you create diagnostic report the next time it happens (before re-pairing)?

FonsdeRond, 2021-11-02:
Diederik, thanks for the swift reply. Without interruption I've been
receiving data on my mobile through an app especially for the Solax
converter. So I conclude data is available in the cloud.

Yesterday evening the tile in Homey reported an error 'Error retrieving
data (Error: this sn did not access!', but in the morning it was
functioning normal. At this very moment it is receiving data. Fingers
crossed.

I'm not sure where to find the diagnostic report. Is it a Homey report?
Of course I hope that it will not occur very soon. Still I'm puzzled why
after a power outage the 'Zonnepalen' app only is deleted from the Homey. I
had 3 outages in the last couple of months and 3 times the app had to be
re-installed.

Fons de Rond
+31 (0) 636 575 676

DiedB, 2021-11-02:
Is the entire app uninstalled, or only the device removed? That is really strange.

Does that error pop up every evening and/or night?

DiedB, 2021-11-02:
You can create a diagnostic report through the Homey Settings screen -> Apps -> Solar Panels -> Send diagnostic report. Only do this when you are experiencing errors, otherwise it is not really useful.

FonsdeRond, 2021-11-02:
The device (the tile) is still there, but in error. The X-symbol (for
Solax) for instance has disappeared. But the app is not visible anymore
(the only one). When I re-install the app it connects to the device and the
flows work without a problem.

Fons de Rond
+31 (0) 636 575 676

DiedB, 2021-11-02:
Ah, that makes more sense and that should be investigated. Will look into it.