DiedB/Homey-SolarPanels #178: Linking Trannergy to Homey doesn't work since portal update

OverlordWolff, 2021-12-07:
When I try to integrate my Trannergy inverter into Homey there is always an error since the Trannergy Portal was updated. I haven't changed anything in my credentials.

I get the message below:
[image]

DiedB, 2021-12-07:
Thanks for the report. Will have to deprecate the driver given that we have no alternative way to connect to Trannergy at this moment.

DiedB, 2021-12-08:
Also, I would remove that screenshot, given that it contains a hash of your password.

OverlordWolff, 2021-12-08:
Thanks! I didn't realize that. Will you coment on this thread when there is any update?

OverlordWolff, 2022-01-18:
> Thanks for the report. Will have to deprecate the driver given that we have no alternative way to connect to Trannergy at this moment.

Is there already a solution?

closed 2023-06-10 by DiedB