DiedB/Homey-SolarPanels #180: 2 solaredge converters in Homey pro

scrub29, 2021-12-18:
HI Is it possible to have two solaredge installed in Homey? i do get an error when I try to install the second one. could not find that pair session.

thanks

DiedB, 2021-12-20:
Could you try the test version of the app? https://homey.app/nl-nl/app/it.diederik.solar/Zonnepanelen/test/

scrub29, 2021-12-20:
Hi Diederik,

Zojuist geprobeerd ik krijg deze melding; 

Klopt het dat ik geen id hoef in te geven? 

Met vriendelijke groet,

Manon en Eugène

DiedB, 2021-12-20:
Zou je het direct op GitHub kunnen posten? Kan het zo niet zien.

Klopt inderdaad dat een ID niet meer nodig is, die wordt automatisch opgehaald.

scrub29, 2021-12-20:
[image]

DiedB, 2021-12-20:
Werkt de eerste inverter wel?

Weet je heel erg zeker dat die API-key klopt? Anders moet je dat even dubbelchecken met je installateur.

scrub29, 2021-12-20:
OKe ik ga morgen weer bellen, dank je

closed 2021-12-20 by DiedB

scrub29, 2021-12-20:
kunnen er wel 2 omvormers op?

DiedB, 2021-12-20:
In principe moet dat geen probleem zijn. Afhankelijk van je installateur zijn het twee aparte 'installaties', dus twee API-keys, of zit alles onder één API-key. Maar als dat zo was had je het gemerkt bij het installeren van het eerste systeem in de nieuwe versie van de app, dan kun je meerdere inverters kiezen.

scrub29, 2021-12-20:
het zijn twee api keys

DiedB, 2021-12-20:
Dan zijn het losse 'installaties', en zal een van de keys niet kloppen. Gaat vaak fout omdat na het genereren van de API-key niet op 'opslaan' wordt geklikt door de installateur.

scrub29, 2021-12-21:
Hoi Diederik,
Het klopte de key was niet opgeslagen! Nu werkt het mijn dank is groot.
Gr Eugene