DiedB/Homey-SolarPanels #181: Growatt

Koymoreno1, 2022-01-18:
Does not work with Growatt SPH 300-600.  Can it be added?

pro-sumer, 2022-01-20:
Unfortunately there's no documented Growatt API.

I implemented support for simple Growatt inverters like my own by investigating the network traffic.

More advanced inverters seem to use a different API/protocol, so we probably need the help of owners of such devices who can add support for them (or at least research the protocol).

Koymoreno1, 2022-01-22:
> @pro-sumer <https://github.com/pro-sumer> Would you be willing to move
> Growatt support into its own app? I'm working on SDK v3 + TypeScript
> migration and with that, I'd like to make the app a little easier to
> support for myself, by only adding (stable) cloud integrations with very
> few exceptions.

Hola,

¿Que debo hacer? No le entiendo muy bien y no se que necesita exactamente.
Sin embargo, le agradezco el interés.

Espero su respuesta, saludos
Juan Moreno

pro-sumer, 2022-01-25:
> @pro-sumer Would you be willing to move Growatt support into its own app? I'm working on SDK v3 + TypeScript migration and with that, I'd like to make the app a little easier to support for myself, by only adding (stable) cloud integrations with very few exceptions.

I understand, but I'm not sure I have the skills/time to do that now.

What other options would we have?

pro-sumer, 2022-03-05:
Your code is released under GPL, so I am allowed to make a personal/private fork that only adds my Growatt inverter?

pro-sumer, 2022-03-05:
What happens if you release this version with less supported inverters to the production channel?

Can people stick to the old version or will they lose this functionality (and Insights data)?

DiedB, 2022-03-07:
That's a good question, and because the effects are yet unknown to me, I am holding off on releasing a new version for now outside of the beta channel. I think I'll implement 'dummy' drivers to replace drivers that are being phased out. These would then give a deprecation message and a warning on the device, but would prevent any strange behaviour and would cause insights data to be retained.

DiedB, 2022-03-07:
> I am allowed to make a personal/private fork that only adds my Growatt inverter?
Absolutely, and if you want to make and release a Growatt driver based on my code, feel free to do so.

MP-ICT-Diensten, 2022-07-10:
I continue with this issue, it's about the Growatt support, it is wordking form my Convertor, but I do also get often errors that the Growatt server is not reachable and then several minutes later it is fine again.  I don't know how often the Solarpanel app queries the Growatt server, but maybe we need to have a larger delay in polling, because Growatt maybe blocks the once who poll to much?

It is just a thought, maybe possible to add the delay als configurable parameter in the device page (advanced settings)?

pro-sumer, 2022-07-15:
I have some Shortcuts that only run every 4 hours during daytime, and even those often fail due to server errors.

Growatt seems to have serious issues with server/API availability...

pro-sumer, 2022-12-01:
Created #222 to fix the Growatt server login issue.

closed 2022-12-11 by DiedB