DiedB/Homey-SolarPanels #185: Support for Sungrow

murreen, 2022-03-04:
Hi
I have a Sungrow inverter.Any plans of adding it to the app?

DiedB, 2022-03-07:
Might be possible, but needs more research. Sungrow seems to support Modbus, but would need to know if this implementation is the same for all inverters. Also, need to know whether there is enough interest to implement this, if it turns out to be possible. Please thumbs up the original issue comment if you need Sungrow support as well and let me know what model of inverter you have.

murreen, 2022-03-12:
That would be great.. I have the SG20RT inverter.

ericedberg, 2022-04-18:
SH10RT and SBR096 battery

Fsahlin, 2022-06-27:
SH6.0RT invertera.

Would be great with with support for this.

bergh-io, 2022-08-25:
Yes, please see if sungrow can get implemented 👍

Greatman61, 2022-08-27:
Double SH10RT and battery

drape76, 2022-08-28:
SH80RT and battery

frebra83, 2022-09-28:
Hi, just got my Sungrow SG12RT installed and need the integration. 
Thanks in advance

perkarom, 2022-10-08:
SH10RT and SBR096 battery

Gnalf, 2022-10-25:
That would be really nice, I have a SG10RT.

megardt, 2022-10-31:
Yes please, SH10RT with battery

haadeess, 2022-11-30:
SH8RT please! Would be very good! Thanks in advance.

Ruuudi, 2023-01-25:
That would be great! SH8RT

jfhallberg, 2023-05-17:
Hi would like it for SH10RT with SBR096 battery

Melectro1, 2023-07-20:
Hello
As written before I have also the interest in Sungrow SG12RT connection

JensA23, 2023-09-29:
Two SG10RT

hAMMERlized, 2024-01-12:
Would appreciate it if you could add support for sungrow inverter. I have implemented mine in home assistant via modbus so far with the following Integration (https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant).

It's all modbus and yaml based. As far as i can see there is not mich difference between the models and batterys.

Any idea when of if it will be supported? Thanks.

burrikido, 2024-02-04:
I am interested, too. 
I have an SG6.0RT inverter

schioth, 2024-02-13:
I am interrested as well. I have Sungrow SH6.0RT-V112 with batteries 9.6KWh from Sungrow

burrikido, 2024-02-13:
Pretty old thread. Not sure if the dev is still up for it :(

mellbratt, 2024-02-18:
SH10RT + SBR battery

erikfriman, 2024-04-06:
SG8.0RT + SGH8.0RT-V112 + SBR battery

Patrikwall, 2024-11-18:
I'm also interested in this, mainly just for reading data, I have personally no need for controlling anything.
SG17RT

dvalfrid, 2024-11-30:
Two SH10RT with battery