DiedB/Homey-SolarPanels #186: Homey Envoy

triple-it, 2022-03-08:
Hi,

Great APP! Where is the Donate button?

Is the following URL always available without installer/enphase password?
http://192.168.178.142/api/v1/production
(Envoy-S Standard)

Because I would like to build some flows when the PV is producing.
E.g. start charging IPADs via Shelly's only when there is SUN.

Currently the Homey Envoy defaults to IP address 169.254.120.1:80. how can we fix/set the IP address in the Homey Envoy app?

Thnx!

DiedB, 2022-03-08:
Thanks for your donation, I really appreciate that!

The app tries to discover the Enphase in your network through mDNS-SD. If I read your issue correctly, I think this process does not go well given your network configuration. Did it work previously, or does it pair and then starts giving errors immediately? Do you have multiple subnets? Could `169.254.120.1` be a router? Don't worry if you don't have answers to the last two questions, but I presume the underlying reason for this process going wrong is quite technical in nature.

triple-it, 2022-03-08:
Hi, 

I will troubleshoot mDNS-SD then. 

It is a simple architecture with a Ziggo ConnectBox as WIFI, and a Cisco ws-3560CX as switch.
The Homey APP will go over Wifi, via the switch towards the Envoy (1 vlan, 1 subnet: 192.168.178.0/24).

I thought mDNS-SD would work standard in that scenario... maybe not.

But I will inspect both devices if they are capable of multicasting or setting some mDNS service-lists
(https://tinyurl.com/4895k8wv for Cisco)

Or I could set the Envoy (via login:installer) also on Wifi?
And see if mDNS discovery works instantaneously then.

It is a new setup, the envoy-S standard was installed yesterday.  I don't even have enlighten access yet! :) 
I read your comment that enlighten version needs developer account, etc.  

But I found some hacking URL's to all the JSON api's, and the installer menus.

But if someone would be routing (e.g. have multiple subnets), 
wouldn't it be nice to be able to input an IP ADDRESS in the app anyway? 
Just like with the enlighten version I saw you are able to input a key and an uid.
[https://github.com/DiedB/Homey-SolarPanels/blob/master/drivers/enphase/pair/pair.html](https://github.com/DiedB/Homey-SolarPanels/blob/master/drivers/enphase/pair/pair.html)

Also some people will give there Envoy a static ip, or reserve a static DHCP lease, and would prefer to enter an ip address instead of relying on discovery?

The end goal for me is just:
- Have an event/trigger "Vermogen werd groter dan 1000", to start loading the IPADs :-)

Thnx for your feedback!

triple-it, 2022-03-08:
Over wifi (from macos), so mDNS-sd is not (directly) the issue?
 
 % dns-sd -B _enphase-envoy._tcp 
Browsing for _enphase-envoy._tcp
DATE: ---Tue 08 Mar 2022---
20:44:36.790  ...STARTING...
Timestamp     A/R    Flags  if Domain               Service Type         Instance Name
20:44:36.792  Add        2   6 local.               _enphase-envoy._tcp. envoy
^C
 % dns-sd -Q envoy.local         
DATE: ---Tue 08 Mar 2022---
20:44:42.059  ...STARTING...
Timestamp     A/R    Flags if Name                          Type  Class   Rdata
20:44:42.061  Add 40000002  6 envoy.local.                  Addr   IN     192.168.178.142
^C
 %

DiedB, 2022-03-08:
Could you try installing https://homey.app/nl-nl/app/it.diederik.solar/Zonnepanelen/test/ on the Homey App Store and pair using that version?

The mDNS-SD discovery should happen on the Homey itself, so I do not see any possible influence from the device used in the pairing process. I would say, monitor it for a while and let me know if any issues occur.

On supporting manual IP address entry - I would like this to be an option, but it would make the app more difficult to use for 99% of its users. In the current Homey SDK, I cannot support both options. mDNS-SD is very robust, as long as your network is configured correctly. Also through static leases, Homey should adapt very quickly when paired with mDNS.

DiedB, 2022-03-08:
Oh, I found a possible explanation for your issue: https://support.enphase.com/s/article/What-does-it-mean-when-the-Envoy-displays-the-IP-address-169-254-120-1

Somehow, the Envoy does not receive a lease from your router, but mDNS seems to reach Homey. I cannot explain that yet, but this is definitely the direction where you would need to search.

triple-it, 2022-03-08:
Ok yes.
But in this case the flow was:
- I had a working mDNS from wifi Macbook
- Delete && Add & Delete .. && Add cycle made the Zonnepanelen-envoy device still resolving to 169.254.120.1
- I manually runned and installed your app on the Homey Pro via Homey CLI, and the mDNS resolving worked there, problem solved.
- After shutting down Homey CLI, I had to reinstall the Zonnepanelen APP on the Homey
- Problem was still solved.

So I would think the mDNS resolving on the Homey was not OK. The systems (router + switch) weren't reloaded whatsoever.
Only a manual Homey Cli install, and/or a Zonnepanelen Driver reinstall fixed the issue.

Brainstorm:
- So I would think the Homey was not able to resolve the mDNS, or had a (very very) old entry? 
- What's the TTL of these kind of mDNS records anyway?
- I do have the Envoy UTP cabled connected (most people have it connected via wifi) that's a difference.

Is there some quick and handy way, to check mDNS on the Homey Pro itself? Or should some "Discovery APP" be written?

triple-it, 2022-03-08:
> Could you try installing https://homey.app/nl-nl/app/it.diederik.solar/Zonnepanelen/test/ on the Homey App Store and pair using that version?
>
> The mDNS-SD discovery should happen on the Homey [...]

Ok, v5.0.22 "Experimental" is installed, and seems to work also.
If you want me to verify some things with this version pls let me know.

If manual and mDNS-SD cannot be supported both, I would go for only mDNS-SD indeed.
Why is that limitation there anyway?

DiedB, 2022-03-09:
> So I would think the Homey was not able to resolve the mDNS, or had a (very very) old entry? 

Could be, hard to check where the issue is as it is within the Homey SDK and not this app.

> What's the TTL of these kind of mDNS records anyway?

You should be able to check in the mDNS traffic the Envoy is sending, but it should be around 2 minutes.

> Why is that limitation there anyway?

This is up to Athom, I think this should be built into the SDK instead of single apps. Athom (rightfully) wants Homey to be user-friendly so I see why they would choose for mDNS here.

If you want to debug this, you could build a simple 'discovery' app which has a very broad discovery strategy. https://apps.developer.homey.app/wireless/wi-fi/discovery

triple-it, 2022-04-10:
Looks like shelly h&t discovery implements an "advanced settings" ip setting. Do they also use the Athom recommended md-dns discovery via the sdk? Because then both configuration paths are possible!

DutchITPro, 2022-12-19:
When I go to https://envoy.local/ in my browser, I get to the login page of my Enphase Gateway (after having accepted an invalid SSL certificate).
Also when I ping from my laptop to envoy.local I get the IP address back.
This works both wired and WiFi. The homey is on the same WiFi network and in the same vlan as my laptop and the homey.

Also tested with Bonjour Browser. This one finds him too
Type: _enphase-envoy._tcp
Name: envoy

I experience the problem with both version 4.7.0 and 5.1.2 (test version). A restart of the homey, reinstall of the app unfortunately does not make a difference.

On the envoy I have version D7.0.88 (5580b1) running.

Any idea?

DutchITPro, 2023-01-29:
> When I go to https://envoy.local/ in my browser, I get to the login page of my Enphase Gateway (after having accepted an invalid SSL certificate). Also when I ping from my laptop to envoy.local I [...]

The issue has been resolved for me. In the end I have no idea what the solution was. Tried again after a few weeks and then it worked in one go. (Still getting an SSL error, but that's a known issue #215

In the meantime, the power of all equipment has been once, that's the only thing I can think of why it works now. I felt I had already restarted all involved components once.
In Dutch we say: Vraag niet hoe het kan, maar geniet ervan.

closed 2023-06-10 by DiedB