DiedB/Homey-SolarPanels #192: SUN2000-8KTL-M1

B3rndBu, 2022-04-11:
Hi, 

I saw that my inverter is not supported at the moment. When I try to connect it it gives an error. Ports not open. Are there any ports that needs to be open for the 2000L version? Maybe I can try that and it works. 

Bernd

nauti23, 2022-04-18:
@DiedB 
Heute habe ich mit einem Sun2000 MKT6l und Batterie auf einem Rapsberry in 5 Minuten EVCC.io installiert, in der Condor Auswahl des huawei, ip Adresse lan Port 502 und alles geht, die solarpanels App sagt da : Port Not Open. Vielleicht ein Hinweis mal in deren Code zu schauen? https://github.com/evcc-io/config Muss ja gehen :)

DiedB, 2022-04-22:
@nauti23 That would indicate a network issue, could you debug that? My implementation and their implementation are practically functionally identical.

nauti23, 2022-04-22:
@DiedB how could that be?
- homey in wlan1
- rapserry in wlan 1
- Huawei connected via lan
- Try to ad with Static internal IP in our App: time out
- try to das with same ip via wizzard Form EVCC: Instant connection 
- 
Where could be the difference?

closed 2023-06-10 by DiedB