DiedB/Homey-SolarPanels #196: Huawei sun2000 - time out when adding inverter

blexter, 2022-05-01:
Hi,
I found out that Modbus tcp had to be activated on the inverter by installer login. I did and Modbus is now working, however the inverter still can’t be joined in your homey app, can we do a “second try” on this?

_Originally posted by @blexter in https://github.com/DiedB/Homey-SolarPanels/issues/154#issuecomment-1113345548_

LTE-STHLM, 2022-05-16:
+1 on this issue. Manually activated Modbus TCP (full) and get time out on pairing. Connection rejected before modbus enabling.

gsommer, 2022-08-16:
Hi, I have the same problem and found out that the modbus tcp port for the internal wifi has changed from 502 to 6607.

It would help if the port would not be hardcoded but changeable in the connection dialog.

Kind Regards,
Gerhard

ludwich, 2022-11-15:
I didn´t get an installer account when they installed it so how can i then open Modbus?

closed 2023-06-10 by DiedB