DiedB/Homey-SolarPanels #201: App crashes since running v7.4.x RC firmware

PeterKawa, 2022-06-20:
Hi Diederik,
First of all thanks for your great app.

I have an issue: The app crashes since Homey is running v7.4.x RC firmware. I use a Growatt inverter. 
It all ran fine since september 2020, so it's probably due to an Athom related change.

I'm not sure it's the RC firmware, but I'm sure it's occurs since v7.4.x 

Diagn.report 4ea25ec7-15d2-48ff-8e5c-f79d24ebb677

Thanks in advance
~Peter

pro-sumer, 2022-06-25:
I also have a Growatt inverter (and wrote the Growatt driver) and for me the App still works with Homey firmware 7.4.2.

I see 7.4.3 is available, but can't update right now.

What firmware version do you use?

And which App? (iOS/Android/web?)

erlendoyen, 2022-06-25:
Hi

Have been getting this error for growatt inverter for the last week or so

[image]

I asked growatt about the api and I got this:

[image]

pro-sumer, 2022-06-25:
Their server has indeed been really unstable lately (I also use their API in an iOS shortcut that creates a graph every 4 hours; it often failed the last few weeks).

erlendoyen, 2022-06-25:
After beeing I contact with support about 2 weeks ago the stability of the shinephone app is better, and they admitted that they have had stability issues, so right now the app is very stable, but the api is not (from homey at least)

PeterKawa, 2022-06-26:
> I also have a Growatt inverter (and wrote the Growatt driver) and for me the App still works with Homey firmware 7.4.2.
> 
> I see 7.4.3 is available, but can't update right now.
> 
> What firmware version do you use?
Thanks for your reply,
v7.4.2 (+ rc's), and v7.4.3 since june 23rd

> And which App? (iOS/Android/web?)
Android Homey bèta, and v6.11.5.956 since june 24th.

The app seems stable now, but the connection fails often.
I read it's the Growatt API and out of your hands.
Thanks for looking into it!

erlendoyen, 2022-07-08:
Growatt reported back that the api shouod be stable.
I have seen less intermittent issues with the connection from homey over the last 3 days

erlendoyen, 2022-07-21:
Hm, the api/server access still seems to be very unstable.
Don't know if anything can be improved in the driver

closed 2022-12-11 by DiedB