DiedB/Homey-SolarPanels #206: Growatt connection not stable

Zakraket, 2022-07-12:
Since a few weeks (can't be more specific) both my Growatt's report unavailable, but it varies during the day, multiple times not available, and then available again. Also not always both at the same time

I changed my Growatt password some time ago, but I have the correct username/password (otherwise should not work at all)
[image]
[image]
[image]

Are there problems with the Growatt services? I'm not experiencing any problems with the data on the Growatt website

pro-sumer, 2022-07-15:
Yes, Growatt seems to have serious issues with server/API availability...

(see other related Growatt issues reported by Solar Panels users)

Zakraket, 2022-07-19:
I could not really find a general forum for this app or Growattusers, you mean reported here on Github?

What would be a solution/workaround?

I'm using my solarpanels to direct screens, but that does not work if the API is down.

pro-sumer, 2022-07-19:
See #181 and #201

Zakraket, 2022-07-19:
Tnx, I'll check these!

Outlook voor Android<https://aka.ms/AAb9ysg> downloaden

pro-sumer, 2022-12-01:
Created #222 to fix the Growatt server login issue.

closed 2022-12-11 by DiedB