DiedB/Homey-SolarPanels #209: Request - Growatt SPH-series

BHylsebeck, 2022-07-29:
Hi 

I hav a request for integration with the SPH-series. I would be very happy if it would be possible to add support for this series. 
API is made. I have a SPH-unit online/cloud connection if I can help. 
I hope it can be made.

pro-sumer, 2022-07-29:
What do you mean with "API is made"?

BHylsebeck, 2022-08-08:
Hi - as I can see growatt has made a api so it is possible to connect to the SPH-series and get data out of the units. 
Is it possible for you to add support for Growatt SHP series in your Growatt APP ?

pro-sumer, 2022-08-09:
Unfortunately, there's no documentation for the Growatt API.

Are you able to collect the API calls for your SPH-unit?

PS: @DiedB is planning to drop Growatt support in the next release, so I'm not sure it makes sense to spend time on this.

BHylsebeck, 2022-08-10:
Hi, Ok, 
How about communication with the Growatt server and get all the reading that way ? I am connected to the Growatt server and gets all data thru APP and Webportal. 
Is it possbile to get all these readings from Growatt server to Homey ? Today I cant thru the Homey Growatt APP.

pro-sumer, 2022-08-12:
My Growatt driver is based on API calls.

If you want to scrape the website you might want to start a separate driver from scratch.

LasseBang, 2022-08-12:
I also need support for the SPH series, 

Found this mabye it can help?
https://growatt.pl/wp-content/uploads/2020/01/Growatt-Server-API-Guide.pdf

pro-sumer, 2022-08-13:
This seems to be documentation for an API for installers, who get a token from Growatt to use this API.

As far as I know Growatt does not hand out such tokens to individuals (though maybe you can try to get one?).

LasseBang, 2022-08-16:
Oh! What about using HA Intergration as base?
https://github.com/home-assistant/core/tree/dev/homeassistant/components/growatt_server

LasseBang, 2022-08-16:
https://github.com/indykoning/PyPi_GrowattServer

pro-sumer, 2022-08-17:
> https://github.com/indykoning/PyPi_GrowattServer

Seems to use the same API as the/my Growatt driver currently uses, but with support for newer devices (TLX, Mix, storage).

Does this library work for you with your SPH-series device?

LasseBang, 2022-08-17:
> > https://github.com/indykoning/PyPi_GrowattServer
> 
> Seems to use the same API as the/my Growatt driver currently uses, but with support for newer devices (TLX, Mix, storage).
> 
> Does this library work for you with your SPH-series device?

Yes work fine :) using it to schedule charge from grid on battery when power is cheap

pro-sumer, 2022-08-18:
Excellent!

Which API calls from this library do you use for that?

LasseBang, 2022-08-18:
Line 72 and down here
https://github.com/indykoning/PyPi_GrowattServer/blob/master/examples/settings_example.py

pro-sumer, 2022-08-18:
So you use `update_mix_inverter_setting`.

Can you get the total energy production for today (so far) (in kWh) and current power (in W) using some of the other "mix" functions? (How?)

(those are the two values that the driver needs to report to the App)

LasseBang, 2022-08-18:
Total solar production today is mix_info['epvToday']

Current W is (its in kW) mix_status['ppv']

LasseBang, 2022-08-19:
I did this to calculate to W current production

calc_pv_total = (float(mix_status['pPv1']) + float(mix_status['pPv2']))*1000

DiedB, 2022-08-19:
@pro-sumer I haven't read the entire thread, but it seems that this (newer?) API does support all Growatt devices that were also supported in the current integration?

In that case I'm inclined to just give it a go in the new Solar Panels app version. Lots of Growatt users and would be nice if it's supported - given that it is stable enough!

pro-sumer, 2022-08-20:
@DiedB That would be great news!

I haven't had much time to look into this (busy with other projects), but it looks like this updated version of the Python Growatt library still uses the same domain name and API calls as currently used in SolarPanels (for "legacy" devices). However, it does add support for newer ("tlx" and "mix"?) devices.

We would have to find out how to detect which kind of inverter is used and what API calls need to be made for getting the energy production for today and the current power for that specific inverter. I can check that the code still works for my "legacy" inverter; @LasseBang seems to be capable to help for the "mix" inverters, and maybe we can find a "tlx" "power user" as well?

Perhaps with some teamwork we can make this work?

LasseBang, 2022-09-23:
Im fresh! :-)

DiedB, 2022-12-12:
@pro-sumer Added Growatt support to the SDK v3 version of this app, could you check whether it works by running it locally?

https://github.com/DiedB/Homey-SolarPanels/tree/sdk-v3

pro-sumer, 2022-12-13:
I'll try to test this later this week/weekend.

pro-sumer, 2022-12-14:
Installed 5.1.2 today (after sunset).

App shows 0 W plus today's production in kWh (OK), but I will have to wait for tomorrow (sunlight) to test current power.

pro-sumer, 2022-12-15:
I did get a single value for current power once, but no updates after that.

Growatt refreshes the data every 5 minutes, but I see you have `interval = 60` in the code. What’s the unit of that?

Does this mean "every hour"? If so, is that a mistake or intentional? (Why?)

DiedB, 2022-12-15:
Should be 60 seconds.

Will have a look at the code to see what goes wrong, and why it doesn't update without giving any errors.

DiedB, 2022-12-15:
@pro-sumer Could you manually send a crash report? Want to check whether the scheduled interval runs.

pro-sumer, 2022-12-15:
Please reconsider accessing the Growatt server every minute, which means 5x as much requests compared to the current implementation.

(it might cause Growatt to get even more defensive about third-party API usage)

DiedB, 2023-06-10:
@pro-sumer It's currently 15 minutes (before it was 60 minutes), I was mistaken.

@BHylsebeck please supply me with Growatt credentials at slower_04_machete@icloud.com so I can see if I can (easily) add support for SPH series.

JBrynaa, 2023-09-20:
Is this issue fixed or???
I can't make heads or tail in this thread.

When I use the growatt homey-app, my Growatt SPH10000TL3 is recognized, but when I open the device in homey I get an error That it is not supported yet.
[image]

When I try to use the solar panels homey-app i get to where I can enter my growatt credentials, but then "no new devices have been found"

Is this a Growatt problem? or a homey problem ??

llBIKSll, 2024-05-05:
I would like to connect to my SPH3600