DiedB/Homey-SolarPanels #214: ‘Vermogen’ empty

Dqsp, 2022-09-05:
Sometimes my Omnik inverter looses his WiFi connection, so the Solarman database isn’t updated. Solar panels work though, as I can see in my P1 app. Tried to make a flow in Homey to warn me when the panel looses connection or doesn’t get any connection at the start of the day. I cannot use 0 or “”. Is there a possibility to add a variable option ‘empty’ or otherwise? Any other suggestion also greatly appreciated!
Thanks for any reaction in advance. Rob

Vlieger200, 2022-09-24:
If your Homey makes direct contact with the Omnik inverter, Advanced Flows gives you the option at the bottom of the box to report a problem/error. (Watch the video of Advanced Flows from Homey.)

If your inverter connects via the Solarman server, the story becomes a lot more difficult because you no longer have direct information.

I monitor my WiFi connections from the inverters (several brands) separately by using the Unifi app and giving a message when a connection is lost.

Dqsp, 2022-09-27:
Thanks for the suggestion. Unify app unfortunately doesn’t work in my homey, due to use of mfa. But would be a great solution.

closed 2022-12-11 by DiedB