DiedB/Homey-SolarPanels #215: Error after adding envoy

castle87, 2022-09-20:
Just installed the app and installed my envoy device. But after the pairing i get the following error when clicking on the device:

Error retreiving data: (FetchError: request to https://192.168.1.79/production.json?details=1 failed, reason: self signed certificate)

Is this a known problem and/or is there a solution for this?

Thanks in advance!!

DiedB, 2022-12-11:
Will be fixing this soon.

castle87, 2022-12-12:
Dont know if you still need the feedback, but i've just tried the new version, getting the following error now:

_An unknown error occurred trying to fetch the inverter data_

Tried several things like a fresh installation after reboot.

DiedB, 2022-12-12:
Did it work before the update?

castle87, 2022-12-12:
No it didnt, still had the error as described earlier

DiedB, 2022-12-12:
Alright, will get to it some time this month. Needs quite some changes to get this fixed, as your Envoy likely needs authentication to get inverter data.

castle87, 2022-12-12:
Thanks, let me know if you need any additional information or testing.

Dites33, 2022-12-14:
Available to help too :) (dites33 here)

JohanvanZanten, 2022-12-21:
the local connection to the Envoy aka IQ Gateway is now token based; a token has to be requested at https://entrez.enphaseenergy.com after which it will allow retrieval of data more info here: https://support.enphase.com/s/question/0D53m00008V8Bs4CAF/token-authentification-local-api-with-curl

Dites33, 2022-12-21:
> the local connection to the Envoy aka IQ Gateway is now token based; a token has to be requested at https://entrez.enphaseenergy.com after which it will allow retrieval of data more info here: https://support.enphase.com/s/question/0D53m00008V8Bs4CAF/token-authentification-local-api-with-curl

Ok so i asked for the token, and tonight i will add it my local Gateway.

Dites33, 2022-12-22:
> the local connection to the Envoy aka IQ Gateway is now token based; a token has to be requested at https://entrez.enphaseenergy.com after which it will allow retrieval of data more info here: https://support.enphase.com/s/question/0D53m00008V8Bs4CAF/token-authentification-local-api-with-curl

I requested the token. I added it in local.envoy using browser. Data is here BUT nothing changed using the app.
So I understand that in fact your comment was addressed more to the developer of the Solar Panels app rather than us, the end users since for me it does not change the error message.

JohanvanZanten, 2022-12-22:
@Dites33 
Oh yeah, this needs to be incorporated in the setup and communication between the app / homey and the Envoy

lowiess, 2023-01-04:
Hello i have the same issue here keep up the good work and hope you can find a fix

DiedB, 2023-01-27:
Will work on fixing this integration as soon as possible.

Slofware, 2023-02-02:
Got it working in Python

https://community.homey.app/t/enphase-zonnepanelen-toevoegen/9609/69?u=slof

ketra, 2023-02-10:
because i could not find a working solution for my enphase V7, I have created an app to test if i could get it to work, if someone wants this can be used as a reference to add it to the solar panels app https://github.com/ketra/nl.ketra.enphase.local

Slofware, 2023-02-14:
Hello Kevin, Got it installed and running, i am getting data but as it is night now I will check it tomorrow. Met vriendelijke groeten, With kind regards, Mark Slof Software engineer Tel: +31(0)634683580 Mail: ***@***.*** ***@***.***) www: www.Slofware.nl (http://www.slofware.nl/)

Slofware, 2023-02-17:
Got it running in a test version now

DiedB, 2023-03-04:
Great effort! Will integrate it into the test version of this app. 

Two questions (@ketra):
- Can we get the serial from an API endpoint somehow? Would be more user friendly to do it that way.
- Can we find the Envoy on the network through mDNS, just like in the old version?

Can anyone port forward their V7 Enphase for a while to the outside, and send me IP address, port and credentials at slower_04_machete@icloud.com so I can test the integration?

ketra, 2023-03-04:
Yes, I believe mDNS is still working like before. 

I looked into the getting the serial, and the https://envoy/info.xml endpoint does not require authentication and does contain the serial of the gateway.

I have done some more testing and investigation the last week. Most old URL's are still available just hidden behind authentication, this can be either a bearer token and/or cookie authentication.

rv39, 2023-03-12:
hello, I just tried to connect my new solar system (Enphase envoy) et get this message. Any help ?
[image]

DeWoltie, 2023-04-01:
Encounter the same error. Is there progress on this issue with the local token: https://enphase.com/download/accessing-iq-gateway-local-apis-or-local-ui-token-based-authentication. Thanks

niekvugteveen, 2023-05-03:
Possibly this line has to be added to the code:

process.env['NODE_TLS_REJECT_UNAUTHORIZED'] = 0;

It's a really ugly solution, but it works.

lowiess, 2023-05-04:
> Possibly this line has to be added to the code:
> 
> process.env['NODE_TLS_REJECT_UNAUTHORIZED'] = 0;
> 
> It's a really ugly solution, but it works.

Where must this be added, in a script to read data form the envoy gateway ?

DiedB, 2023-06-17:
If anyone wants to try, support for newer Envoy firmware is on the branch `feature/envoy-v7`. You can install it on your Homey through the Homey CLI (only recommended for power users). Feedback is appreciated.

DiedB, 2023-07-03:
I have pushed this as a test version to the Homey app store, please try it out and let me know if it works for you.

https://homey.app/a/it.diederik.solar/test/

Remove your old Envoy device from Homey and use the 'Enphase Envoy v7' device while pairing.

DeWoltie, 2023-07-03:
It is working!

JohanvanZanten, 2023-07-03:
see data coming in 🤪

DiedB, 2023-07-03:
Thanks! Will make some time soon to get consumption metrics (for metered Envoy) in as well.

mikkelnl, 2023-07-04:
> I have pushed this as a test version to the Homey app store, please try it out and let me know if it works for you.
> 
> https://homey.app/a/it.diederik.solar/test/
> 
> Remove your old Envoy device from Homey and use the 'Enphase Envoy v7' device while pairing.

I tried the test version, thanks! Only issue I seem to have is that 'Energieproductie (vandaag)' is showing my lifetime kWh's instead of 'today'... ;)

mikkelnl, 2023-07-05:
> I tried the test version, thanks! Only issue I seem to have is that 'Energieproductie (vandaag)' is showing my lifetime kWh's instead of 'today'... ;)

This seems to work now somehow :)

castle87, 2023-07-05:
Got it working as well! Thanks!

DiedB, 2023-07-09:
Does someone have an Envoy with CTs (consumption data)?

Could you post the output you get when requesting `/ivp/meters` and `/ivp/meters/readings`?

JohanvanZanten, 2023-07-09:
/ivp/meters
[
    {
        "eid": 704643328,
        "state": "enabled",
        "measurementType": "production",
        "phaseMode": "three",
        "phaseCount": 3,
        "meteringStatus": "normal",
        "statusFlags": []
    },
    {
        "eid": 704643584,
        "state": "enabled",
        "measurementType": "net-consumption",
        "phaseMode": "three",
        "phaseCount": 3,
        "meteringStatus": "normal",
        "statusFlags": []
    }
]

/ivp/meters/readings

[

    {

        "eid": 704643328,

        "timestamp": 1688895561,

        "actEnergyDlvd": 16248067.788,

        "actEnergyRcvd": 28585.537,

        "apparentEnergy": 17685426.697,

        "reactEnergyLagg": 2066133.228,

        "reactEnergyLead": 6852.781,

        "instantaneousDemand": 18123.302,

        "activePower": 18123.302,

        "apparentPower": 18156.911,

        "reactivePower": 364.249,

        "pwrFactor": 0.998,

        "voltage": 729.374,

        "current": 74.610,

        "freq": 50.000,

        "channels": [

            {

                "eid": 1778385169,

                "timestamp": 1688895561,

                "actEnergyDlvd": 5427218.068,

                "actEnergyRcvd": 6398.427,

                "apparentEnergy": 5891489.910,

                "reactEnergyLagg": 668995.748,

                "reactEnergyLead": 1052.575,

                "instantaneousDemand": 6057.837,

                "activePower": 6057.837,

                "apparentPower": 6069.339,

                "reactivePower": 115.796,

                "pwrFactor": 0.998,

                "voltage": 243.143,

                "current": 24.938,

                "freq": 50.000

            },

            {

                "eid": 1778385170,

                "timestamp": 1688895561,

                "actEnergyDlvd": 5407100.028,

                "actEnergyRcvd": 13017.370,

                "apparentEnergy": 5941242.942,

                "reactEnergyLagg": 699435.983,

                "reactEnergyLead": 4966.046,

                "instantaneousDemand": 6021.940,

                "activePower": 6021.940,

                "apparentPower": 6034.108,

                "reactivePower": 103.426,

                "pwrFactor": 0.998,

                "voltage": 242.673,

                "current": 24.841,

                "freq": 50.000

            },

            {

                "eid": 1778385171,

                "timestamp": 1688895561,

                "actEnergyDlvd": 5413749.692,

                "actEnergyRcvd": 9169.740,

                "apparentEnergy": 5852693.846,

                "reactEnergyLagg": 697701.496,

                "reactEnergyLead": 834.161,

                "instantaneousDemand": 6043.525,

                "activePower": 6043.525,

                "apparentPower": 6053.464,

                "reactivePower": 145.027,

                "pwrFactor": 0.999,

                "voltage": 243.558,

                "current": 24.830,

                "freq": 50.000

            }

        ]

    },

    {

        "eid": 704643584,

        "timestamp": 1688895561,

        "actEnergyDlvd": 12587637.679,

        "actEnergyRcvd": 10815578.444,

        "apparentEnergy": 26399363.011,

        "reactEnergyLagg": 482947.500,

        "reactEnergyLead": 5756960.580,

        "instantaneousDemand": -16058.550,

        "activePower": -16058.550,

        "apparentPower": 16234.845,

        "reactivePower": -1439.737,

        "pwrFactor": -0.990,

        "voltage": 729.473,

        "current": 66.681,

        "freq": 50.000,

        "channels": [

            {

                "eid": 1778385425,

                "timestamp": 1688895561,

                "actEnergyDlvd": 4539949.131,

                "actEnergyRcvd": 3483946.176,

                "apparentEnergy": 9209790.462,

                "reactEnergyLagg": 122687.139,

                "reactEnergyLead": 1785913.042,

                "instantaneousDemand": -5292.839,

                "activePower": -5292.839,

                "apparentPower": 5392.455,

                "reactivePower": -559.792,

                "pwrFactor": -0.982,

                "voltage": 242.958,

                "current": 22.159,

                "freq": 50.000

            },

            {

                "eid": 1778385426,

                "timestamp": 1688895561,

                "actEnergyDlvd": 6194166.905,

                "actEnergyRcvd": 2864618.848,

                "apparentEnergy": 10320444.219,

                "reactEnergyLagg": 61442.041,

                "reactEnergyLead": 3173718.375,

                "instantaneousDemand": -4734.602,

                "activePower": -4734.602,

                "apparentPower": 4802.023,

                "reactivePower": -583.425,

                "pwrFactor": -0.986,

                "voltage": 242.798,

                "current": 19.762,

                "freq": 50.000

            },

            {

                "eid": 1778385427,

                "timestamp": 1688895561,

                "actEnergyDlvd": 1853521.644,

                "actEnergyRcvd": 4467013.420,

                "apparentEnergy": 6869128.330,

                "reactEnergyLagg": 298818.319,

                "reactEnergyLead": 797329.163,

                "instantaneousDemand": -6031.110,

                "activePower": -6031.110,

                "apparentPower": 6040.368,

                "reactivePower": -296.520,

                "pwrFactor": -0.999,

                "voltage": 243.717,

                "current": 24.760,

                "freq": 50.000

            }

        ]

    }

]

DiedB, 2023-07-09:
@JohanvanZanten Thanks! Could you update to 5.2.4 and see if consumption (grid and self-consumption) show up?

JohanvanZanten, 2023-07-09:
I got an invalid capability: measure_power.consumption

DiedB, 2023-07-09:
@JohanvanZanten Weird, could you restart the app? And if it happens again, submit a crash report?

JohanvanZanten, 2023-07-09:
I removed the enphase, app and reinstalled both, although the device is now named Enohase IQ gateway instead of the Enphase Envoy v7, could that be an interference?

DiedB, 2023-07-09:
That's correct and that should not cause the interference.

If you have a crash report that would be great!

SiskoBen, 2023-07-10:
Hi @DiedB ,
The new test version 5.2.4 works and gives me now also the "Power (self-consumption)" and "Power (grid)" measurements along side the "Power (production)" measurement and "Daily energy yield" meter that were already there before. (note that i had to refresh the web pages on my PC to get the new measurements shown for the envoy device but that is a Homey problem not of your solarpanels APP)
Thanks very much!
Best regards,
Arjan

ldikmans, 2023-08-13:
> I have pushed this as a test version to the Homey app store, please try it out and let me know if it works for you.
> 
> https://homey.app/a/it.diederik.solar/test/
> 
> Remove your old Envoy device from Homey and use the 'Enphase Envoy v7' device while pairing.

I have tried this approach, but when I try to add a device, it tells me it can't find any new devices. 

Zonnepanelen version: v5.2.6 experimental
Envoy software version: D7.6.175 (f79c8d)  software build on 22 Jun, 2023 12:43 PM

Adding the device: 
1. Select "Enphase IQ Gateway", I don't see an option "Enphse envoy v7".  
2. Enter username/password of the system owner (which is correct, because I can connect locally)
3. After 4 seconds "gegevens controleren"  it comes back with the message "Er zijn geen nieuwe apparaten gevonden" (no new devices found)

any thoughts?