DiedB/Homey-SolarPanels #217: Growatt MAX 50KTL-XL

elgen1981, 2022-10-30:
Hi!
Can you please add support for this inverter? 
Growatt MAX 50ktl-Xl     
Your app looks Awesome, so I (and others) would be very happy if this is possible.

pro-sumer, 2022-10-30:
Please run the HomeyScript listed in #216 and post the result (here).

elgen1981, 2022-10-31:
Thank you for responding.

Returned:
[
  "QRH2BJH00D: inverter"
]

pro-sumer, 2022-10-31:
The type "inverter" is the same as mine and should be supported...

What's going wrong for you? (Can you add the inverter? Do you see production data?)

elgen1981, 2022-10-31:
Ohh.. It adds ok, and seems perfect (with data) for a few minutes, then it goes red. Error retrieving data (Error: could not login to Growatt server)

User pass, checked.. I was able to add the device.. So why wont it work?

Alesschen, 2022-11-01:
Same issue here: unable to login to Growatt server. Showing red icon on device.

martijngeerts, 2022-11-01:
> Ohh.. It adds ok, and seems perfect (with data) for a few minutes, then it goes red. Error retrieving data (Error: could not login to Growatt server) User pass, checked.. I was able to add the device.. So why wont it work?
> […](#)

Same problem here.

erlendoyen, 2022-11-02:
Same issue here.

pro-sumer, 2022-11-02:
The login error does not mean that your model is not supported:

Unfortunately Growatt's server performance is really bad lately...

Their App (which uses the same server) seems to perform better, so maybe we should investigate why?

erlendoyen, 2022-11-02:
It boils to growatt doing what they can to stop unsupported methods for accessing data. 
They do not respond to emails anymore. 

Some larger 3 party vendors get access with official api keys and access like tibber her in Norway, it's showing stable realtime data in the tibber app.

Didiervu91, 2022-11-03:
Can we help in simulating the connections of the app to see what would be the problem/difference?

pro-sumer, 2022-11-03:
It would indeed be interesting to see which API calls this "tibber" App makes.

erlendoyen, 2022-11-04:
> It would indeed be interesting to see which API calls this "tibber" App makes.

The request most likely goes from Tibber backend>>growatt.

I have updated doc for the growatt api, from growatt but without a working api key you don't get any data. 

Alternatives to getting the data from growatt

Proxy
https://github.com/johanmeijer/grott/wiki

Or custom firmware for the wifi model 
https://github.com/otti/Growatt_ShineWiFi-S

pro-sumer, 2022-11-04:
These alternatives both get data from the inverter itself, not from the Growatt website:

> The purpose of Grott is to read, parse and forward the raw metrics as they are sent to Growatt servers. This means other applications can consume the raw Growatt metrics without relying on the Growatt API and servers and without delay.

> It does not communicate to Growatt Cloud at all.

That does not seem feasible for the Homey SolarPanels App, but maybe @DiedB can say more about that?

I don't want to try/support either of those solutions myself.

It's sad that Growatt servers respond so bad to third-party API requests (like Homey and iOS Shortcuts).

erlendoyen, 2022-11-12:
Possible solution https://github.com/home-assistant/core/issues/81951

pro-sumer, 2022-11-12:
> Possible solution

Interesting! Will try to investigate later.

But isn't that about being totally blocked out? (I suffer from random server issues, but I'm able to log in now and then)

pro-sumer, 2022-11-12:
Read all the comments. Seems like the suggested fix is using a fake User-Agent header?

I already experimented with that myself not that long ago (in my iOS shortcut; not in the SolarPanels App), but it did not help...

erlendoyen, 2022-11-13:
yes, i suspect it will be a constant development from growatt to block it
but rendering elements/values from the web site itself should always be possible as long as they allow login from any browser and not just the app
i will be testing the proxy solution next moth, of course it's not as easy to setup.

koenvanmil, 2022-11-17:
In the HA topic mentioned above their fix still seems to work 5 hours ago? Would it be possible following the same path / solution to use as a fix? Maybe the developer there even can explain you some more?

pro-sumer, 2022-11-17:
I have experimented with the User-Agent from the ShinePhone App today and I'm happy to report that I have had no login error today!

However, using a fixed User-Agent might require too many App/driver updates in the future, so I want to investigate whether I can make this a setting.

Thank you @erlendoyen for pointing me to https://github.com/home-assistant/core/issues/81951!

koenvanmil, 2022-11-17:
Maybe do a one time update now in the meanwhile with a hardcoded user-agent? Or should we just be a bit more patient?

pro-sumer, 2022-11-17:
Please be a bit more patient.

I have made the User-Agent an Advanced Setting, but it's already dark here, so I need to wait until tomorrow for further testing.

Additionally, I will need some help from @DiedB before I can make a Pull Request; I need to figure out how to (also) access this setting in `driver.js` (for logging in the first time, when a new user adds an inverter to the Homey App).

koenvanmil, 2022-11-17:
I’m very sorry. Not trying to push, just thinking along. Please don’t feel pressured. And thanks for the effort already, it is really appreciated

Ronflon, 2022-11-21:
Same issue here. Hope it can be fixed. Thanks for your effort so far!

pro-sumer, 2022-11-21:
Testing a different approach today... Seems to work so far!

Wouwmw, 2022-11-22:
I follow the comments

Wouwmw, 2022-11-29:
Is there a progress for an update? I have no connection with the growatt server for some two weeks. 
Thanks for solving the problem.

pro-sumer, 2022-11-29:
Unfortunately I did not have much time to look into this, but the few approaches I tried all seem to work.

I now need to decide which solution I prefer and then find the time to clean up the code and submit a Pull Request.

pro-sumer, 2022-12-01:
Decided to try KISS ("Keep It Simple, Stupid") first...

Created #222 to fix the Growatt server login issue.

DiedB, 2022-12-11:
Is going to be fixed in the next update, thanks @pro-sumer!

closed 2022-12-11 by DiedB

Ronflon, 2022-12-12:
Hi, I've installeren the update, but it fails with that can be seen at the screenshot attached. 

Can you please fix this?  If you need more info or additional logging info,  please let me know. 

[image]

pro-sumer, 2022-12-13:
@Ronflon Can you please restart the SolarPanels App and try again?