DiedB/Homey-SolarPanels #218: Homey Pro 2023

Henk-Darkw00d, 2022-11-05:
Hi,

Any chance that there will be a version for the Homey Pro 2023?
That would be awesome.
Thanks!

ArjenEV6, 2022-11-06:
I like to support this request.

stortkoker, 2022-11-08:
A test version is allready availble in the app store, haven't tried it myself though.

ArjenEV6, 2022-11-09:
Yeah I guess we've all seen that and also that it doesn't work. Therefore these requests. Meanwhile patiently waiting for upcoming news and updates. Thx for  pointing it out though.

ludwich, 2022-11-11:
Will the new version support Huawei?

DiedB, 2022-12-11:
https://homey.app/nl-nl/app/it.diederik.solar/Zonnepanelen/test/

Not all drivers will be ported to SDKv3 however. Some are constantly breaking and require too much support from my side.

closed 2022-12-11 by DiedB