DiedB/Homey-SolarPanels #219: Can't login anymore. Need to specify country

meer1963, 2022-11-12:
Login from growatt changed. They are now asking for your country. 
I can't login anymore.

pro-sumer, 2022-11-12:
> They are now asking for your country.

Where do you see that?

meer1963, 2022-11-13:
Hoi, als ik via de Growatt app inlog krijg ik tegenwoordig dit inlog
screen.

Met tegenwoordig een land erbij.

Groeten Willem.

pro-sumer, 2022-11-13:
I don't see a screenshot of a login screen?

However, this morning I have been experimenting with my Growatt iOS shortcut (for another reason) and made a lot of login calls, using username and password only. These all succeeded (no country needed).

meer1963, 2022-11-13:
Cool, thanks!

Wouwmw, 2022-11-15:
I have the same problem since three days. After restart the app it goes to the Brocken app.

Code of homey for the developper.
cdd9cab9-39aa-4df2-b1eb-86213d325ccf

pro-sumer, 2022-11-15:
Where/how do you log in in the App? And what is the "Brocken" App?

I'm still logged in, but I did not get any data the last 4 days it seems... 😢

(hope to debug this later this week or next weekend)

Wouwmw, 2022-11-15:
> Where/how do you log in in the App? And what is the "Brocken" App?
> 
> I'm still logged in, but I did not get any data the last 4 days it seems... 😢
> 
> (hope to debug this later this week or next weekend)

The app zonnepanelen for homey. I have log in with my credentials. Name and password. The app works. But after a minute it is broken. 
It is not available.  Error retrieving data. Could not log in to growatt server.

pro-sumer, 2022-11-15:
But when does something (what?) ask for your country?

Wouwmw, 2022-11-15:
It is not asking for a country. I have make a diagnostic report for you.
cdd9cab9-39aa-4df2-b1eb-86213d325ccf

meer1963, 2022-11-15:
I logged out, and now i'm not able to log in again.
I always get: not able to login.
I have tried an other app called Growatt. Which allows me to login with the
same password and name I use for the zonnepanelen app. But that one does
not work for me.
Please fix the app.

Groeten Willem

pro-sumer, 2022-11-15:
@Wouwmw Then you confused me by writing `I have the same problem`. You mean you have (generic) login issue as everyone, but not anything country specific, which is the topic of this particular GitHub issue? 

@meer1963 What is that other app called "Growatt"? Can you please share a link to it?

PS: Since I'm not the developer of this App, but only wrote the Growatt driver I can't look into diagnostic reports.

pro-sumer, 2022-11-15:
@DiedB Which version of Node do you use for development?

I currently have v19.0.1 installed on my Mac, but then the "prebuild-install" of some serial port related code seems to fail...

Homebrew refuses to install Node v12 (stil used by Homey?), because it is not supported upstream.

As a result I cannot perform `homey app run` to debug this issue... 😢 Help?!