DiedB/Homey-SolarPanels #220: Can't login to Growatt server with Homey zonnepanelen app

meer1963, 2022-11-18:
I can login with the official shinepro app. That's not from Homey. On Homey there is an app called Growatt, with this I can also login. But my device is not supported.
With the zonnepanelen app it was working fine. But for a few days now I can not login anymore. 
Anyone got an idea?

pro-sumer, 2022-11-18:
Please don't submit [the same issue](https://github.com/DiedB/Homey-SolarPanels/issues/219#event-7844621010) multiple times.

pro-sumer, 2022-11-18:
Additionally, see #217 for the current status/progress.

pro-sumer, 2022-12-01:
Created #222 to fix the Growatt server login issue.

DiedB, 2022-12-11:
Will push an update today! Thanks @pro-sumer

closed 2022-12-11 by DiedB

meer1963, 2022-12-12:
Thank you very much! It's working again.

john1v, 2022-12-13:
Thanks, it is working here, too.

RobinBol, 2022-12-13:
Thanks for the fix! Working again :)