DiedB/Homey-SolarPanels #235: Enphase Enlighten v4 API

JohanvanZanten, 2022-12-18:
the v2 of the Elighten API will expire jan 18th, the new v4 is already live but entitles an entire;y different concept with paid plans, although there will be a free plan (but with the least features)

will you support the v4 API ?

ferrydl, 2023-01-04:
There is also a local API, which off course is free to acces. I would really like it if it is possible to connect to that. The local API is: http://[local-ip]/api/v1/production/.
The resulat are:
```
{
  "wattHoursToday": 1852,
  "wattHoursSevenDays": 16777,
  "wattHoursLifetime": 24426931,
  "wattsNow": 568
}
```
and it is updated about every 5 minutes.

Slofware, 2023-01-27:
Tryed that with a webhook and I got these error:
request to https://X.X.X.X/api/v1/production/ failed, reason: self signed certificate
Same as when you add a Enphase Envoy.
Is that a Homey thing or a Enphase thing.

DiedB, 2023-01-27:
Will work on this as soon as I can find the time. @ferrydl the local API is covered in the Enphase Envoy integration of this app for two years already. However, Enphase changed things there as well and the app needs to adapt. @Slofware you are also refering to Envoy, if you have any comments there please add them to issue #215.

DiedB, 2023-03-04:
Enlighten V4 goes back to 1000 hits a month, which would mean 2-3 updates per hour if we make it smart (no requests at night).

I don't have an Enphase system myself, is there anyone who could share credentials with me so I can update the implementation? You can send me an email at slower_04_machete@icloud.com.

RoadXY, 2023-03-04:
I've send you my credentials.

I'm also a Homey developer (though not as good 😊) and a IT admin.

Let me know if/where I can help

KnarfOfSpam, 2023-05-30:
Do not know if anyone seen this but here goes from the Enphase Forum (Sorry don't know how to deep link):

3 of 907

[alexbk66](https://support.enphase.com/s/profile/0052G00000AAnKVQA1)

Edited May 10, 2023 at 8:37 PM
Not sure if it was mentioned before, but Enphase finally released a new brief which includes the "API" to get he token. Which is a great news as they now made their "API" official!

Quote:

"Accessing IQ Gateway local APIs or local UI with token-based authentication"

For users who want to retrieve a token programmatically, Enphase provides a URL to retrieve a token and mentions the duration for which the token will be valid.

The token is valid only for one year to ensure safety and you must generate a new token upon expiry. The following table lists a few local REST APIs of IQ Gateway.

 

https://click.email.enphase.com/?qs=787eba345119af180f82c885585570bddc60a3d49a9ff9f55248059addae0c9c440e87fc1f9cbfd60eabe6154d4f0488cf5be0c48bb8c133

[Tech Brief-IQ Gateway Local API or UI Access using Token-EN-US-12-14-2022.pdf](https://github.com/DiedB/Homey-SolarPanels/files/11598543/Tech.Brief-IQ.Gateway.Local.API.or.UI.Access.using.Token-EN-US-12-14-2022.pdf)

DiedB, 2023-06-10:
Can someone help me with such a token and remote access to an Envoy (through port forwarding or Wireguard for example)? Hit me up at slower_04_machete@icloud.com

RoadXY, 2023-06-10:
I can help you with that.

DiedB, 2023-06-10:
For reference: https://community.home-assistant.io/t/enphase-local-api-with-firmware-7-x-my-setup/563828/12

SiskoBen, 2023-06-14:
Hi @DiedB,
I have been working on this issue as well and gave up on a Homey App as development/debugging on that is not working for me. I have however managed to get a Javascript proof of concept working for Envoys with the D7 firmware (currently hardcoded local envoy IP address [which your solar panels app can get from the mDNS discovery], username and password [to the Enphase Enlighten account used to get a authentication token for the local envoy which the homey app could ask to the user].
I can send you the javascript code i have for you to use to develop/incorporate in your solar panels app. Let me know if you want to look at my javascript code?
Best regards/Groetjes,
Arjan

DiedB, 2023-06-14:
Hi Arjan! Great work, but I sorted it myself already :) Didn't take too much time luckily!

SiskoBen, 2023-06-14:
Hi @DiedB ,
Great! I hope to see the new (test) app soon ;-)

DiedB, 2023-06-17:
If anyone wants to try, support for newer Envoy firmware is on the branch `feature/envoy-v7`. You can install it on your Homey through the Homey CLI (only recommended for power users). Feedback is appreciated.

SiskoBen, 2023-06-17:
> If anyone wants to try, support for newer Envoy firmware is on the branch `feature/envoy-v7`. You can install it on your Homey through the Homey CLI (only recommended for power users). Feedback is appreciated.

Hi Diederik,
I don't know how the Homey CLI works but I looked at the code. Is it true to say that the code does read only the  data in "production.json" (.wNow if relevant/active) and not the data from each individual microinverter data (.lastReportWatts from "api/v1/production/inverters/")?

DiedB, 2023-06-17:
Currently, yes. The app does not record data on a micro-inverter level, only the total power/energy for an entire plant. This was also the case for the previous Envoy implementation.

JohanvanZanten, 2023-06-17:
[image0.png]

DiedB, 2023-06-17:
I cannot see the image, uploading images only works directly on GitHub.

JohanvanZanten, 2023-06-17:
Sorry, my bad ;) 

here you go 
[image]

DiedB, 2023-06-17:
@JohanvanZanten Should be fixed, please pull and run the app again.

JohanvanZanten, 2023-06-18:
Hi Diederik,

the previous error is now gone but, this is always a nice one: "An unknown error occurred while fetching inverter data", is there any way of logging available ?

grtx Johan

ketra, 2023-06-20:
@DiedB @JohanvanZanten , i checked the code, and there is a 401 unauthorized error, even when the Token is received, This is because there also needs to be an authenticated cookie in the session before its possible to request data from the production.json endpoint.

JohanvanZanten, 2023-06-26:
were there any changes as I still have the same 
[image]

WouterAar, 2023-06-27:
Hoi Diederik,

Ik heb ook het probleem dat ik met jou top apps, ni t kan connecten ivm benodigde API Key, heb op internet volgende bij enphase gevonden voor ontwikkelaars van apps zoals jij. Mede ivm API.

Heb je hier iets aan?
Zal onderstaande de schermafbeeldingen toevoegen.

Laat maar even weten of ik je verder kan helpen? 

Gr wout, 50 panels enphase installation with IQ7+ omvormers.

WouterAar, 2023-06-27:
[image]

WouterAar, 2023-06-27:
[image]

WouterAar, 2023-06-27:
[image]

WouterAar, 2023-06-27:
[image]

WouterAar, 2023-06-27:
Bovenstaande is de tekst en toelichting voor app ontwikkelaars in relatie tot API

WouterAar, 2023-06-27:
@DiedB 

User id heb ik als systeem eigenaar wel, maar hij vraagt ook om de API key als ik je app envoy enlighten installeer en open. Die API heb ik niet, komt volgens mij omdat er als ik teksten enphase lees hierboven, in de software eea ingebouwd moet zijn ivm API, incl goedkeuring en aanmelding vd app door de programeur bij enphase.

WouterAar, 2023-06-27:
Als ik je app installeer op homey pro early 2019 Dan krijg ik onderstaande melding op scherm, denk dat jij dit beter snapt dat ik.
[image]

ketra, 2023-06-27:
@WouterAar deze melding krijg je omdat de huidige app in de store nog niet kan praten met de nieuwe firmware van de enphase gateway. hier wordt aan gewerkt.

WouterAar, 2023-06-27:
> @WouterAar deze melding krijg je omdat de huidige app in de store nog niet kan praten met de nieuwe firmware van de enphase gateway. hier wordt aan gewerkt.

Ok, maar na installatie vraagt app ook om API key en die kun je als systeem eigenaar nergens in de enphase envoy Metered s gateway vinden, nemen jullie dat ook mee?

Gr wout

WouterAar, 2023-06-27:
Overigens gebruikte ik tot voor kort jullie app enphase envoy en die vraagt niet om user ID en API KEY zoals jullie app enphase enlighten wel doet, ik weet niet wat verschil is tussen deze twee apps van jullie tav werking en flows.

ketra, 2023-06-27:
> Ok, maar na installatie vraagt app ook om API key en die kun je als systeem eigenaar nergens in de enphase envoy Metered s gateway vinden, nemen jullie dat ook mee?
>
> Gr wout

De nieuwe feature die nu getest wordt praat lokaal met de gateway en zal alleen een enphase eigenaar account login nodig hebben

WouterAar, 2023-06-27:
> De nieuwe feature die nu getest wordt praat lokaal met de gateway en zal alleen een enphase eigenaar account login nodig hebben

Dat klinkt goed, is er bekend wanneer de nieuwe app te downloaden is?

En Heb je nog hulp nodig hierbij om te testen op een installatie, dan hou ik me aanbevolen om te helpen met info die nodig hebt.

ketra, 2023-06-27:
> Overigens gebruikte ik tot voor kort jullie app enphase envoy en die vraagt niet om user ID en API KEY zoals jullie app enphase enlighten wel doet, ik weet niet wat verschil is tussen deze twee apps van jullie tav werking en flows.

Voor flows geen verschil, de enphase enlighten praat lokaal met de gateway, enphase enlighten via de cloud api, waar een developer account voor nodig is, en daarnaast is die api betaald vanaf een x aantal verzoeken

WouterAar, 2023-06-28:
> Voor flows geen verschil, de enphase enlighten praat lokaal met de gateway, enphase enlighten via de cloud api, waar een developer account voor nodig is, en daarnaast is die api betaald vanaf een x [...]

Hoi thx, maar ik denk dat er iets in de tekst van je niet goed gaat, je schrijft twee keer enphase enlighten, kun je de tekst corrigeren zodat die klopt? Wat is verschil in app enphase enlighten VS enphase envoy

ketra, 2023-06-28:
@WouterAar comment aangepast.

WouterAar, 2023-07-03:
> > Overigens gebruikte ik tot voor kort jullie app enphase envoy en die vraagt niet om user ID en API KEY zoals jullie app enphase enlighten wel doet, ik weet niet wat verschil is tussen deze twee apps van jullie tav werking en flows.
> 
> 
> 
> Voor flows geen verschil, de enphase envoy praat lokaal met de gateway, enphase enlighten via de cloud api, waar een developer account voor nodig is, en daarnaast is die api betaald vanaf een x aantal verzoeken

Hi, zag vandaag automatische update app zonnepanelen naar versie 5.2. Heb app getest maar krijg bij enphase envoy (lokaal) onderstaande foutmelding. Is 5.2 geschikt voor de enphase met de laatste firmware of adviseer je mij om anders te testen?

[image]

SiskoBen, 2023-07-04:
Hi, 
I just installed the test version (from https://homey.app/en-us/app/it.diederik.solar/Solar-Panels/test/ ), I could add my Envoy and (after entering my envoy login credentials) it is reading the Power (production) as well as the Energy yield (today). So the token authentication system is working. It would be nice to also have these (power and energy) for (total-consumption), the (net-consumption) and the (storage [i.e. battery]) values from the local Envoy (provided the CTs for this and storage are installed and configured obviously). The individual inverter production (as reported by each individual microinverter to the Envoy with 5 minute intervals) would be a nice to have as well but not crucial.
Thanks for getting it working!
Best regards/Groetjes,
Arjan

DiedB, 2023-07-04:
@WouterAar The 5.2 version does not support Envoy v7 yet.

Please install this version, remove your Enphase Envoy from Homey, and then re-add the device as 'Enphase Envoy v7': https://homey.app/nl-nl/app/it.diederik.solar/Zonnepanelen/test/

Slofware, 2023-07-04:
I have got 5.2.1 and i can add the Envoy V7 but it keeps giving a value of -0- on both Vermogen and Energie Production

DiedB, 2023-07-04:
If it gives you 0, that would mean that the app is able to get data from your inverter and that it reports back those 0 values. What if you manually log into your Envoy, do you see production data?

Slofware, 2023-07-04:
Jup then I see the day production and current

DiedB, 2023-07-04:
Can you copy paste the data you get from https://192.168.x.x/api/v1/production?

DiedB, 2023-07-04:
> Hi, The Energy yield (Today) seems to be the value for lifetime production (or else i already produced 1302 kWh [more than 1mWh ] in 12 hours).

You are right, this has been fixed in the latest test version.

Slofware, 2023-07-04:
I have made a time ago a test version in Python where I also used “api/v1/production”  what I saw in you’re code and also then got -0-‘s as Data. When I used “ /production.json” it was oke and got all the correct Data

DiedB, 2023-07-04:
Hmm, that's weird. There is good reason to go with the API now - production.json is not accessible without a cookie, while the endpoints are meant for programmatic access, and a token is enough.

Can you find any other ocassions on the internet where this data is reported to be zero? What could be the cause?

Slofware, 2023-07-04:
CollectToken() {

        // Create basic client
        const client = axios.create({
            withCredentials: true,
        });
        // Get sessionID for later use to get token.
        const loginPayload = { 'user[email]': this.username, 'user[password]': this.password };
        let options = {
            headers: { 'content-type': 'application/x-www-form-urlencoded' },
        };
        client.post(ENLIGHTEN_AUTH_JSON_URL, loginPayload, options).then((res: AxiosResponse) => {
            const SessionId = res.data.session_id;
            const tokenPayload = { session_id: SessionId, serial_num: this.serial, username: this.username };

            options = {
                headers: { 'content-type': 'application/json' },
            };
            client.post(TOKEN_URL, tokenPayload, options).then((response: AxiosResponse) => {
                const token = <string>response.data.trim('\n');
                console.log(`Setting token to ${token}`);
                this.token = token;
                this.client.defaults.headers.common['Authorization'] = `Bearer ${token}`;

            });
        });
    }

    GetCookie() {
        this.client.defaults.headers.common['Authorization'] = `Bearer ${this.token}`;
        this.client.get('auth/check_jwt').then((result: any) => {

            if (result.data.toString().includes("Valid token.")) {
                this.sessionID = result.headers['set-cookie'][0].substring(10, 42);
                console.log(`Valid token, sessionId: ${this.sessionID}`);

            }
            else {
                this.token = '';
                this.sessionID = '';
            }
        });
    }

    static parseJwt(token: any) {
        const jsonToken = JSON.parse(Buffer.from(token.split('.')[1], 'base64').toString());
        return jsonToken;
    }

    GetData() {

        return new Promise((resolve, reject) => {
            this.client = axios.create({
                baseURL: `https://${this.hostname}/`,
                timeout: 10000,
                withCredentials: true,
                headers: {
                    'Content-Type': 'application/json',
                    Accept: 'application/json',
                },
                httpsAgent: new https.Agent({
                    rejectUnauthorized: false,
                }),
            });

            if (this.token === '') {
                console.log('Token not found, collecting token.');
                this.CollectToken();
            }

            if (this.token !== '') {
                console.log('Check token');
                this.GetCookie();
            }

            if (this.sessionID !== '') {
                console.log('Get production data');
                this.client.defaults.headers.common['Cookie'] = `sessionid=${this.sessionID}`;
                this.client.get('production.json').then((data: { data: object; }) => {
                    resolve(data.data);
                    console.log(data.data);
                }).catch((err: any) => {
                    reject(err);
                });

            }

        });
    }

Slofware, 2023-07-04:
This was working for me on a Homey2016, a bit stollen from you’re older app ;)

WouterAar, 2023-07-05:
> @WouterAar The 5.2 version does not support Envoy v7 yet.
> 
> 
> 
> Please install this version, remove your Enphase Envoy from Homey, and then re-add the device as 'Enphase Envoy v7': https://homey.app/nl-nl/app/it.diederik.solar/Zonnepanelen/test/

Het werkt met versie 5.2.2 experimantal, ik log in met mijn enphase account login en wachtwoord. Thx
Vraag is wel, werkt de app ook als je niet op de locatie vd zonnepanelen bent, dus op afstand middels 5G? En niet op de lokale WiFi zit.

DiedB, 2023-07-09:
Enlighten and local (IQ Gateway) are being confused in this topic a little bit. Let's switch over the discussion on Envoy to #215.

On Enlighten, I have experimented a little bit this morning, but I'm afraid it's gonna be quite complicated to support this. A reference implementation is on `feature/enlighten-v4`. Not entirely working but should be quite close. Main culprit is that everyone has to create their own OAuth app on the 'Watt' plan with Enlighten, and we'll have to ask for the client ID and secret in the pairing flow. Not sure if Athom will even approve that because it's a process that isn't too easy to follow.

Wondering how many people would be interested in having Enlighten supported as well (e.g., cases where local Envoy access does not work)? Please thumbs up this message.