DiedB/Homey-SolarPanels #236: Enphase Envoy

kneibdev, 2023-01-06:
When I add an Enphase Envoy device it says it is found but then not available. I think because it has the wrong IP. Must be 192.168.10.x

Device Unvailable
Error retrieving data (FetchError: request to http://**169.254.120.1**:80/production.json?details=1 failed, reason: connect ETIMEDOUT 169.254.120.1:80)

ferrydl, 2023-01-19:
Same here. Would be nice if you can inputoredit the IP adress manualy. In my case the IP adress is: 192.168.107.131. Most probably the IP is different per user.

kneibdev, 2023-01-19:
I also enabled wifi and now it detects the cable up 😉.
But now the cert problem

Slofware, 2023-01-27:
request to https://X.X.X.X/api/v1/production/ failed, reason: self signed certificate

Is that a Homey thing or a Enphase thing?

DiedB, 2023-01-27:
Both, Enphase changed some things and the Homey app needs to adapt to that.

Closing this issue as it is a duplicate of #215. Will work on a fix for this as soon as I can find the time.

closed 2023-01-27 by DiedB