DiedB/Homey-SolarPanels #240: Trigger missing Actual Energy Usage

peterverwolf, 2023-02-06:
Using Enphase Metered version. Making use of the Enphase Envoy device and the local API and getting the correct values in the device values. Also the realtime power consumption from home is correct.

But when creating a flow I do not have an option to create a trigger based on the power consumption. Is this by design? I would like to start a flow when power consumption is x KW. Is this possible?

DiedB, 2023-03-04:
https://homey.app/nl-nl/app/it.diederik.solar/Zonnepanelen/test/

Do you have the flow triggers if you install the new version?

closed 2023-03-04 by DiedB

peterverwolf, 2023-03-05:
Hey,
Yes the flow triggers are now in the test version. Thank you very much!

Outlook voor iOS<https://aka.ms/o0ukef> downloaden

https://homey.app/nl-nl/app/it.diederik.solar/Zonnepanelen/test/<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fhomey.app%2Fnl-nl%2Fapp%2Fit.diederik.solar%2FZonnepanelen%2Ftest%2F&data=05%7C01%7C%7Cd470df6cd87346961cb408db1cb52c56%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638135336685077660%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=bgPmHh9prpqCy69zFzNSbIYq8ScmGyct6QnY8gxgBzk%3D&reserved=0>