DiedB/Homey-SolarPanels #244: App zonnepanelen toegang growatt

Wouwmw, 2023-02-26:
Sinds enkele weken komt de app telkens weer in error. Na herstart doet hij weer even. Helaas de fix van december niet 100% geholpen

pro-sumer, 2023-02-26:
I noticed this as well, unfortunately.

Growatt seems to be very aggressive about preventing third-party access to their API.

We need to figure out which check they added now. Suggestions are welcome!

pro-sumer, 2023-02-26:
> The API data is available in the Energy app per 15 minutes. 

https://helpdesk.homewizard.com/en/articles/6121231-how-to-connect-my-growatt-inverter

Maybe we should experiment with that interval as well?

DiedB, 2023-03-04:
Rate limiting could be a thing indeed. Maybe good to test that on your local install @pro-sumer?

pro-sumer, 2023-03-17:
I'm first going to monitor it a bit in the Homey App, because it might also be just one of the many Growatt server issues...

pro-sumer, 2023-03-17:
Worked fine; the entire day.

(So unlikely that we need to change the update frequency now)

Wouwmw, 2023-04-09:
Do you know when this update comes? Last week I have to much errors with no connection with the app and no data.
Thanks for change the frequence. I hope that works fine.

Wouwmw, 2023-04-09:
I have also find this. At home assistance they had the same problems.
https://community.home-assistant.io/t/growatt-integration-roadmap/510221

Zakraket, 2023-05-18:
Yes, it doesn't work a lot of the time (like today, totally nothing). Yesterday it seems to have worked fine though

pro-sumer, 2023-05-18:
I haven’t had any issues for quite some time (unless my watchdog flow is broken…).

Can you still see your data at the Growatt website when the SolarPanels App is having issues accessing the Growatt servers?

koenvanmil, 2023-05-19:
I think I have the same issue. My growatt account is blocked regularly. Might have to do with rate limiting (max amount of API calls)? I have 2 Growatt inverters, both in homey, so probably make double the api calls compared to the average user.

When Homey cannot login anymore, the iPhone app is blocked as well. Using the website I still login and see my data.

Error in the iPhone app "de lopende rekening is vergrendeld"

Zakraket, 2023-05-19:
I also have two Growatts (and a SolarEdge)

On my android phone I'm getting: current account is locked. The website works.

koenvanmil, 2023-05-19:
Looks like exact same behaviour as mine. Only the dutch translation for the error message in the app is a bit "weird"

pro-sumer, 2023-05-19:
The “account locked” error is completely new to me and I’m surprised that even their own App is blocked then.

If this happens to a lot of users, we might have to start decreasing the update frequency to find a safer value.

koenvanmil, 2023-05-19:
I just read this topic, apparently their seems to be a rate limit, since february 2023. Just a conincedence that I added my second inverter around that time:
https://community.home-assistant.io/t/growatt-integration-roadmap/510221

pro-sumer, 2023-05-19:
The frequency is currently hard-coded.

Maybe @DiedB can make it configurable?

DiedB, 2023-06-10:
Improved the implementation such that the app makes less API requests. Let's see if it helps. Report back if you see this issue again with the newest version.

closed 2023-06-10 by DiedB

Wouwmw, 2023-08-19:
With the new version i had no problems till this week. I can not see what has changed by homey or growatt. But i have no connection anymore.

Please help

pro-sumer, 2023-08-19:
> With the new version i had no problems till this week. I can not see what has changed by homey or growatt. But i have no connection anymore.

Nobody seems to know yet; see #293 (please join the discussion there, instead of here).