DiedB/Homey-SolarPanels #248: Homey Pro 2023

Chopper8, 2023-03-07:
Hi! I am using this app. In the next weeks hope to receive the new Pro. Possible to have a look at the update? I am using the SolarCloud inverter. Many thanks for your time and effort!

Fischmaster3000, 2023-03-08:
+

Yes, would be awesome to get an update for this :)

Ronflon, 2023-03-11:
Hi, I fully support this request: it would be great if you could upgrade this app to be compatible with homey pro 2023. I am using it for reading out a Growatt inverter.

Many thanks for your time and effort!

EffectICT, 2023-03-15:
Hi, Ialso  fully support this request: it would be great if you could upgrade this app to be compatible with homey pro 2023. I am using it for reading out SolarEdge SK7 (true Api Key). When, do you think, the new  will  be aviable. Thanks in advanced!!

wolck, 2023-03-19:
Received my homey pro 2023, would be great to keep using the solar panels app:) Vote for this one;)

ictcorebiz, 2023-03-22:
I received my first homey (pro 2023), can you indicate when you plan to upgrade the app so that i can monitor my enlighten enphase solarpanels?

Anteboyan, 2023-03-23:
Would love an update for the 2023 version - testversion also! Would donate if needed!

DiedB, 2023-03-29:
Thanks for posting the link @WoutR. I'm working together with Athom on getting it published in the store.

I need some screencasts of the pairing process for each brand, if you could make a GIF and post it here I can share it with Athom. Would greatly help in getting the app published in the store.

spoelly, 2023-04-02:
First of all thanks for all the effort @DiedB, after reading @WoutR his comment i had to try the test version. But when i tried to add my Solar Edge system i could not give my system id anywhere. Is this a bug in the current test version? I can only add the API token on the screen, after that i get an error saying to check my system id and api token.

fagern, 2023-04-25:
Hi
Just installed the Beta/test app on my new 2023 PRO, and it works like a charm and found both my Solar Edge inverters.
But there is a little issue
It looks like the Power (Production), and Energy yeld (Today) shows the the combined (totalized) values from both inverters, and shows the same values in both tiles.
While the Inverter temperature, and Enery yeld (Total) shows the correct values.

RonaldNeed, 2023-05-03:
Hi, Thanks for the update. I installed the test version from the link from WoutR. Looks like it is working with my Enphase systems. To run my new Homey 2023 I used a backup/restore from Homey 2019. It (app) runs without any installation or settings. Al the settings were available when the test App was installed. Tomorrow there will be a sunny test day. Checking how the app behaves.

DiedB, 2023-06-10:
@fagern Feel free to supply me with an API key through slower_04_machete@icloud.com so I can debug that issue.

closed 2023-06-10 by DiedB