DiedB/Homey-SolarPanels #253: New solar panel owner

ECort, 2023-04-21:
Hi Diederik,
We had our solar panels installed this week. I hope to install my installation with your app. I'm not a very experienced user and 'api's' are difficult. I tried lot's of stuff but so far our Enphase Enlighten could not be created. Can you help?

Steps so far:
Enlighten
- created account at https://developer-v4.enphase.com/admin/account/stripe
(V4 api, v2account seems seperate, only on invite and expires?)
- account details: mail/ password. Username first 'random', afterwords changed on 'complex long code' User Id api in 'enlighten app - conncetion info'
- created 'app' with ficticous name. Succesfull and api key
- tried to use 'authorisarion link' but did not work and did not understand

Homey
- tried to add Enphase enlighten
- api key: from v4 api key (enlighten)
- user id: complex long users id from enlighten app

No idea id this is the peopwr procedure

Unfortunately 'object/object' error on enter.

Hope you can help or give instructions.
Kind regard, Eric

niekvugteveen, 2023-05-02:
Have you tried adding the Enphase Envoy instead?

DiedB, 2023-06-10:
Is there any way you can find the software version of your Envoy?

I would indeed strongly recommend going with Envoy (local connection), but Enphase changed their local API recently so that might be broken currently.

DiedB, 2023-07-10:
@ECort did you get this sorted by using Envoy?

closed 2023-07-10 by DiedB