DiedB/Homey-SolarPanels #256: Solar panel with SDK3

alechenne, 2023-05-07:
Dear Support,

I used your app with Homey 2019 and I was very satisfied. Did you plan to upgrade it to SDK3 for Homey 2023?

Thanks

DiedB, 2023-06-10:
It will soon be supported, you can test it here: https://homey.app/nl-nl/app/it.diederik.solar/Zonnepanelen/test/

closed 2023-06-10 by DiedB

alechenne, 2023-06-11:
Hi Diederik,

Thanks coming back to me. In the mean time, I have asked to the SolarEdge
support to activate the Modbus TCP on my inverter. In that case, I can use
the SolarEdge + Growatt TCP app for Homey 2023 and have higher refresh data
rate than through the Cloud API from SolarEdge.

Nevertheless, I installed your test app version and:
- This time I can add the device (by the past, it wast the case)
- After few seconds, I got all the 4 data fields with data; all the data
are ok
- After 15 min (the period interval of the settings), all the 4 data are
refreshed
- I don't test any workflow cards

Kind regards

Alexandre Léchenne
Les Pâquerettes 16
CH-2608 Courtelary
(Switzerland)

Mobile: +41 79 307 40 25
Email: ***@***.***