DiedB/Homey-SolarPanels #258: Homey pro 2023 app - Not able to add devices

koenvanmil, 2023-05-19:
When I try to add a Growatt device to the Homey pro 2023 app I get the following error after login in with my growatt password/username:

“invalid json response body at [http://server-apr.growatt.com/error.do?errorMess=errorNoLogin 1](http://server-apr.growatt.com/error.do?errorMess=errorNoLogin) reason: Unexpected token < in JSON at position 2”

pro-sumer, 2023-05-19:
Is “apr” a manual typo? Shouldn’t that be “api”?

koenvanmil, 2023-05-19:
Yes, but from my side, not from Diederik in this app... Sorry...

pro-sumer, 2023-05-19:
I think I have seen this error during development, but I can’t remember what triggered/fixed it.

Most probably the server is returning an HTML webpage instead of JSON data; I have seen that before.

Are you familiar with tools like curl/wget/HTTPie and Charles/Proxyman? (To get more insight)

koenvanmil, 2023-05-19:
No, not really...

I do know the wget comment from back in my days when I had a linux server, but that's been a while.

But with some instructions I'm always open to give it a try

pro-sumer, 2023-05-19:
I don't have a 2023 model myself and have not looked at all the software changes for that model.

Maybe @DiedB is better equipped to help here?

pro-sumer, 2023-05-21:
I was hoping to provide a [HomeyScript](https://homey.app/en-us/app/com.athom.homeyscript/HomeyScript/) script for testing, but all login requests using such scripts fail for me, while the same request in [RapidAPI for Mac](https://paw.cloud) works fine...

pro-sumer, 2023-05-21:
Growatt seems to be having server issues again; I keep getting 504 errors:

```
<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html>
<head><title>504 Gateway Time-out</title></head>
<body>
<h1>504 Gateway Time-out</h1>
<p>The gateway did not receive a timely response from the upstream server or application.<hr/>Powered by Tengine</body>
</html>
```

This could be the HTML response causing the error message reported in this bug...

(If so, there's nothing we can do...)

pro-sumer, 2023-05-21:
And 502's as well:

```
<html>
<head><title>502 Bad Gateway</title></head>
<body bgcolor="white">
<center><h1>502 Bad Gateway</h1></center>
<hr><center>nginx</center>
</body>
</html>
```

koenvanmil, 2023-05-21:
> Growatt seems to be having server issues again; I keep getting 504 errors:
>
> ```
> <!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
> <html>
> <head><title>504 Gateway Time-out</title></head>
> <body> [...]

If it’s the Growatt server, why does it still work in Homey 2019 then?

pro-sumer, 2023-05-21:
> If it’s the Growatt server, why does it still work in Homey 2019 then?

That's a good question!

I guess we're just really unlucky today...

koenvanmil, 2023-05-21:
I just tried. At this moment it still works in homey pro 2019 and not in 2023

Maybe we should wait to @DiedB  has time to have a look indeed

DiedB, 2023-05-22:
Strange, then there is probably some difference between the old SDKv2 code for Homey 2019 and the new implementation for SDKv3.

Here's the (apparently non-working) source of the SDKv3 code: https://github.com/DiedB/Homey-SolarPanels/tree/main/drivers/growatt
And here's the old one: https://github.com/DiedB/Homey-SolarPanels/tree/old/drivers/growatt

Could someone have a look and/or do some testing to see where the difference lies?

DiedB, 2023-05-22:
By the way, I installed a Growatt system at my relatives a few weeks ago so I also have access to my own Growatt account now. Could help with testing/supporting this in the future.

DiedB, 2023-06-10:
Tried today, pairing works fine with my account.

@koenvanmil If the issue persists, please share your credentials with me through slower_04_machete@icloud.com and I will see what I can do.

DiedB, 2023-06-10:
I reworked the Growatt implementation a little bit to support devices across multiple plants, thanks @koenvanmil for sharing your account for testing.

As a side-effect, all previously existing Growatt devices will need to be re-paired in Homey. An added benefit is that the app will make significantly less API calls to the Growatt server. As a result, I increased the data fetching interval to 15 minutes (from 60 minutes).

closed 2023-06-10 by DiedB